esp8266/Arduino #8212: ESP8266WebServer - Force client disconnect

jaggzh, 2021-07-12:
When dynamically sending out unknown length data, using sendContent() (one or more) -- upon completion on the server side -- how do we force disconnect the client?  At present the HTTP_MAX_CLOSE_WAIT stays in effect (2000ms).  (This means a curl of a URI that uses sendContent(), non-chunked, and unknown length content, will always take a minimum of 2 seconds, even if the content is delivered immediately).

(sendContent("") is only evaluated by ...-impl.h if the content is _chunked, where it then sets _chunked = false).

You can see my code here:
https://github.com/jaggzh/esp8266-wifi-lcd-touch/blob/main/esp8266-wifi-lcd-touch.ino#L317

The sendContent("k\n"); is received by curl immediately; however, the server doesn't close the connection for an additional 2 seconds (of course, for how would it know?)

I've tested various combinations of server.getServer().available().stop() and stuff, but nothing works.

So how do we trigger the connection to close?

Thank you all. I hope you are as safe as can be.

d-a-v, 2021-07-12:
Did you try with `server.client().close()` ?

jaggzh, 2021-07-12:
Thank you!  It's not close() but stop() though.
So far it seems to be working.
(Man I spent so many hours, if not days, on that.)

closed 2021-07-14 by jaggzh

jaggzh, 2021-08-30:
Can we add this clearly to the docs, perhaps in multiple places?