esp8266/Arduino #8217: SPISlave.h  Missing packets and all packets after first 32 byte packet are received incorrectly

ben-watkinson, 2021-07-14:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:			        Ai Thinker ESP-12F
- Core Version:			3.0.1
- Development Env:		Arduino IDE
- Operating System:		Windows 10

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			unknown
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

I am using the SPIslave.h library to receive data from a board developed by another company. the board is setup as the master and transmits 2000 SPI packets to the ESP8266 board that is configured as a slave.  Each packet is 32 bytes long and there is a 20ms delay between each packet. When receiving data the ESP8266 board only reads in just under 1000 packets. When the ESP8266 board receives the data it correctly receives the first packet of 32 bytes correctly but all subsequent packets are read incorrectly. I have set the sketch up to print out the first 5 packets to the Serial monitor and looks like this:

78,0,0,255,229,244,0,127,248,128,112,7,255,192,12,255,129,10,0,64,2,128,196,4,0,224,2,255,75,246,0,135,
204,0,128,2,128,165,248,0,64,0,255,192,12,0,191,255,0,248,11,255,64,8,255,86,12,0,133,255,75,246,0,135,
204,0,0,128,12,128,228,7,255,192,0,255,187,8,0,127,248,128,250,75,255,86,255,86,12,0,133,255,75,246,0,135,
204,0,0,12,128,226,111,255,64,8,255,18,98,0,63,255,0,63,163,255,63,245,127,15,173,255,150,255,75,246,0,135,
204,0,127,251,127,113,167,255,0,12,128,130,160,0,224,10,255,124,174,0,255,246,128,119,36,0,83,255,75,246,0,135

The first packet in this output matches what I see on an oscilloscope but the rest don't match the scope output. Also the last 5 bytes of all subsequent packets are always the same and i have no idea why.

I have included the SPISlave.onData function below but let me know if it would be worth including the full sketch. i based the function off the SPISlave example. the SPISlave.onData function is set up to add the data received over SPI into a buffer. I have also had an issue of the first 2 bytes of every packet being empty but i have managed to work around this.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

SPISlave.onData([](uint8_t * data, size_t len) 
  {
    
      
    
      (void) len;   
    
      if(count==0)
      {
        buff[count] = 0x4e; // this sets the start packet byte to the correct value to compensate for the missing first 2 bytes
        count++;
      }
      else
      {
        buff[count] = 0xcc; // this sets the start packet byte to the correct value to compensate for the missing first 2 bytes
        count++;
      }
      buff[count] = 0;
      count++;
      for(int i = 0; i<(len-2); i++) //loop to add the data to the buffer
      {  
        buff[count] = data[i];
        count++;
              
      }
      packet_count++;     //this is just used to count the number of packets recieved
     
   ;
      
  });
```

JiriBilek, 2021-07-14:
Just a question: are you aware of proprietary format of SPI messages received by the ESP8266? In the default setup, each SPI packet has a 2 bytes prembule. In your case I believe it should be 0x02 0x00 and then 32 bytes of data.

ben-watkinson, 2021-07-14:
I wasnt aware of that, would that need to be transmitted at the start of every packet by the SPI master or is it automatically added to the received packet by the ESP8266?

JiriBilek, 2021-07-14:
AFAIK, the packets have to have 2+32 bytes. ESP8266 sends the extra 2 bytes when acting as a master, I hope. But your case is different - your SPI master has to send the extra 0x02 0x00 bytes.

ben-watkinson, 2021-07-15:
ok so that must be why I'm losing the first 2 bytes of each packet. but it doesn't explain why i'm all packets after the first are read incorrectly and why im missing over half of the packets

JiriBilek, 2021-07-15:
The preambule is mandatory although it can be shorter than 2 bytes with changing values in ESP registers. But you can't get rid of it. 
The first byte is the command (0x00 to 0x03), write 2 bytes, read 2 bytes, write 32 bytes and read 32 bytes respectively. I don't know how and if the ESP evaluates the other bits in the first byte.

ben-watkinson, 2021-07-15:
ah ok so its not reading the data in correctly without the correct preamble at the start. but if thats the case then why is the first packet of 32 bytes read correctly

JiriBilek, 2021-07-15:
I don't know, there is no documentation about the ESP8266 SPI controller.

ben-watkinson, 2021-07-15:
Ok So do you think that adding the correct preamble to the front of the data packet will solve the problem?

Just to make sure ive understood correctly the transmitted packet should look something like this:

0x02, 0x00, 78,0,0,255,229,244,0,127,248,128,112,7,255,192,12,255,129,10,0,64,2,128,196,4,0,224,2,255,75,246,0,135,

JiriBilek, 2021-07-15:
I think so. 
There may be an issue with SS timing, in case of troubles give several microseconds before the SS signal is down and the transmission begins. Look at my previous code snippet and here is a function that changes the SS:
```cpp
void _pulseSS(bool start)
    {
        if (_ss_pin >= 0)
        {

            if (start) {
                digitalWrite(_ss_pin, LOW);
                delayMicroseconds(20);  // 10us is low (some errors), 15 is ok, 25 us is safe for speeds > 4MHz
            }
            else {
                digitalWrite(_ss_pin, HIGH);
            }
        }
    }

```

ben-watkinson, 2021-07-15:
ok thank you so much for your advice. i'll see if i can get those changes made to the transmitter and let you know if it works