esp8266/Arduino #8219: Time display problem in version 3.0.1

zoszabo72, 2021-07-16:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:			[ESP8286 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I have a display with date, time, temperature and humidity displayed on it.
Everything worked fine with 2.7.4 version, but till then the date is wrong and the ESP8266 is restarting.
What has been changed since 2.7.4 version?

Thanks for your help!

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <WiFiUdp.h>
#include <NTPClient.h>               // Include NTPClient library
.........................
WiFiUDP ntpUDP;

// 'time.nist.gov' is used (default server) with +2 hour offset (7200 seconds) 300 seconds (300000 milliseconds = 5 min) update interval
NTPClient timeClient(ntpUDP, "time.nist.gov", 7200, 300000);

................

void timeDisplay() {

  timeClient.update();

  unsigned long epochTime = timeClient.getEpochTime();

  String formattedTime = timeClient.getFormattedTime();

  //Get a time structure
  struct tm *ptm = gmtime ((time_t *)&epochTime);

  int monthDay = ptm->tm_mday;

  int currentMonth = ptm->tm_mon + 1;

  int currentYear = ptm->tm_year + 1900;

  //Print complete date: %4d - 4 decimal places, %02d - 2 decimal places with 0 leadings
  sprintf_P(DateString, PSTR("%4d.%02d.%02d"), currentYear, currentMonth, monthDay);

  // Send time and date to serial monitor
  Serial.println(" ");
  Serial.print("► Idő (time.nist.gov): ");
  Serial.print(DateString);
  Serial.print(" ");
  Serial.println(formattedTime);

  // Display time and date on the 0.96" LCD
  display.setTextSize(1);
  display.setTextColor(SSD1306_WHITE);
  display.setCursor(5, 25);
  display.print(DateString);
  display.print(" ");
  display.print(formattedTime);
  display.display();
}

```

### Debug Messages

```

12:44:27.913 -> ...START.........  
12:44:28.984 -> ► Idő (time.nist.gov): 5378547.03.08 02:00:19       HERE IS THE PROBLEM!!!
12:44:29.031 ->  Hőmérséklet °C (KOZTECH): 27.98
12:44:29.031 ->  Páratartalom % (KOZTECH): 48.34
12:44:29.031 ->  Légnyomás hPa  (KOZTECH): 1013.47
12:44:29.031 ->  Magasság m     (KOZTECH): 111.00
12:44:29.031 -> .................
12:44:29.031 ->  WiFi status: 3
12:44:29.031 ->  WiFi flag: 1
12:44:29.031 ->  WiFi SSID: Koztech
12:44:29.031 ->  WiFi RSSI: -64 dB
12:44:29.031 ->  WiFi IP:  192.168.0.160
12:44:29.031 ->  WiFi MAC: E0:98:06:85:E0:1D
12:44:29.031 ->  MQTT kapcs.: 1
12:44:29.031 -> ...........END...
12:44:29.031 -> 
12:44:37.904 -> ...START......... 

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
 
Exception (28):
epc1=0x40208933 epc2=0x00000000 epc3=0x40241fc7 excvaddr=0x00000064 depc=0x00000000
 
>>>stack>>>
 
ctx: cont
sp: 3ffffd30 end: 3fffffc0 offset: 0190
3ffffec0:  00000003 3ffeefe4 3ffef018 3ffeefdc  
3ffffed0:  3fffdad0 3ffeefd0 3ffeedac 40208a16  
3ffffee0:  40204cd4 00484345 3ffeedac 4020236b  
3ffffef0:  00000003 3ffeefe4 3ffef018 40202900  
3fffff00:  3fffdad0 3ffeefd0 3ffef364 40203da1  
3fffff10:  00000000 00686365 00000000 00000000  
3fffff20:  000f000f 00000000 40100505 00000000  
3fffff30:  00007071 2e313131 00003030 06000000  
3fffff40:  33313031 0039342e 07000000 302e3734  
3fffff50:  00000030 05000000 302e3832 00000039  
3fffff60:  05000000 3ffef5b8 40100505 00000000  
3fffff70:  f0722f17 3fef94c7 0000000a 4020a2fc  
3fffff80:  4020b385 0000000a 3ffef4a4 3ffef4a4  
3fffff90:  3fffdad0 00000000 3ffeef38 402045d4  
3fffffa0:  feefeffe 00000000 3ffef490 4020a41c  
3fffffb0:  feefeffe feefeffe 3ffe8654 40100f55  
<<<stack<<<
 
--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

enjoyneering, 2021-07-16:
NTPClient.h is the old way to get time from SNTP server. Everything is now included in Arduino, see example - [NTP-TZ-DST.ino](https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino)

d-a-v, 2021-07-16:
@zoszabo72 This is not an issue with the official Arduino library "NTPClient", because your *partial* MCVE works with current master of https://github.com/arduino-libraries/NTPClient without issue.

You have ticked this line:
> [] If there is a stack dump, I have decoded it.

But you did not decode the stack trace that was given by the esp.
Without it we can't further help.

```
14:45:08.646078131 192.168.43.207
14:45:10.898356750 
14:45:10.902407011 ► Idő (time.nist.gov): 5378547.03.08 02:00:08
14:45:10.906183990 5378547.03.08
14:45:10.907008998 02:00:08
14:45:14.142755340 
14:45:14.147085169 ► Idő (time.nist.gov): 5378598.09.21 14:45:13
14:45:14.151098997 5378598.09.21
14:45:14.153365152 14:45:13
14:45:17.143451065 
14:45:17.147616776 ► Idő (time.nist.gov): 5378598.09.21 14:45:16
14:45:17.151296072 5378598.09.21
14:45:17.152497581 14:45:16
14:45:20.145075527 
14:45:20.149154826 ► Idő (time.nist.gov): 5378598.09.21 14:45:19
14:45:20.152919266 5378598.09.21
14:45:20.154309150 14:45:19
14:45:23.146429599 
14:45:23.150998596 ► Idő (time.nist.gov): 5378598.09.21 14:45:22
14:45:23.154717562 5378598.09.21
14:45:23.156605726 14:45:22
14:45:26.147945414 
14:45:26.152299124 ► Idő (time.nist.gov): 5378598.09.21 14:45:25
14:45:26.155904905 5378598.09.21
14:45:26.157712460 14:45:25
14:45:29.149144069 
14:45:29.153661457 ► Idő (time.nist.gov): 5378598.09.21 14:45:28
14:45:29.157177598 5378598.09.21
14:45:29.158987828 14:45:28
```

zoszabo72, 2021-07-26:
Thanks for all. I will try it.

earlephilhower, 2021-08-19:
Closing w/no feedback and no problem found in the core after 24 days.

closed 2021-08-19 by earlephilhower