esp8266/Arduino #8220: Compile error when using fixed size integer macros from inttypes.h

ardui-no, 2021-07-16:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[other]
- Core Version:			[2021-07-16]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dout]
- Flash Size: 		[1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Compile error when using the fixed size integer macros from inttypes.h with library versions 3.0.0+ but was OK in previous versions (e.g. v2.7.4). The code example also compiles OK on GNU GCC v7.1.1 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <stdint.h>
#include <inttypes.h>
#include <stdio.h>

void setup() {
  char buffer[100] = {0};
  uint32_t v = 12345;
  
  sprintf(buffer,"%" PRIu32, v);
}

void loop() {
}

```

### Debug Messages
Error message is:
```
MCVE.ino: In function 'void setup()':
MCVE:10:21: error: expected ')' before '__INT32'
   10 |   sprintf(buffer,"%" PRIu32, v);
      |          ~          ^
      |                     )
exit status 1
expected ')' before '__INT32'
```

earlephilhower, 2021-07-17:
Repro'd.  The newlib needs an update to remove the stale xtensa-specific _intsup.h and use the standard one.  So a new toolchain build is required.

earlephilhower, 2021-07-17:
Reopening until I actually get a build running with the newly compiled toolchain!

PR #8222 (2021-07-26): Fix PRxxx printf format macros
Update toolchain (newlib) to supply proper definitions for PRxxx macros.
Fixes #8220

Added a PRxxx macro to an example to ensure CI will catch any problem like
this in the future.
commit cfb6d508 (2021-07-26): Fix PRxxx printf format macros (#8222)

* Fix PRxxx printf format macros

Update toolchain (newlib) to supply proper definitions for PRxxx macros.
Fixes #8220

Added a PRxxx macro to an example to ensure CI will catch any problem like
this in the future.
closed 2021-07-26 by d-a-v