esp8266/Arduino #8223: Add DHCP options while the ESP is requesting a server

1e1, 2021-07-17:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-01|Wemos D1 mini R2]
- Core Version:			[09c4e331]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2]
- Flash Mode:			[DOUT]
- Flash Size: 		[1MB|4MB]
- lwip Variant:   [v2 Lower Memory|v2 Higher Bandwidth]
- Reset Method:		[dtr]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

The ESP8266 is not sending the Client Identifier when it request a DHCP server. 

Some routers-gateway-dhcp require this option to register the ESP's hostname into their DNS.
On devices with several network interfaces it prevents the DHCP to assign an new address per interface. 

Can we amend the Client Identifier (option 61) to the DHCP query? 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
// cf https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/examples/mDNS_Web_Server/mDNS_Web_Server.ino
```

### Debug Messages

```
me@ime  /Users/me sudo dhcpdump -i en0
  TIME: 2021-07-18 00:30:26.692
    IP: 0.0.0.0 (b4:e6:2d:ZZ:ZZ:ZZ) > 255.255.255.255 (ff:ff:ff:ff:ff:ff)
    OP: 1 (BOOTPREQUEST)
 HTYPE: 1 (Ethernet)
  HLEN: 6
  HOPS: 0
   XID: 51fcZZZZ
  SECS: 0
 FLAGS: 0
CIADDR: 0.0.0.0
YIADDR: 0.0.0.0
SIADDR: 0.0.0.0
GIADDR: 0.0.0.0
CHADDR: b4:e6:2d:ZZ:ZZ:ZZ:00:00:00:00:00:00:00:00:00:00
 SNAME: .
 FNAME: .
OPTION:  53 (  1) DHCP message type         1 (DHCPDISCOVER)
OPTION:  57 (  2) Maximum DHCP message size 1500
OPTION:  55 (  5) Parameter Request List      1 (Subnet mask)
					      3 (Routers)
					     28 (Broadcast address)
					      6 (DNS server)
					     42 (NTP servers)

---------------------------------------------------------------------------

  TIME: 2021-07-18 00:30:26.794
    IP: 0.0.0.0 (b4:e6:2d:ZZ:ZZ:ZZ) > 255.255.255.255 (ff:ff:ff:ff:ff:ff)
    OP: 1 (BOOTPREQUEST)
 HTYPE: 1 (Ethernet)
  HLEN: 6
  HOPS: 0
   XID: 51fcZZZZ
  SECS: 0
 FLAGS: 0
CIADDR: 0.0.0.0
YIADDR: 0.0.0.0
SIADDR: 0.0.0.0
GIADDR: 0.0.0.0
CHADDR: b4:e6:2d:ZZ:ZZ:ZZ:00:00:00:00:00:00:00:00:00:00
 SNAME: .
 FNAME: .
OPTION:  53 (  1) DHCP message type         3 (DHCPREQUEST)
OPTION:  57 (  2) Maximum DHCP message size 1500
OPTION:  50 (  4) Request IP address        192.168.1.20
OPTION:  54 (  4) Server identifier         192.168.1.254
OPTION:  55 (  5) Parameter Request List      1 (Subnet mask)
					      3 (Routers)
					     28 (Broadcast address)
					      6 (DNS server)
					     42 (NTP servers)

OPTION:  12 ( 10) Host name                 ESP-ZZZZZZ
---------------------------------------------------------------------------

```

VERSUS a right DHCP request:

```bash
me@ime /Users/me sudo dhcpdump -i en0
  TIME: 2021-07-18 00:36:27.113
    IP: 0.0.0.0 (98:1:a7:ZZ:ZZ:ZZ) > 255.255.255.255 (ff:ff:ff:ff:ff:ff)
    OP: 1 (BOOTPREQUEST)
 HTYPE: 1 (Ethernet)
  HLEN: 6
  HOPS: 0
   XID: 7519ZZZZ
  SECS: 0
 FLAGS: 0
CIADDR: 0.0.0.0
YIADDR: 0.0.0.0
SIADDR: 0.0.0.0
GIADDR: 0.0.0.0
CHADDR: 98:01:a7:ZZ:ZZ:ZZ:00:00:00:00:00:00:00:00:00:00
 SNAME: .
 FNAME: .
OPTION:  53 (  1) DHCP message type         3 (DHCPREQUEST)
OPTION:  55 ( 11) Parameter Request List      1 (Subnet mask)
					    121 (Classless Static Route)
					      3 (Routers)
					      6 (DNS server)
					     15 (Domainname)
					    114 (URL)
					    119 (Domain Search)
					    252 (MSFT - WinSock Proxy Auto Detect)
					     95 (LDAP)
					     44 (NetBIOS name server)
					     46 (NetBIOS node type)

OPTION:  57 (  2) Maximum DHCP message size 1500
OPTION:  61 (  7) Client-identifier         01:98:01:a7:ZZ:ZZ:ZZ
OPTION:  50 (  4) Request IP address        192.168.1.11
OPTION:  51 (  4) IP address leasetime      7776000 (12w6d)
OPTION:  12 (  3) Host name                 iMe
---------------------------------------------------------------------------

  TIME: 2021-07-18 00:36:28.529
    IP: 0.0.0.0 (98:1:a7:ZZ:ZZ:ZZ) > 255.255.255.255 (ff:ff:ff:ff:ff:ff)
    OP: 1 (BOOTPREQUEST)
 HTYPE: 1 (Ethernet)
  HLEN: 6
  HOPS: 0
   XID: 7519ZZZZ
  SECS: 1
 FLAGS: 0
CIADDR: 0.0.0.0
YIADDR: 0.0.0.0
SIADDR: 0.0.0.0
GIADDR: 0.0.0.0
CHADDR: 98:01:a7:ZZ:ZZ:ZZ:00:00:00:00:00:00:00:00:00:00
 SNAME: .
 FNAME: .
OPTION:  53 (  1) DHCP message type         3 (DHCPREQUEST)
OPTION:  55 ( 11) Parameter Request List      1 (Subnet mask)
					    121 (Classless Static Route)
					      3 (Routers)
					      6 (DNS server)
					     15 (Domainname)
					    114 (URL)
					    119 (Domain Search)
					    252 (MSFT - WinSock Proxy Auto Detect)
					     95 (LDAP)
					     44 (NetBIOS name server)
					     46 (NetBIOS node type)

OPTION:  57 (  2) Maximum DHCP message size 1500
OPTION:  61 (  7) Client-identifier         01:98:01:a7:ZZ:ZZ:ZZ
OPTION:  50 (  4) Request IP address        192.168.1.11
OPTION:  51 (  4) IP address leasetime      7776000 (12w6d)
OPTION:  12 (  3) Host name                 iMe
---------------------------------------------------------------------------

  TIME: 2021-07-18 00:36:28.634
    IP: 192.168.1.254 (0:24:d4:ZZ:ZZ:ZZ) > 192.168.1.11 (98:1:a7:ZZ:ZZ:ZZ)
    OP: 2 (BOOTPREPLY)
 HTYPE: 1 (Ethernet)
  HLEN: 6
  HOPS: 0
   XID: 75198106
  SECS: 0
 FLAGS: 0
CIADDR: 0.0.0.0
YIADDR: 192.168.1.11
SIADDR: 0.0.0.0
GIADDR: 0.0.0.0
CHADDR: 98:01:a7:ZZ:ZZ:ZZ:00:00:00:00:00:00:00:00:00:00
 SNAME: .
 FNAME: .
OPTION:  53 (  1) DHCP message type         5 (DHCPACK)
OPTION:  54 (  4) Server identifier         192.168.1.254
OPTION:  51 (  4) IP address leasetime      864000 (1w3d)
OPTION:   1 (  4) Subnet mask               255.255.255.0
OPTION:   3 (  4) Routers                   192.168.1.254
OPTION:   6 (  8) DNS server                212.27.888.888,212.27.888.888
---------------------------------------------------------------------------
```

As you can see, the default Client Identifier is: 
OPTION:  61 (  7) Client-identifier         01:{{macAddress}}

And the field should be optional:
https://datatracker.ietf.org/doc/html/rfc4361

d-a-v, 2021-07-19:
DHCP client configuration is [configured there](https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/7421258237b7c8f61629226961af498a0a6e0096/glue-lwip/arduino/lwipopts.h#L3083-L3110).

Do you think you could propose a pull request on this repository ?

1e1, 2021-07-20:
@d-a-v I can try.

1e1, 2021-07-24:
fix at https://github.com/d-a-v/esp82xx-nonos-linklayer/pull/50

PR #8319 (2022-01-06): lwIP: v2.1.3 + dhcp fixes
fixes #8223 #8247
Also thanks to
- @1e1 (https://github.com/d-a-v/esp82xx-nonos-linklayer/pull/51)
- @mcspr (https://github.com/d-a-v/esp82xx-nonos-linklayer/pull/46) for the CMake build system
closed 2022-01-06 by d-a-v