esp8266/Arduino #8228: [Question]: return enum WRONG_Passowrd (ESP8266WIFISTA.cpp)

hasenradball, 2021-07-21:
Hello is it here better to return the Status 6 == WRONG_PASSWORD?

https://github.com/esp8266/Arduino/blob/8a42163a50e97fddd2dd1d7522fbe6bfc5eeb929/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L136

d-a-v, 2021-07-21:
Yes. Would you be ready to create a pull-request for this ?

hasenradball, 2021-07-21:
I will do a PR for this 😉

hasenradball, 2021-07-21:
Have You seen my post in Gitter chat?

d-a-v, 2021-07-21:
> Have You seen my post in Gitter chat?

You have many :) There's always a reason when no one answer to a particular post. If the issue is real/persistent, posting a new entry in the issue tracker like you did is OK and better because is stays visible and opened until one has time to understand and eventually answer.
One could say that you deleted the issue template, and you did, but providing a solution to an issue is, I dare to think, a valid exception.

hasenradball, 2021-07-21:
Hi sorry added the the template and filled out :-)

hasenradball, 2021-07-21:
Hi I also recognized if using `waitForConnectResult` --> it returns also no `WL_WRONG_PASWORD`;

```
int8_t ESP8266WiFiSTAClass::waitForConnectResult(unsigned long timeoutLength) {
    //1 and 3 have STA enabled
    if((wifi_get_opmode() & 1) == 0) {
        return WL_DISCONNECTED;
    }
    // if probing doesn't trip, this yields
    using oneShotYieldMs = esp8266::polledTimeout::timeoutTemplate<false, esp8266::polledTimeout::YieldPolicy::YieldOrSkip>;
    oneShotYieldMs timeout(timeoutLength); // number of milliseconds to wait before returning timeout error
    while(!timeout) {
        if(status() != WL_DISCONNECTED) {
            return status();
        }
    }
    return -1; // -1 indicates timeout
}
```

hasenradball, 2021-07-21:
I tried to get the status `WL_WRONG_PASSWORD` by using the folllowing test code (see below), but giving a wrong password does not result in `Wifi.status() == WL_WRONG_PASSWORD` why???

Does the NOTE has an impact on this?

[image]

```
#include <Arduino.h>
#include <ESP8266WiFi.h>

constexpr const char *SSID {"SSID"};
constexpr const char *PASSWORD {"PASSWD"};

wl_status_t get_wifi_status(void) {
  // keep in mind
  /*
    WL_NO_SHIELD        = 255,   // for compatibility with WiFi Shield library
    WL_IDLE_STATUS      = 0,
    WL_NO_SSID_AVAIL    = 1,
    WL_SCAN_COMPLETED   = 2,
    WL_CONNECTED        = 3,
    WL_CONNECT_FAILED   = 4,
    WL_CONNECTION_LOST  = 5,
    WL_WRONG_PASSWORD   = 6,
    WL_DISCONNECTED     = 7
  */
  wl_status_t status = WiFi.status();
  if (status == WL_NO_SHIELD) {
     Serial.println(F("\n WiFI.status =  NO_SHIELD"));
  }
  else if (status == WL_IDLE_STATUS) {
    Serial.println(F("\n WiFI.status =  IDLE_STATUS"));
  }
  else if (status == WL_NO_SSID_AVAIL) {
    Serial.println(F("\n WiFI.status =  NO_SSID_AVAIL"));
  }
  else if (status == WL_SCAN_COMPLETED) {
    Serial.println(F("\n WiFI.status =  SCAN_COMPLETED"));
  }
  else if (status == WL_CONNECTED) {
    Serial.println(F("\n WiFI.status =  CONNECTED"));
  }
  else if (status == WL_CONNECT_FAILED) {
    Serial.println(F("\n WiFI.status =  CONNECT_FAILED"));
  }
  else if (status == WL_CONNECTION_LOST) {
    Serial.println(F("\n WiFI.status =  CONNECTION_LOST"));
  }
  else if (status == WL_WRONG_PASSWORD) {
    Serial.println(F("\n WiFI.status =  WRONG_PASSWORD"));
  }
  else if (status == WL_DISCONNECTED) {
    Serial.println(F("\n WiFI.status = DISCONNECTED"));
  }
  else {
     Serial.println(F("\n No appropriate Status available!"));
  }
  return status;
}

bool wifi_connect(const char *_SSID, const char * _PASSWORD, const char *_hostname = nullptr){
  Serial.printf("Set WiFi mode to WIFI_STA - %d\n", WiFi.mode(WIFI_STA));
  if (_hostname != nullptr) WiFi.hostname(_hostname);
  // Warte auf Verbindung
  WiFi.begin(_SSID, nullptr);
  while (WiFi.status() != WL_CONNECTED) {
     Serial.printf(" 1 - Wifi status: %d\n", wifi_station_get_connect_status());
     Serial.printf(" 2 - Wifi status: %d\n", WiFi.status());
     Serial.printf(" 3 - Wifi status: %d\n", get_wifi_status());
     delay(500);
  }
  return true;
}
  /*
   * if (WiFi.waitForConnectResult() == WL_CONNECTED) {
    Serial.printf(" Wifi status: %d\n", get_wifi_status());
    Serial.println("\n Connection - success! <<<");
    return true;
  }
  else {
    Serial.printf(" Wifi status: %d\n", WiFi.status());
    Serial.printf(" Wifi status: %d\n", get_wifi_status());
    Serial.println("\n Connection - failed! <<<");
    return false;
  }
}
*/

void setup() {
  Serial.begin(115200);
  while (!Serial) yield();
  Serial.print("\n\n");
  WiFi.setAutoReconnect(false);
  WiFi.setAutoConnect(false);
  WiFi.persistent(false);

  time_t tic {micros()};
  if (wifi_connect(SSID, PASSWORD)) {
    Serial.printf("time until connect: %lld µs\n", (micros() - tic));
    delay(500);
    WiFi.disconnect();
  }
}

void loop() {
  // put your main code here, to run repeatedly:

}
```