esp8266/Arduino #8231: Unable to set HTTPClient to a std::optional

paulocsanz, 2021-07-23:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12
- Core Version:			3.0.1
- Development Env:		Platformio
- Operating System:		Ubuntu

### Problem Description

We are unable to set a HTTPClient to a std::optional<HTTPClient> that already exists. I have no idea why, the compiler generates hundreds of lines of error all related to the fact that `std::optional<HTTPClient>(std::optional<HTTPClient>&&)` is implicitly deleted. So right now it's impossible to set a HTTPClient to a std::optional, and I have no idea how to make it work, or provide a PR for this (although I'm willing if pointed in the right direction). But this seems a bug. And is something we actually need (we memset the unused 4kb sys stack to zero and cast it to a struct that contains a bunch of classes, and the easier way to properly initialize them is to wrap them in an optional).

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266HTTPClient.h>
#include <optional>

void setup() {
    std::optional<HTTPClient> http;
    http = std::make_optional(HTTPClient());
}

void loop() {

}

```

### Debug Messages

It's a huge compiler error: https://pastebin.com/2NZzvrgY

The core of it seems this:
```
include/loop.hpp:189:57: error: use of deleted function 'std::optional<HTTPClient>& std::optional<HTTPClient>::operator=(std::optional<HTTPClient>&&)'
  189 |       this->data->http = std::make_optional(HTTPClient());

.platformio/packages/toolchain-xtensa@2.100200.0/xtensa-lx106-elf/include/c++/10.2.0/optional:659:11: note: 'std::optional<HTTPClient>& std::optional<HTTPClient>::operator=(std::optional<HTTPClient>&&)' is implicitly deleted because the default definition would be ill-formed:
  659 |     class optional
      |           ^~~~~~~~

.platformio/packages/toolchain-xtensa@2.100200.0/xtensa-lx106-elf/include/c++/10.2.0/optional:659:11: error: use of deleted function 'std::_Enable_copy_move<false, false, false, false, _Tag>& std::_Enable_copy_move<false, false, false, false, _Tag>::operator=(std::_Enable_copy_move<false, false, false, false, _Tag>&&) [with _Tag = std::optional<HTTPClient>]'

.platformio/packages/toolchain-xtensa@2.100200.0/xtensa-lx106-elf/include/c++/10.2.0/bits/enable_special_members.h:306:5: note: declared here
  306 |     operator=(_Enable_copy_move&&) noexcept                         = delete;

.platformio/packages/toolchain-xtensa@2.100200.0/xtensa-lx106-elf/include/c++/10.2.0/optional: In instantiation of 'constexpr std::optional<typename std::decay<_Tp>::type> std::make_optional(_Tp&&) [with _Tp = HTTPClient; typename std::decay<_Tp>::type = std::decay<HTTPClient>::type]':
include/loop.hpp:189:57:   required from here .platformio/packages/toolchain-xtensa@2.100200.0/xtensa-lx106-elf/include/c++/10.2.0/optional:1207:62: error: no matching function for call to 'std::optional<HTTPClient>::optional(<brace-enclosed initializer list>)'
 1207 |     { return optional<decay_t<_Tp>> { std::forward<_Tp>(__t) }; }

```

And it goes on and on.

d-a-v, 2021-07-23:
I tried with g++11 and the message is different:
```
arduino/8231//8231.ino: In function ‘void setup()’:
arduino/8231//8231.ino:8:30: error: no matching function for call to ‘make_optional(HTTPClient)’
    8 |     http = std::make_optional(HTTPClient());
      |            ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
In file included from arduino/8231//8231.ino:4,
                 from arduino/8231/8231.ino.cpp:1:
/usr/include/c++/11/optional:1261:5: note: candidate: ‘template<class _Tp> constexpr std::enable_if_t<is_constructible_v<typename std::decay<_Tp>::type, _Tp>, std::optional<typename std::decay<_Tp>::type> > std::make_optional(_Tp&&)’
 1261 |     make_optional(_Tp&& __t)
      |     ^~~~~~~~~~~~~
```

`HTTPClient` does not have move contructor. Did you try to add one ?

paulocsanz, 2021-07-23:
No I haven't, I can do that tonight. But why should it be implicitly deleted? I'm not that keen on rules on default move constructors.

mcspr, 2021-07-23:
Quoting earlier gitter thread:

https://gitter.im/esp8266/Arduino?at=60f7e54a7331d202b5c10f43
> https://godbolt.org/z/PTPabM1z3, try to =delete things and see the same error

> ...and the helper class for optional seems to hide the actual error behind this check https://github.com/gcc-mirror/gcc/blob/e0335bb7d1fc7dd05e91bcdd1f65b2bcf8ec1a09/libstdc%2B%2B-v3/include/std/optional#L673, then https://github.com/gcc-mirror/gcc/blob/e0335bb7d1fc7dd05e91bcdd1f65b2bcf8ec1a09/libstdc%2B%2B-v3/include/bits/enable_special_members.h#L107 and downwards it deletes assignment

and also referencing https://en.cppreference.com/w/cpp/language/default_constructor and https://en.cppreference.com/w/cpp/language/move_assignment, you *must* implement constructors and assignment operators yourself if compiler can't do it for you. For the above, HTTPClient is cannot pass the `__and_v<is_move_constructible<_Tp>, is_move_assignable<_Tp>>` check. So these need to be manually implemented:

```cpp
HTTPClient::HTTPClient(HTTPClient&&);
```
```cpp
HTTPClient& HTTPClient::operator=(HTTPClient&&);
```

My earlier suggestion was to also look at WifiClient, since it can't (?) be moved properly. There are issues with ClientContext* member deleting itself though, but that's whole other can of worms

paulocsanz, 2021-07-23:
It works! The implementation is in PR #8236

paulocsanz, 2021-07-24:
Done!

PR #8237 (2021-10-13): Movable HTTPClient and fixing WiFiClient copy
- =default for default ctor, destructor, move ctor and the assignment move
- use std::unique_ptr<WiFiClient> instead of raw pointer
- implement clone() for the WiFiClient so it's safe to copy the WiFiClientSecure, and not accidentally call the WiFiClient copy instead
- replace headers pointer array with unique_ptr<T[]> to simplify the move operations
- substitute userAgent with the default one when it is empty

Allow HTTPClient to be placed inside of movable classes (e.g. std::optional) or to be returned from functions. Class logic stays as-is, onl [... cut]