esp8266/Arduino #8234: SD Card uses 7x as much power when sleeping under 3.0.1 than 2.7.4

daledj, 2021-07-23:
### Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ x] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [ na] If there is a stack dump, I have decoded it.
- [ x] I have filled out all fields below.

#### Platform

- Hardware:					[LOLIN D1 mini Pro 2.0.0]
- Core Version:			[3.0.1]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[LOLIN(WEMOS) D1 mini pro]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[16MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [921600] (serial upload only)

### Problem Description

When complied an run under core 2.7.4 the sketch below uses less than 1 mA when in deep sleep. When compiled and run with core 3.0.1, the same sketch uses about 7 mA when in deep sleep. This is a problem for applications that are designed to run for long times on a battery power source. Any idea why this is happening or how to debug and fix it?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

#define DEBUG_LOGFILE "test.txt"

#define DEBUG_MSG(...) Serial.printf( __VA_ARGS__ ), confile.printf( __VA_ARGS__ )

#include <SPI.h>
#include <SD.h>

File confile;
const int chipSelect = D8;

void setup() {
  Serial.begin (74880 );
  Serial.setTimeout( 2000 );
  // Wait for serial to initialize.
  while ( !Serial ) { }

  SD.begin(chipSelect);
  confile = SD.open(DEBUG_LOGFILE, "a");
  
  for (int i=0; i<20; i++) {
    DEBUG_MSG("%d: The quick brown fox jumps over the lazy dog.\n", i);
  }
  DEBUG_MSG("\n");
  confile.close();
  SD.end();

  ESP.deepSleep( 5 * 1e6, WAKE_RF_DEFAULT );  // Does not return. Restarts at setup()
}

void loop() {

}

### Debug Messages

daledj, 2021-07-29:
So after more testing I have discovered that the problem only happens with certain (offbrand?) SD cards. Still not sure why it does not happen with core 2.7.4 and it does happen with 3.0.1, but I suspect that the low level code that turns off the SD card has changed in 3.0.1.

earlephilhower, 2021-07-30:
The underlying SDFS code was rev'd from upstream 1.x to 2.0 for the 3.0 release.  We don't change code there, but if you want you can poke around https://github.com/greiman/SdFat and see if there's anything in particular of note.

Seeing as this seems to be only for certain cards, and related to an upstream dependency, closing here.  If there's a new version of SDFat we need to pull in, we can always reopen and work from there.

closed 2021-07-30 by earlephilhower