esp8266/Arduino #8239: [E][WiFiUdp.cpp:183] endPacket(): could not send data: 12

francescot92, 2021-07-26:
Hi guys. 
For my project I use Esp32 W-ROVER IE and Platformio to compiling the sketch. I wish to send a data to a localHost server. For this reasons i use the WiFiUdp library.
Sometimes in SerialTerminal i see these errors:

[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12

this error is a relative error because when the udp packet is not sent i try again in order to the packet will not be lost but,  when this error is a ricorsive error causes the CORRUP HEAP and consequently the ESP32 restart:

[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
[E][WiFiUdp.cpp:183] endPacket(): could not send data: 12
CORRUPT HEAP: Bad tail at 0x3fa5b704. Expected 0xbaad5678 got 0xa9836dd5
abort() was called at PC 0x400899e1 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008d500:0x3ffd9160 0x4008d779:0x3ffd9180 0x400899e1:0x3ffd91a0 0x40089b0d:0x3ffd91d0 0x401067d7:0x3ffd91f0 0x401028a1:0x3ffd94b0 0x40102829:0x3ffd9500 0x400922dd:0x3ffd9530 0x4008266e:0x3ffd9550 0x400d4064:0x3ffd9570 0x4008e7be:0x3ffd95c0

Rebooting...

how can i do? Why does this happen? 

Thanks

d-a-v, 2021-07-26:
> how can i do? Why does this happen?

You should probably be opening an issue case on the esp32/Arduino repository.

francescot92, 2021-07-26:
ok thanks