esp8266/Arduino #8247: Hostname not set for DISCOVER packets - DHCP client hostname not present on some routers

ilium007, 2021-07-27:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				[other]
- Core Version:			[3.0.2]
- Development Env:		[Platformio]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

The hostname is not being set in the DISCOVER packet sent by the ESP8266. Some routers (Mikrotik definitely) reject the DHCP 'Option 12 - hostname' field in the REQUEST packet if it has not been sent in the DISCOVER packet previously.

This same issue has been fixed for the ESP32 in this commit: https://github.com/espressif/esp-lwip/commit/13665c14e19dfda6b2ad30c1e5d88e0415cf8b7a

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

N/A

### Debug Messages
N/A

Detailed explanation and packet captures in this ESPHome issue: https://github.com/esphome/issues/issues/2069#issuecomment-883854019

d-a-v, 2021-07-27:
duplicate #8247 and https://github.com/d-a-v/esp82xx-nonos-linklayer/pull/50 by @1e1

This will be addressed soon

1e1, 2021-07-27:
@ilium007 I've tested the fix on a Mikrotik router with a custom script like here:
https://forum.mikrotik.com/viewtopic.php?t=119469

ilium007, 2021-07-27:
Issue https://github.com/esp8266/Arduino/issues/8223 references "OPTION: 61 ( 7) Client-identifier" This issue I have raised is about OPTION 12 Hostname.

ilium007, 2021-07-27:
> @ilium007 I've tested the fix on a Mikrotik router with a custom script like here:
> https://forum.mikrotik.com/viewtopic.php?t=119469

Thats awesome. Thanks.

d-a-v, 2021-07-27:
Different option, reopening

1e1, 2021-07-27:
@d-a-v @ilium007 
It fixes the hostname so ;) 
=> https://github.com/d-a-v/esp82xx-nonos-linklayer/pull/50/files#diff-710fdb9ab0a46e9cdc5204551f23701152eba187976cd0f68950398ec334b2b8R41

What I get is what you see below (OPTION 12 & OPTION 51): 
```
  TIME: 2021-07-24 17:54:36.163
    IP: 0.0.0.0 (b4:e6:2d:ZZ:ZZ:ZZ) > 255.255.255.255 (ff:ff:ff:ff:ff:ff)
    OP: 1 (BOOTPREQUEST)
 HTYPE: 1 (Ethernet)
  HLEN: 6
  HOPS: 0
   XID: b8c8ZZZZ
  SECS: 0
 FLAGS: 0
CIADDR: 0.0.0.0
YIADDR: 0.0.0.0
SIADDR: 0.0.0.0
GIADDR: 0.0.0.0
CHADDR: b4:e6:2d:ZZ:ZZ:ZZ:00:00:00:00:00:00:00:00:00:00
 SNAME: .
 FNAME: .
OPTION:  53 (  1) DHCP message type         1 (DHCPDISCOVER)
OPTION:  57 (  2) Maximum DHCP message size 1500
OPTION:  12 (  7) Host name                 myTestHostname
OPTION:  61 (  7) Client-identifier         01:b4:e6:2d:ZZ:ZZ:ZZ
OPTION:  55 (  5) Parameter Request List      1 (Subnet mask)
					      3 (Routers)
					     28 (Broadcast address)
					      6 (DNS server)
					     42 (NTP servers)
```

d-a-v, 2021-07-27:
So it is not duplicate but is adressed by the same fix right ?

1e1, 2021-07-27:
Yes, I opened another PR for the Client Identifier, this one is for Hostname.
The function was here but it was never called, I have just made a side-fix.

ilium007, 2021-07-30:
I had noticed client identifier as well as hostname is missing from the Mikrotik DHCP lease table. It would be good to have both included in the DHCP DISCOVER and REQUEST packets.

1e1, 2021-07-30:
@ilium007 it's merging https://github.com/d-a-v/esp82xx-nonos-linklayer/pull/51
Then there is to update the submodule version here.

d-a-v, 2021-09-25:
@ilium007 Can you try the unofficial alpha release v0.0.2 (https://d-a-v.github.io/esp8266/Arduino/) ?

1e1, 2021-10-19:
Hi @d-a-v 
It seems OK on my side.

mcspr, 2022-01-11:
closing via https://github.com/esp8266/Arduino/pull/8319

closed 2022-01-11 by mcspr