esp8266/Arduino #8254: Can't use mDNSResolver library with ESP8266mDNS parallelly

TavakoliAli, 2021-08-01:
Hello
The [mDNSResolver](https://github.com/madpilot/mDNSResolver) and ESP8266mDNS libraries works correctlly when i compile those separately but when i add following section of [mDNS_Web_Server](https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266mDNS/examples/mDNS_Web_Server) example :
```
  if (!MDNS.begin("esp8266")) {
    Serial.println("Error setting up MDNS responder!");
    while (1) {
      delay(1000);
    }
  }
  MDNS.addService("http", "tcp", 80);
```

To [example](https://github.com/madpilot/mDNSResolver/blob/master/examples/mDNSResolver.ino) of mDNSResolver library as follwing :
```
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <mDNSResolver.h>
#include <ESP8266mDNS.h>
#include <WiFiClient.h>

#define WIFI_AP "ssid"
#define WIFI_PASS "pass"
#define NAME_TO_RESOLVE "myhost.local"

using namespace mDNSResolver;

WiFiClient wifiClient;
WiFiUDP udp;
Resolver resolver(udp);

void setup()
{
  Serial.begin(115200);
  int WiFiCounter = 0;

  // We start by connecting to a WiFi network
  Serial.println("Connecting to ");
  Serial.println(WIFI_AP);

  WiFi.disconnect();
  WiFi.mode(WIFI_STA);
  WiFi.begin(WIFI_AP, WIFI_PASS);
  while (WiFi.status() != WL_CONNECTED && WiFiCounter < 30)
  {
    delay(1000);
    WiFiCounter++;
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  Serial.print("Resolving ");
  Serial.println(NAME_TO_RESOLVE);

  resolver.setLocalIP(WiFi.localIP());
  IPAddress ip = resolver.search(NAME_TO_RESOLVE);
  
  if (ip != INADDR_NONE)
  {
    Serial.print("Resolved: ");
    Serial.println(ip);
  }
  else
  {
    Serial.println("Not resolved");
  }

  /************************************************** Added Code *********************************************/
  if (!MDNS.begin("esp8266"))
  {
    Serial.println("Error setting up MDNS responder!");
    while (1)
    {
      delay(1000);
    }
  }
  Serial.println("mDNS responder started");

  // Add service to MDNS-SD
  MDNS.addService("http", "tcp", 80);
  /************************************************** End *********************************************/
}

void loop()
{
  // Required to clear the UDP buffer, as we can't unjoin a multicast group yet
  resolver.loop();

  MDNS.update();
}
```

The MDNS can't be initialized and `Error setting up MDNS responder!` will print on serial console.
If i move added code above the ` IPAddress ip = resolver.search(NAME_TO_RESOLVE);` line it will work correctly.

What is that problem and how can i fix that?

Tnx

hreintke, 2021-08-02:
The issue in not in esp8266 MDNS but in resolver library.

Both libraries try to listen to the MDNS multicast port but only the first can succeed.

The MDNS library checks whether the bind is OK and reports failing.
The responder library just call the listen and expect it to complete OK.

Update in  the mDNSResolver.cpp in read() the

```
udp.beginMulticast(localIP, MDNS_BROADCAST_IP, MDNS_PORT);
```
to
```
      if (udp.beginMulticast(localIP, MDNS_BROADCAST_IP, MDNS_PORT))
      {
    	  Serial.printf("Resolver : Multicast OK\r\n");
      }
      else
      {
    	  Serial.printf("Resolver : Multicast Failed\r\n");
      }
```
And you will see that it fails.

You cannot have a sketch with both libraries active at the same time.

TavakoliAli, 2021-08-03:
Thanks

> And you will see that it fails.

But it resolve host name to IP correctly.

> You cannot have a sketch with both libraries active at the same time.

Are there an easy way to directly resolve a host name to IP with MDNS library? 
I checked the [mDNS-SD_Extended](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/examples/mDNS-SD_Extended/mDNS-SD_Extended.ino) example and saw that how it resolve host name, It first query services and then fetch service details but in my network there aren't  any service attached to host and i don't want use "dns-sd" protocol.

hreintke, 2021-08-03:
Don't know how resolver lib can still resolve hostname.

`WiFi.hostByName(hostname, ipaddress, timeout)`

works for "web addresses" and "local addresses"

TavakoliAli, 2021-08-04:
Thanks
 Is there an easy way to set a local host on esp8266 (only through one function)?

hreintke, 2021-08-05:
WiFi.setHostname(...)
MDNS.begin(....)

closed 2022-07-12 by mcspr