esp8266/Arduino #8268: Problem with compiling <DHT.h> for NodeMCU

amnah96, 2021-08-14:
Hi everyone,

I am trying to program my NodeMCU, in order to read a DHT11 sensor. 
But even when I add the <DHT.h> library to my code I get an error.

for example when I try to add these libraries:

`#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <Hash.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <Adafruit_Sensor.h>
#include <DHT.h>`

 I was wondering, if anyone can help me.I get the following error:

In file included from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/HardwareSerial.h:31,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/Arduino.h:288,
                 from C:\Users\ASUSCenter\Documents\Arduino\test_temp_sens_esp\test_temp_sens_esp.ino:7:
c:\users\asuscenter\documents\arduino\libraries\include\time.h:116:22: error: conflicting declaration 'typedef uint32_t time_t'
  116 |     typedef uint32_t time_t;
      |                      ^~~~~~
In file included from c:\users\asuscenter\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\sys\select.h:15,
                 from c:\users\asuscenter\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\sys\types.h:50,
                 from c:\users\asuscenter\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\stdio.h:61,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/Arduino.h:32,
                 from C:\Users\ASUSCenter\Documents\Arduino\test_temp_sens_esp\test_temp_sens_esp.ino:7:
c:\users\asuscenter\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\sys\_timeval.h:42:18: note: previous declaration as 'typedef __int_least64_t time_t'
   42 | typedef _TIME_T_ time_t;
      |                  ^~~~~~
In file included from c:\users\asuscenter\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\sys\time.h:414,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2/tools/sdk/lwip2/include/lwipopts.h:3635,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2/tools/sdk/lwip2/include/lwip/opt.h:51,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2/tools/sdk/lwip2/include/lwip/init.h:40,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/IPAddress.h:27,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src/ESP8266WiFi.h:31,
                 from C:\Users\ASUSCenter\Documents\Arduino\test_temp_sens_esp\test_temp_sens_esp.ino:8:
c:\users\asuscenter\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\time.h:37:8: error: redefinition of 'struct tm'
   37 | struct tm
      |        ^~
In file included from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/HardwareSerial.h:31,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/Arduino.h:288,
                 from C:\Users\ASUSCenter\Documents\Arduino\test_temp_sens_esp\test_temp_sens_esp.ino:7:
c:\users\asuscenter\documents\arduino\libraries\include\time.h:159:12: note: previous definition of 'struct tm'
  159 |     struct tm {
      |            ^~
In file included from c:\users\asuscenter\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\sys\time.h:414,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2/tools/sdk/lwip2/include/lwipopts.h:3635,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2/tools/sdk/lwip2/include/lwip/opt.h:51,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2/tools/sdk/lwip2/include/lwip/init.h:40,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/IPAddress.h:27,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src/ESP8266WiFi.h:31,
                 from C:\Users\ASUSCenter\Documents\Arduino\test_temp_sens_esp\test_temp_sens_esp.ino:8:
c:\users\asuscenter\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\time.h:57:11: error: conflicting declaration of C function 'double difftime(time_t, time_t)'
   57 | double    difftime (time_t _time2, time_t _time1);
      |           ^~~~~~~~
In file included from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/HardwareSerial.h:31,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/Arduino.h:288,
                 from C:\Users\ASUSCenter\Documents\Arduino\test_temp_sens_esp\test_temp_sens_esp.ino:7:
c:\users\asuscenter\documents\arduino\libraries\include\time.h:128:21: note: previous declaration 'int32_t difftime(time_t, time_t)'
  128 |     int32_t         difftime(time_t time1, time_t time0);
      |                     ^~~~~~~~
In file included from c:\users\asuscenter\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\sys\time.h:414,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2/tools/sdk/lwip2/include/lwipopts.h:3635,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2/tools/sdk/lwip2/include/lwip/opt.h:51,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2/tools/sdk/lwip2/include/lwip/init.h:40,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/IPAddress.h:27,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src/ESP8266WiFi.h:31,
                 from C:\Users\ASUSCenter\Documents\Arduino\test_temp_sens_esp\test_temp_sens_esp.ino:8:
[... 40 lines cut ...]
      |                     ^~~~~~~~
In file included from c:\users\asuscenter\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\sys\time.h:414,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2/tools/sdk/lwip2/include/lwipopts.h:3635,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2/tools/sdk/lwip2/include/lwip/opt.h:51,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2/tools/sdk/lwip2/include/lwip/init.h:40,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/IPAddress.h:27,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src/ESP8266WiFi.h:31,
                 from C:\Users\ASUSCenter\Documents\Arduino\test_temp_sens_esp\test_temp_sens_esp.ino:8:
c:\users\asuscenter\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\time.h:81:12: error: conflicting declaration of C function 'tm* localtime_r(const time_t*, tm*)'
   81 | struct tm *localtime_r  (const time_t *__restrict,
      |            ^~~~~~~~~~~
In file included from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/HardwareSerial.h:31,
                 from C:\Users\ASUSCenter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/Arduino.h:288,
[... cut ...]
Using library ESPAsyncTCP-master at version 1.2.2 in folder: C:\Users\ASUSCenter\Documents\Arduino\libraries\ESPAsyncTCP-master 
Using library ESPAsyncWebServer-master at version 1.2.3 in folder: C:\Users\ASUSCenter\Documents\Arduino\libraries\ESPAsyncWebServer-master 
Using library Adafruit_Sensor-master at version 1.1.4 in folder: C:\Users\ASUSCenter\Documents\Arduino\libraries\Adafruit_Sensor-master 
Using library DHT_sensor_library at version 1.4.2 in folder: C:\Users\ASUSCenter\Documents\Arduino\libraries\DHT_sensor_library 
exit status 1
Error compiling for board NodeMCU 0.9 (ESP-12 Module).

earlephilhower, 2021-08-17:
You haven't followed the requested template or posted a MCVE, closing.

That said, you need to delete the bad/obsolete/??? `time.h` in `c:\users\asuscenter\documents\arduino\libraries\include\`
````
c:\users\asuscenter\documents\arduino\libraries\include\time.h:116:22: error: conflicting declaration 'typedef uint32_t time_t'
  116 |     typedef uint32_t time_t;
````

Actually, that sould include directory probably should not exist and should probably be dumped as well.

closed 2021-08-17 by earlephilhower

amnah96, 2021-08-19:
> You haven't followed the requested template or posted a MCVE, closing.
>
> That said, you need to delete the bad/obsolete/??? `time.h` in `c:\users\asuscenter\documents\arduino\libraries\include\`
>
> ``` [...]

thank you very much!
I tried to resolve the time.h problem but again I got errors. then, I deleted the IDE and all the libraries and installed It again and my problem was solved.