esp8266/Arduino #8278: a digital potentiometer using mcp4131 and wemos d1 mini dev board

mo802, 2021-08-17:
Hello everyone, 
I'm trying to make a digital potentiometer using mcp4131 and wemos d1 mini dev board in order to use it in studying the IV characteristics of a solar cell remotely. The mcp4131 digipot uses SPI communication. I have used a simple Arduino code for testing. the code works correctly with my Arduino Uno, but when i tried it with wemos d1 mini dev board it doesn't work!

**MCP4131 and wemos wiring :**
[image]

**code used with arduino :** 
```
#include <SPI.h>

byte address = 0x00;
int CS= 10;

void setup()
{
pinMode (CS, OUTPUT);
SPI.begin();
digitalPotWrite(100)
}

void loop()
{

}

int digitalPotWrite(int value)
{
digitalWrite(CS, LOW);
SPI.transfer(address);
SPI.transfer(value);
digitalWrite(CS, HIGH);
}

```
**code used with wemos:**
```
#include <SPI.h>

byte address = 0x00;
int CS= D8;

void setup()
{
pinMode (CS, OUTPUT);
SPI.beginTransaction(SPISettings(5000000, MSBFIRST, SPI_MODE0));
digitalPotWrite(100);
}

void loop()
{

}

int digitalPotWrite(int value)
{
digitalWrite(CS, LOW);
SPI.transfer(address);
SPI.transfer(value);
digitalWrite(CS, HIGH);
SPI.endTransaction();
}
```
that's the error message that appears on the serial monitor 
[image]

earlephilhower, 2021-08-17:
Closing as the template was ignored.  Suggest using https://esp8266.com for general questions like this.

closed 2021-08-17 by earlephilhower

mo802, 2021-08-18:
why ?