esp8266/Arduino #8299: ESP8266 can't get DHCP IP addres from Asus router

cniedzi, 2021-09-03:
Hi,

I have a problem described below. Is there any solution?

#### Platform
- Hardware:				ESP8266-12F
- Core Version:			Any higher then 2.5.2
- Development Env:		Arduino IDE
- Operating System:		Windows 10

### Settings in IDE
- Module:  			Wemos D1 R2 & mini
- Flash Mode:			
- Flash Size: 		         4MB no SPIFFS
- lwip Variant:                   v2 Lower Memory
- Reset Method:		
- Flash Frequency:	
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:             912600

### Problem Description

I use Wemos D1 mini R2 with Arduino core for ESP8266. With any core version higher than 2.5.2 ESP can't get DHCP IP address from ASUS RT-AX86U router (although according to debug data it succesfully connects to wifi).  

When I revert back to core 2.5.2 or 2.5.0 ESP sucessfully connects to Asus router and gets IP address. Is it known issue?

Regards

cniedzi, 2021-09-11:
Any comment pls?

TD-er, 2021-09-25:
Not really a helpful reply from me, but I am quite curious to see an answer, as it is reported often by users of my software too.

cniedzi, 2021-09-25:
Any core higher then 2.5.2 is useful with asus routers :( I stay with 2.5.

TD-er, 2021-09-25:
Well it seems also to depend on the firmware of those Asus routers, as reported by some. (I don't have Asus routers here)

cniedzi, 2021-09-25:
> Well it seems also to depend on the firmware of those Asus routers, as reported by some. (I don't have Asus routers here)

Thx. I'll pay attention when new asus' fws will be releasing.

TD-er, 2021-09-25:
It was more like downgrade made it work (again), but I have no idea what specific router it was.

d-a-v, 2021-09-25:
@cniedzi 
> Any core higher then 2.5.2 is useless with asus routers :( I stay with 2.5.

With latest release, can you try, in he tools menu:
- 1. to select generic esp8266 board
- 2. then select firmware version "nonos-sdk 2.2.1 (legacy)"
- 3. flash and report

Did you also try to flash OpenWrt on your asus router ?

cniedzi, 2021-09-25:
> @cniedzi
> 
> > Any core higher then 2.5.2 is useless with asus routers :( I stay with 2.5.
> 
> With latest release, can you try, in the tools menu:
> 
> * 1. to select generic esp8266 board
> * 1. then select firmware version "nonos-sdk 2.2.1 (legacy)"
> * 1. flash and report
> 
> Did you also try to flash OpenWrt on your asus router ?

Hi, thx for reply. 
nonos-sdk 2.2.1 (legacy) doesn't work for me with the latest release.

I didn't try OpenWrt because I use Mesh system.

Regards

rolupusoru, 2021-11-05:
Hi @cniedzi ,

I have an ASUS Rt-AC86U running Merlin firmware. My IoT device runs tasmota wit Arduino Core 2.7.4.9 and I have the same problem with DHCP addresses. I t
[sp111_trace_NOK.zip](https://github.com/esp8266/Arduino/files/7489206/sp111_trace_NOK.zip)
ook a trace on the router and I can clearly see that the device ignores the DHCP offer r from the router,

d-a-v, 2021-11-05:
#8319 *may* fix this issue (follow the links for explanations).
It is not merged yet into master branch but will be soon.
In the meantime, any report about it is welcome (by merging the PR or by using the [alpha version 0.0.2 board manager installer for the Arduino IDE](https://d-a-v.github.io/esp8266/Arduino/index.html)).

movodos, 2021-11-12:
Having the same issue with tasmota zbridge connecting to a ASUS RT-AX88U running wrt-merlin.  The bridge required several manual power cycles after a reboot before it would reconnect to wifi. Following the guide below and another forum post, once "Wi-Fi Agile Multiband" was disabled on the 2.4 band the issue completely dissapeared.   

https://www.asus.com/us/support/FAQ/1042475
 
Setting the IP on the device will probably be a better short term solution but maybe this info will help someone else until the fix above is implemented.

barbudor, 2021-11-12:
@movodos we have a test version of Tasmota available with the fix
Please join the discussion and test at https://github.com/arendst/Tasmota/discussions/13623

rolupusoru, 2021-11-14:
Tasmota 10.0.0.2 test version with #8319 did not help. I still have the same DHCP issues on my ASUS router.  Trace attached.
[esp8266_trace.zip](https://github.com/esp8266/Arduino/files/7532973/esp8266_trace.zip)
.

cniedzi, 2021-12-27:
Hi, any news about this issue? Core 3.0.2 still doesn't work for me :( 
PS. Sometimes I am able to sucessfully connect to Asus router, most often just after uploading new sketch.

d-a-v, 2021-12-27:
> Hi, any news about this issue? Core 3.0.2 still doesn't work for me :(

[There are suggestions](https://github.com/esp8266/Arduino/issues/8299#issuecomment-962268021) after your last message that we'd like to have reports about.

cniedzi, 2021-12-28:
> > Hi, any news about this issue? Core 3.0.2 still doesn't work for me :(
> 
> [There are suggestions](https://github.com/esp8266/Arduino/issues/8299#issuecomment-962268021) after your last message that we'd like to have reports about.

I've tried Alpha board manager and to install 0.0.2 core but there were errors during uploading sketch (for normal releases everyting goes fine). Should I set something in addition?

```
Arduino: 1.8.19 (Windows 10), Board: "LOLIN(WEMOS) D1 mini Pro, Sketch, Off, Off, 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 16MB (FS:14MB OTA:~1019KB), v2 Lower Memory, Disabled, HTTP_UPDATE, Only Sketch, 921600"

Executable segment sizes:

ICACHE : 32768           - flash instruction cache 

IROM   : 298964          - code in flash         (default or ICACHE_FLASH_ATTR) 

IRAM   : 31497   / 32768 - code in IRAM          (IRAM_ATTR, ISRs...) 

DATA   : 1708  )         - initialized variables (global, static) in RAM/HEAP 

RODATA : 3712  ) / 81920 - constants             (global, static) in RAM/HEAP 

BSS    : 27168 )         - zeroed variables      (global, static) in RAM/HEAP 

Build-Dir: C:\Users\Czarek\AppData\Local\Temp\arduino_build_511358

C:\Users\Czarek\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1/python3 -I C:\Users\Czarek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\0.0.2/tools/upload.py -fi 0 -up 1 --port COM3 --baud 921600 --before default_reset --after hard_reset --sk 0x0 C:\Users\Czarek\AppData\Local\Temp\arduino_build_511358/Sterownik_WiFi.ino.bin --fs 0x200000 C:\Users\Czarek\AppData\Local\Temp\arduino_build_511358/Sterownik_WiFi.ino 

Port: 'COM3', does not exist!

...Upload aborted

An error occurred while uploading the sketch
```

cniedzi, 2021-12-28:
The core 3.0.2-42-gb5f3d1d6 (0.0.1) uploads correctly and I have to say it connects to my Asus router every time I try. I will test it more.

cniedzi, 2021-12-28:
The core 3.0.2-42-gb5f3d1d6 (0.0.1) uploads correctly.

cniedzi, 2021-12-30:
Hi, any suggestions how to install 0.0.2 core?

d-a-v, 2021-12-31:
0.0.1 and 0.0.2 installation process are similar, unofficial and experimental.
If 0.0.1 installs fine, then 0.0.2 can be installed with the same process.
In doubt or in case of a previous installation, you can quit the IDE, remove `arduino15/staging/packages/esp8266-0.0*` and `arduino15/packaging/esp8266/hardware/esp8266/0.0*` and retry.

cniedzi, 2021-12-31:
> 

I've done what you suggested - 0.0.1 uploads fine while 0.0.2 with error: "Port: 'COM3', does not exist!".
So, I have to wait for official release...

cniedzi, 2021-12-31:
> 0.0.1 and 0.0.2 installation process are similar, unofficial and experimental. If 0.0.1 installs fine, then 0.0.2 can be installed with the same process. In doubt or in case of a previous [...]

I've done what you suggested - 0.0.1 uploads fine while 0.0.2 with error: "Port: 'COM3', does not exist!".
So, I have to wait for official release...

d-a-v, 2022-01-04:
COM port management is managed by the Arduino IDE.
To my knowledge this esp8266 Arduino core has no impact on the serial port management and I'm afraid I can't see any reason why 0.0.1 and 0.0.2 should behave differently regarding your COM3 port.

You can try to manually use `esptool`:
In the Arduino IDE preference, enable verbose upload, then retry to flash, then copy-pase the esptool command line from the Arduino IDE log window onto a command-line shell, then try to play with the arguments.
In both version 0.0.1 and 0.0.2, esptool is the same.

cniedzi, 2022-01-05:
> COM port management is managed by the Arduino IDE. To my knowledge this esp8266 Arduino core has no impact on the serial port management and I'm afraid I can't see any reason why 0.0.1 and 0.0.2 [...]

2.5.0 uses esptool - DOES upload
`C:\Users\Czarek\AppData\Local\Arduino15\packages\esp8266\tools\esptool\2.5.0-3-20ed2b9/esptool.exe -vv -cd nodemcu -cb 921600 -cp COM3 -ca 0x00000 -cf C:\Users\Czarek\AppData\Local\Temp\arduino_build_151226/Sterownik_WiFi.ino.bin `

0.0.1 uses upload.py instead of esptool - DOES upload
`C:\Users\Czarek\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1/python3 -I C:\Users\Czarek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\0.0.1/tools/upload.py --chip esp8266 --port COM3 --baud 921600 --before default_reset --after hard_reset write_flash 0x0 C:\Users\Czarek\AppData\Local\Temp\arduino_build_391240/Sterownik_WiFi.ino.bin `

0.0.2 uses upload.py - DOES NOT upload
`C:\Users\Czarek\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1/python3 -I C:\Users\Czarek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\0.0.2/tools/upload.py -fi 0 -up 1 --port COM3 --baud 921600 --before default_reset --after hard_reset --sk 0x0 C:\Users\Czarek\AppData\Local\Temp\arduino_build_151226/Sterownik_WiFi.ino.bin --fs 0x200000 C:\Users\Czarek\AppData\Local\Temp\arduino_build_151226/Sterownik_WiFi.ino `

With 0.0.2 no matter which port I set, "Port: 'COM3', does not exist!". 
I've sucessfully used esptool for 0.0.2 and the main issue still exists - ESP8266 still can't connect to Asus router.

cniedzi, 2022-01-06:
I can say 0.0.1 is more or less the same as 3.0.2 - ESP8266 connects to router after ca. 3-5 min. while 2.5.0 instantly.
In both cases WiFi connection is set instantly, but there is a problem with DHCP negotiation for 2.5.0+.

d-a-v, 2022-01-06:
You can now remove 0.0.1 and reinstall it, or [try the "git version"](https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version),

DHCP updates have just been integrated in master branch.

cniedzi, 2022-01-06:
> You can now remove 0.0.1 and reinstall it, or [try the "git version"](https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version),
> 
> DHCP updates have just been integrated in master branch.

I have installed 3.1.0-dev and it looks promising. I can connect to router similarly to 2.5.0.

sblantipodi, 2022-02-07:
> > You can now remove 0.0.1 and reinstall it, or [try the "git version"](https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version),
> > 
> > DHCP updates have just been integrated in master branch.
> 
> I have installed 3.1.0-dev and it looks promising. I can connect to router similarly to 2.5.0.
> 
> EDIT: Looks like I've stated premature opinion :( Sometimes I can connect immediately, somtimes I can't connect at all or I have to wait long time... Still testing.

It looks promising, but does it worked?
Does 3.1.0-dev solved the problem?

Same problem here with AX56U

sblantipodi, 2022-02-13:
it seems that the latest version on the master branch completely fixed the problem on my Asus AX56U.
I had a the same problem even using my provider's router (Vodafone Station) and it is fixed on that router too.

Congrats guys... is there an ETA for this patch to be released in the next release?

zpin, 2022-08-17:
Same problem here, the workaround in issue #8412 (using `WiFi.setPhyMode(WIFI_PHY_MODE_11G);`) worked for me.

PedanticPete, 2022-09-17:
Thanks, all of my espI266s have been working fine with my new ASU's xt8 until I had to update one of them ...  this 11G solution works just fine! Thanks again!

> Same problem here, the workaround in issue #8412 (using `WiFi.setPhyMode(WIFI_PHY_MODE_11G);`) worked for me.

jasenpashov, 2023-02-06:
Thank you very much. That fixed everything - WiFi.setPhyMode(WIFI_PHY_MODE_11G);

dustinsievers, 2023-05-15:
> Same problem here, the workaround in issue #8412 (using `WiFi.setPhyMode(WIFI_PHY_MODE_11G);`) worked for me.

This solved my issues

LabriePierre, 2023-11-01:
Had the same problem with a Giga Hub from Bell (Sagecom Fast 5689E). Used ESP8266 boards (3.02) in Arduino. The ESP01 was connecting after many minutes but did not receive an IP address.  Using WiFi.setPhyMode(WIFI_PHY_MODE_11G); worked for me also. The connection is in 5 seconds now.

flywhc, 2024-06-19:
I had the same problem with x86 openwrt router. I had to use static IP to connect to router. Tried WiFi.setPhyMode and other ways but no luck.

I downloaded the source code of this package, removed installed package and tried to debug the issue. Then it got IP in 5 seconds without any code changes!
I am sure I tested many different distributed version of packages.

ZakKemble, 2024-06-20:
Hey, I was having this same issue with an ASUS RT-AX86U router (Merlin firmware). Using `WiFi.setPhyMode(WIFI_PHY_MODE_11G);` or disabling 11n on the router side (legacy mode) solves it.

I used Wireshark on my PC in my attempts at tracking down the problem and I could see the DHCP discovery broadcast packets coming from the ESP, but the DHCP server on the router simply did not respond to them unless the ESP8266 phy had been set to 11g. The discover packets that wireshark picked up were identical on both 11g and 11n networks (apart from the transaction ID).

I also noticed that if I connected my phone to the 2.4G network it would kick start things into working again, and the ESP would be able to get an IP no problem when using 11n!

Same thing happens with v3.1.1 and v3.1.2.

**Also be careful when testing with and without `WiFi.setPhyMode()` - the ESP must be power-cycled after removing the call otherwise it will remember the previous setting between resets.**

madmacks59, 2024-07-13:
I would like to add to the issue. I've been struggling for 3-4 days trying to get my Hiletgo ESP8266 NODEMCU devices working reliably and found this open ticket. I reverted to the 2.5.2 core, removed wifi scanning from my sketch (apparently nor supported in the old core), and bingo, the board connects reliably to my ASUS RT-AX86U Pro router.  THIS ISSUE SHOULD GET A HIGHER PRIORITY as we're currently at 3.1.2 core and this problem should be fixed!

ericswpark, 2024-07-13:
Same issue here. Here's a link to the repo that houses the sketch I'm using: https://github.com/ericswpark/nodemcu_plant_watering

I had to use the 802.11g downgrade workaround to get the board to reliably connect to my RT-AX56U:
https://github.com/ericswpark/nodemcu_plant_watering/blob/ac86b6fb836b187824d3cb05c1137a7017b5cca8/nodemcu_plant_watering.ino#L122

It connected fine on the first connection attempt, but subsequent connections failed wrt DHCP.

reepotah, 2025-05-25:
Am a beginner, today I was trying to get a connection from my ESP and was confused why isn't it working until I found this issue.

I have ASUS RT-AX58U.

Any one of setting ESP to 11G or following [this guide](https://www.asus.com/us/support/faq/1042475/) worked.