esp8266/Arduino #8300: fatal error: WiFiClientSecureAxTLS.h: No such file or directory after upgrade to version 3

svdrummer, 2021-09-03:
fatal error: WiFiClientSecureAxTLS.h: No such file or directory

After upgrade to version 3, I get the above error.
It seems WiFiClientSecureAxTLS.h was not included with version 3 and yet ESP8266WiFi.h is still calling it?

It seems uploading the missing file does not fix the issue.

earlephilhower, 2021-09-03:
AxTLS was removed from 3.0 after about 1.5yrs of deprecation.  Please see the examples for info on using the new standard BearSSL.

You may also have an older version of the ESP8266WiFi library in your private libs directory because there are no current refs to the deleted file in 3.x.  Delete them and potentially re-install 3.0 from library manager and you should be good.

For add'l assistance, please use https://esp8266.com or https://gitter.im/esp8266/Arduino because this isn't a code issue.

closed 2021-09-03 by earlephilhower

svdrummer, 2021-09-03:
Thank you. I have the latest of all libs. The library site just said to revert to version 2.7.4.
I must be other libs such as stream, which is calling a function in the wifi lib.