esp8266/Arduino #8306: AnalogWriteFreq function not working properly after Arduino 1.8.15 update

santhiaggod, 2021-09-08:
#### Platform

- Hardware:				ESP8266
- Core Version:			
- Development Env:		Arduino IDE
- Operating System:		Windows 10

### Settings in IDE

- Module:  			Node MCU 1.0
- Flash Size: 		        4MB
- CPU Frequency:		160MHz
- Upload Using:		SERIAL
- Upload Speed:              115200

### Problem Description

Hello all, I have been trying to use a simple infrarred LED at 38 kHz using the 12th GPIO on the NodeMCU dev kit writing a PWM signal with ~60% duty cycle at 38kHz. The built-in analogWriteFreq function did just fine using the Arduino IDE v. 1.8.14 but stopped working after updating to the 1.8.15 version. At this point the output at said pin is simply a HIGH level. 

```cpp

void setup() {
  pinMode(12, OUTPUT); 
}

void loop() {
  analogWriteFreq(38000); 
  analogWrite(12, 600);
}

```

dok-net, 2021-09-08:
Off the top of my head, while earlier releases of the ESP8266 core used a default analog scale of 1024, for which your 600 would result in the approx. 60% duty cycle, later it was fixed to adhere to generic Arduino core. Take into consideration this comment in `cores/esp8266/core_esp8266_wiring_pwm.cpp`:
```
// Per the Arduino docs at https://www.arduino.cc/reference/en/language/functions/analog-io/analogwrite/
// val: the duty cycle: between 0 (always off) and 255 (always on).
// So if val = 0 we have digitalWrite(LOW), if we have val==range we have digitalWrite(HIGH)
```

earlephilhower, 2021-09-09:
As @dok-net said, this was part of the 3.x release notes and changes to match standard Arduino behavior.

You can add an `analogWriteRange()` call to your `setup()` function to set the limits back to 2.7.x values.

closed 2021-09-09 by earlephilhower