esp8266/Arduino #8307: DHCPS included in the Core and the old SDK implementation

mcspr, 2021-09-09:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware: any
- Core Version: 3.0.x

### Problem Description

Something noticed while building lwip2 [here](https://github.com/d-a-v/esp82xx-nonos-linklayer/pull/46)

Since #6680 (https://github.com/esp8266/Arduino/commit/51c2a1437b720d584d2f8dd7e01efeb608f991a4#diff-39ab7a280ef46486a19b45c179a5a160a836d57ab164f627ade75cb92ee94d54), old lwip-bundled dhcps was extracted and rewritten into a class

Current example for SoftAP replaced the original calls with the Arduino-API-like global object:
https://github.com/esp8266/Arduino/blob/211606fe9cc781d576a1ae074607e172050b7d64/libraries/ESP8266WiFi/examples/StaticLease/StaticLease.ino#L74-L80
And the internal WiFi code is also forced to work through the wrapper:
https://github.com/esp8266/Arduino/blob/211606fe9cc781d576a1ae074607e172050b7d64/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp#L247

While the original functions are still in the SDK header. Since the lwip1 code is removed, this is just a linking error for the existing programs pre 3.0.0 that use any of `wifi_softap_..._dhcps_...` funcs
https://github.com/esp8266/Arduino/blob/211606fe9cc781d576a1ae074607e172050b7d64/tools/sdk/include/user_interface.h#L385-L395

I'd propose *keeping* the SDK API, but still have a custom class for an netif when actually warranted (maybe in cases like Ethernet?). This will also remove the need for the WiFi class code to reference the SoftAP DHCPS through an Arduino sketch-like API (...which also means, there will be no need to have headers with carefully placed externs)

cc @d-a-v

also, there is slightly more up-to-date rtos sdk implementation (which also has english comments)
https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/components/lwip/apps/dhcpserver/dhcpserver.c

but, it is interesting that the esp32 variant of idf does not include any alternative to add/remove reserved mac feature
https://github.com/espressif/esp-idf/blob/master/components/lwip/include/apps/dhcpserver/dhcpserver.h

d-a-v, 2021-09-13:
#6680 was merged before lwIP-v1 was removed.
So I think these functions from `user_interface.h` could be safely hidden or removed.
A prerelease of lwIP-2.1.3 is out, changes can happen at that time

mcspr, 2021-09-17:
> So I think these functions from user_interface.h could be safely hidden or removed.

I actually meant to restore them + slightly adjust the additional code introduced by the lib. Or it's a no-go?
Roughly this -> https://github.com/esp8266/Arduino/compare/master...mcspr:issue8307?expand=1

mcspr, 2022-02-20:
Another (possibly minor) note about using objects. We are in C++ environment, and we also may run into an issue of init order when dealing with state inside of certain objects declared across the Core

https://gcc.gnu.org/onlinedocs/gcc-10.3.0/gcc/C_002b_002b-Attributes.html#index-init_005fpriority-variable-attribute
While Ctor & Dtor order is stable inside of a single translation unit (aka .cpp -> .o), we can't guarantee order between the Core libraries, Core startup and the sketch itself and it depends purely on the order object files are processed

PR #8546 (2022-06-01): Initialize SoftAP DhcpServer object on demand
Remove dependency on global ctor, and just construct the object when
someone asks us to do it. Only dependency right now is netif_git, which
is expected to be initialized by the lwip code some time before
dhcps_start happens.

Removing ip_info from begin(), since we never reference later on.
Also removing the specific check for netif id and simplify the ctors.
Fixing _netif->state check left over from the old code.

Update tests and recover old nonos-sdk dhcps functions that were not implemented.

resolve #7871
resolve #8307
resolve #8471
closed 2022-06-01 by d-a-v