esp8266/Arduino #8326: Unable to connect to ESP web server in softAP mode if there is no delay between WiFi.softAP() and WiFi.softAPConfig()

enjoyneering, 2021-09-29:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[3.0.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[80Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Client connects to softAP no problem, but web browser cannot connect to ESP web server in softAP mode if there is no delay between WiFi.softAP() and WiFi.softAPConfig(). The WiFi.softAPConfig() detects no problem and returns true.

```
  if (WiFi.softAP(ssid, password, channel, hidden, maxUsers) != true) //ssid, password, hidden=false, maxUsers=4
  {
    Serial.println(F("faild to start soft AP"));

    return false;
  }

  delay(100);  //web browser cannot connect to ESP web server in softAP mode if there is no delay here !!!

  if (WiFi.softAPConfig(ip, gateway, subnet) != true) //ip=192.168.4.1, gateway = 0.0.0.0, subnet=255.255.255.0
  {
    Serial.println(F("failed to assign soft AP settings"));

    return false;
  }
```

jandrassy, 2021-09-29:
first configure, then start

jandrassy, 2021-09-29:
sleepMode and phyMode are properties which apply on a layer common for STA and SoftAP.
changing phy mode requires reconnection so it makes sense doing it before joining AP or beginning SoftAP.
I guess you can change sleep mode any time
https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/generic-class.html#wifi-power-management-dtim

closed 2021-09-29 by enjoyneering