esp8266/Arduino #8328: Upload on Linux Mint 20.2 - kernel 5.4.0-88

roboticboyer, 2021-10-02:
After the upgrade of Linux Mint 20.2 to kernel 5.4.0-88, Esptool,py is not yet able to upload the sketches on the ESP8266
I've tested this bug on using 2 different PCs
If I'll use a previous version of  the kernel all is running

This is error:
```
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "/home/giorgio/Applicazioni_gio/arduino_ESP8266/portable/packages/esp8266/hardware/esp8266/3.0.2/tools/upload.py", line 66, in <module>
    esptool.main(cmdline)
  File "/home/giorgio/Applicazioni_gio/arduino_ESP8266/portable/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool/esptool.py", line 3552, in main
    esp.connect(args.before, args.connect_attempts)
  File "/home/giorgio/Applicazioni_gio/arduino_ESP8266/portable/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool/esptool.py", line 529, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
```

roboticboyer, 2021-10-02:
There is a work around

Settings prior to uploading the sketch:

Boardtype: Generic ESP8266
Flash size: 4MB
Reset method: " no dtr (aka ck)"

Then press Flash button (or Ground D3 (GPIO0) on Wemos D1 mini)
Press Reset shortly
Then release Flash

Press upload in the IDE

earlephilhower, 2021-10-02:
There was a change to the USB serial driver in a recent kernel revision that borked the RS232 control line support used to reset the ESP boards.  Looks like you hit it.  There's nothing here or in esptool to do, you'll have to up or down rev the kernel.  There was thread here and at the esptool repo with more info.  Good luck!

closed 2021-10-02 by earlephilhower

florianfa, 2021-10-20:
I'm facing the same issue with Linux Mint 20.2 and kernel 5.4.0-88. Reboot Linux Mint with an older kernel helps as workaround . E.g. 5.4.0-81-generic