esp8266/Arduino #8344: Arduino 2.0.0-beta.12 No hostname shown in OTA uppdate, only IP address...

tomtdl, 2021-10-17:
[image]

Arduino 1.8.13 will list IP address AND hostname of all online  ESP8266's
Arduino 2.0.0-beta.12 ONLY lists IP addreses

This is difficult as I have many OTA-updateable ESP's online and most are DHCP addressed.

Anyone else notice this different behavior?  Believe me, I googled the issue, checked the forums
and found no answer.  This is in Linux Mint 20 on a new installation of Arduino 2.0

I am new to posting here.  Please forgive me if this is the wrong way to ask or post an issue.

Tom Leitschuh

(screenshot attached as png)

vincentVDB, 2021-10-20:
I haven-t Port item under Board in Tools menu !
i'm in beta 12

mcspr, 2022-09-09:
Fixed in https://github.com/arduino/arduino-ide/pull/1335

closed 2022-09-09 by mcspr