esp8266/Arduino #8365: ets Jan  8 2013,rst cause:4, boot mode:(3,6)  wdt reset load 0x4010f000, len 3460, room 16  tail 4 chksum 0xcc load 0x3fff20b8, len 40, room 4  tail 4 chksum 0xc9 csum 0xc9 v0006bc70 ~ld

Phoenixpranav, 2021-11-07:
ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v0006bc70
~ld
HELP US WITH THIS ERROR

hi I am working on a fitness watch that has 2 i2c devices 1)a oled display 2)MAX 30100 heartrate sensor. the program gets complied and uploaded without any errors but facing issue during runtime. Output is none of the sensors are working. If a seperate code is used for only heartrate monitoring ,the sensors are working.when combined with the below code we r facing this issue 
 
The following is the code running on esp12e D1 mini .

#define BLYNK_TEMPLATE_ID "TMPLxYVkj3uJ"
#define BLYNK_DEVICE_NAME "esp"
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#include <EasyButton.h>
#include "MAX30100_PulseOximeter.h"
#include "MAX30100.h"
#define BLYNK_PRINT Serial//redefine  BLYNK_PRINT as Serial
#include <ESP8266WiFi.h>//header file for controlling wifi
#include <BlynkSimpleEsp8266.h>//header for blynk
#include <TimeLib.h>//header file for counter
#include <WidgetRTC.h>// header file for Real Time Clock
#include "Ubidots.h"
long int update_timer=millis();
long int button_timer=millis();
long int on_screen=millis();
long int on_screen_heart=millis();
long int random_time=millis();
#define BUTTON_PIN 7
#define REPORTING_PERIOD_MS      1000
#define PULSE_WIDTH              MAX30100_SPC_PW_1600US_16BITS
#define IR_LED_CURRENT           MAX30100_LED_CURR_50MA  
#define LED_CURRENT              MAX30100_LED_CURR_20_8MA 
#define SAMPLING_RATE            MAX30100_SAMPRATE_1000HZ
PulseOximeter pox;
MAX30100 sensor;
#define OLED_RESET 4
Adafruit_SSD1306 display(-1);
uint32_t tsLastReport = 0;
uint32_t last_beat=0;
bool initialized=false;
int HRclean;
int SpO2;
char context[50];
char lattitude[20];
char longitude[20];
int ThermistorPin =A0;
int Vo;
float R1 = 4500;
float logR2, R2, T, Tc, Tf;
float c1 = 1.009249522e-03, c2 = 2.378405444e-04, c3 = 2.019202697e-07;
int menu=1;
int beat=0;
int oxygen=0;
int heart=0;
int heart_screen=0;
int oxygen_screen=0;

/*=================================================================================================================*/

char auth[] = "TMPLxYVkj3uJ";// indentification of your board like phone number
[... 466 lines cut ...]
  }
  else if(menu==4)
  {
    temperature(1);
    delay(150);
  }
  else if(menu==5)
  {
    gps();
    delay(150);
  }
  else if(menu>5)
  {
    menu=1;
  }
}
else
{
  Serial.println("sleep"); 
  menu=0;
  display.clearDisplay();
  display.display();
}
  Blynk.run();//communicate to mobile
}

earlephilhower, 2021-11-07:
You've ignored and deleted the template.  This is not the appropriate place and there is no MCVE.  Closing.  Use gitter or esp8266.com for help on things like this, as the README.md and template says.

closed 2021-11-07 by earlephilhower

aldolo69, 2022-12-07:
have you found something? the same error happens here with the unmodified blynk sample code