esp8266/Arduino #8366: StreamString.h:89:29: error: 'min' is not a member of 'std'

adams13x13, 2021-11-09:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12
- Core Version:			3.0.2
- Development Env:		Arduino IDE 1.8.15
- Operating System:		Ubuntu (Linux Mint)

### Settings in IDE

- Module:  			"Nodemcu 1.0 (ESP-12 Module)"
- Flash Mode:			?? [qio|dio|other]
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		?? [ck|nodemcu]
- Flash Frequency:	?? [40Mhz]
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200 (serial upload only)

### Problem Description

When StreamString.h is included from the *.cpp file and \<algorithm\> is not included before, following error message is given by the compiler:

> error: 'min' is not a member of 'std'

```
In file included from /home/testuser/git/StreamStringBugExample/Test1.cpp:4:
/home/testuser/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/StreamString.h: In member function 'virtual int S2Stream::read(uint8_t*, size_t)':
/home/testuser/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/StreamString.h:89:29: error: 'min' is not a member of 'std'
   89 |             size_t l = std::min(len, (size_t)string->length());
      |                             ^~~
```
The problem does not appear if StreamString.h is included from the *.ino file.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch (two files)

StreamStringBugExample.ino
```cpp

void do_test();

void setup() {
  do_test();
}

void loop() {
}
```
Test1.cpp
```cpp
// #include <WString.h> // not needed as it is included from StreamString.h
#include <Stream.h> // this include is also needed for the StreamString.h
// #include <algorithm> // with this not commented out, the code would compile
#include <StreamString.h>

void do_test()
{
  StreamString s;
  s.println("This is a test");
}
```

### Possible solution

At the beginning of StreamString.h, one should include the headers that are used by this file:
```cpp
#include "Stream.h"
#include <algorithm>
```

earlephilhower, 2021-11-09:
@adams13x13 you've found a bug and have a good fix.  Would you like to make a PR implementing it, so you get credit in the `git` logs for fixing it?

adams13x13, 2021-11-10:
Like this? #8367

earlephilhower, 2021-11-10:
Yes, great.  Thx!  Closing per #8367

closed 2021-11-10 by earlephilhower