esp8266/Arduino #8368: 'class BearSSL::WiFiClientSecure' has no member named 'setCACert_P'

ozzcet, 2021-11-11:
### Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				nodemcuv2
- Core Version:			[latest git hash or date]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

I have used to Arduino IDE until found PlatformIO. When I lift my code to PlatformIO almost everything runs super. But ın my case, Although the code runs super at Arduino IDE ,it is not run at PlatformIO, so wierd. I have tried to find why in *.h files but notthing... 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <WiFiClientSecure.h>

void setup() {
function();
}

void loop() {

}

void function(){
// Load root certificate in DER format into WiFiClientSecure object
  bool res = client.setCACert_P(caCert, caCertLen);
  if (!res) {
    Serial.println("Failed to load root CA certificate!");
    while (true) {
      yield();
    }
  }
}
```

### Debug Messages

```
'class BearSSL::WiFiClientSecure' has no member named 'setCACert_P'
   42 |   bool res = client.setCACert (caCert, caCertLen);
```

mcspr, 2021-11-11:
Must be Core version difference between IDE and PIO installations. PlatformIO would use the latest Core 3.0.2 that removes both `setCACert` and `setCACert_P` (ref. 85ba53a24994db5ec2aff3b7adfa05330a637413). See the BearSSL documentation related to cert API
https://github.com/esp8266/Arduino/blob/da6ec83b5fdbd5b02f04cf143dcf8e158a8cfd36/doc/esp8266wifi/bearssl-client-secure-class.rst#x509-certificates

x509list class works seamlessly with both with der and pem formats as input. pass certs object to the client, and you should be good to go
https://github.com/esp8266/Arduino/blob/69f8cd693447404be8696527e94f3665d6318b8e/libraries/ESP8266WiFi/examples/HTTPSRequest/HTTPSRequest.ino#L26
https://github.com/esp8266/Arduino/blob/69f8cd693447404be8696527e94f3665d6318b8e/libraries/ESP8266WiFi/examples/HTTPSRequest/HTTPSRequest.ino#L66

*([note-to-self, unrelated to the issue: keywords.txt still contain these old methods](https://github.com/esp8266/Arduino/blob/0761f6f672dd0cf18984dec666886c3f429f25e1/libraries/ESP8266WiFi/keywords.txt#L148-L161))*

closed 2021-11-11 by mcspr

ozzcet, 2021-11-11:
> Must be Core version difference between IDE and PIO installations. PlatformIO would use the latest Core 3.0.2 that removes both `setCACert` and `setCACert_P` (ref. [85ba53a](https://github.com/esp8266/Arduino/commit/85ba53a24994db5ec2aff3b7adfa05330a637413)). See the BearSSL documentation related to cert API https://github.com/esp8266/Arduino/blob/da6ec83b5fdbd5b02f04cf143dcf8e158a8cfd36/doc/esp8266wifi/bearssl-client-secure-class.rst#x509-certificates
> 
> x509list class works seamlessly with both with der and pem formats as input. pass certs object to the client, and you should be good to go
> 
> https://github.com/esp8266/Arduino/blob/69f8cd693447404be8696527e94f3665d6318b8e/libraries/ESP8266WiFi/examples/HTTPSRequest/HTTPSRequest.ino#L26
> 
> 
> https://github.com/esp8266/Arduino/blob/69f8cd693447404be8696527e94f3665d6318b8e/libraries/ESP8266WiFi/examples/HTTPSRequest/HTTPSRequest.ino#L66
> 
> _([note-to-self, unrelated to the issue: keywords.txt still contain these old methods](https://github.com/esp8266/Arduino/blob/0761f6f672dd0cf18984dec666886c3f429f25e1/libraries/ESP8266WiFi/keywords.txt#L148-L161))_

Thanks @mcspr , however, please get me as newbie... I used to old fashioned way, click top of the lock icon to extract ssl of web page and ı have turn it code by getting some info form  [https://youtu.be/Wm1xKj4bKsY?t=781](url). Actually I have done how andreas Spiess explain... But I realy do not know how could I extract certs.h for any web page like below example 
[https://github.com/esp8266/Arduino/blob/69f8cd693447404be8696527e94f3665d6318b8e/libraries/ESP8266WiFi/examples/HTTPSRequest/certs.h](url)

Best regards.

nrnw, 2024-07-26:
The BearSSL::WiFiClientSecure class does not have the setCACert, setCertificate, or setPrivateKey methods. Instead, you need to use BearSSL::WiFiClientSecure with BearSSL::X509List. Check the example below.

```
//Initialize the WiFi 
  BearSSL::WiFiClientSecure wifiClient;

//Set certificates
  BearSSL::X509List cert(client_cert);
  BearSSL::X509List caCert(ca_cert);
  BearSSL::PrivateKey key(client_key);
```