esp8266/Arduino #8372: SPIFFS println weird behavior with long strings

fabianoriccardi, 2021-11-13:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					Wemos D1 mini
- Core Version:			commit: bf2882d
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Wemos D1 mini
- Flash Mode:			dio
- Flash Size: 		        4MB
- lwip Variant:                  v2 Lower Memory
- Reset Method:		default
- Flash Frequency:	        40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   916200

### Problem Description

Sporadically, File::println("message") fails, and instead of recording "\r\n" it prints 0xFFFF. This happens only with long messages (i.e. >350 chars). It happens with Arduino Core 3.x.x, as well as the latest commit, but it works fine in previous releases.

I know that SPIFFS is deprecated, but this seems a regression and it should be mentioned. 

### Sketch

```cpp
# print 10 lines on file and then print back to Serial.

#include <FS.h>

void setup() {
  Serial.begin(115200);
  while(!Serial);
  Serial.println();
  
  // You mays need to format the flash before using it
  //SPIFFS.format();

  if(SPIFFS.begin()){
    Serial.println("Filesystem mounted successfully");
  }else{
    Serial.println("Filesystem NOT mounted. System halted");
    while(1) delay(100);
  }
}

void loop() {
  for(int i=0; i<10; ++i){
    Serial.println(i);
    File f = SPIFFS.open("/mytext.txt", "a");
    if(f){
      f.println("{\"glossary\":{\"title\":\"example glossary\",\"GlossDiv\":{\"title\":\"S\",\"GlossList\":{\"GlossEntry\":{\"ID\":\"SGML\",\"SortAs\":\"SGML\",\"GlossTerm\":\"Standard Generalized Markup Language\",\"Acronym\":\"SGML\",\"Abbrev\":\"ISO 8879:1986\",\"GlossDef\":{\"para\":\"A meta-markup language, used to create markup languages such as DocBook.\",\"GlossSeeAlso\":[\"GML\",\"XML\"]},\"GlossSee\":\"markup\"}}}}}");      
      f.close();  
    } else {
      Serial.println("error");
    }
    delay(1000);
  }

  File f = SPIFFS.open("/mytext.txt", "r");
  while(f.available()){
    Serial.println(f.readStringUntil('\n'));
  }
  f.close();
  SPIFFS.remove("/mytext.txt");
}

```

### Debug Messages

Sometimes 2 messages are printed on the same line. It seems that println prints 0xFFFF instead of "\r\n". An example (there are 2 '⸮' in the middle of the string):

```
{"glossary":{"title":"example glossary","GlossDiv":{"title":"S","GlossList":{"GlossEntry":{"ID":"SGML","SortAs":"SGML","GlossTerm":"Standard Generalized Markup Language","Acronym":"SGML","Abbrev":"ISO 8879:1986","GlossDef":{"para":"A meta-markup language, used to create markup languages such as DocBook.","GlossSeeAlso":["GML","XML"]},"GlossSee":"markup"}}}}}⸮⸮{"glossary":{"title":"example glossary","GlossDiv":{"title":"S","GlossList":{"GlossEntry":{"ID":"SGML","SortAs":"SGML","GlossTerm":"Standard Generalized Markup Language","Acronym":"SGML","Abbrev":"ISO 8879:1986","GlossDef":{"para":"A meta-markup language, used to create markup languages such as DocBook.","GlossSeeAlso":["GML","XML"]},"GlossSee":"markup"}}}}}
```

If I use `f.print("mymessage\r\n")` the sketch works as expected

mcspr, 2021-11-13:
also see https://github.com/esp8266/Arduino/issues/8070

with the examples above...
f.print("...") 360 byte string, plus f.println() results in 0xff 0xff instead of \r \n
f.print("...") 360 byte string, plus f.print('\r') and .fprint('\n') also has the same result
single print("...\r\n") still works though (but, I wonder if there's also a limit when it no longer works)

for failing cases, 0xff is 0xff can be seen on the flash, so I'd guess this is something broken on writes and not something wrong with the reading
```
00001300: 0100 0800 7c53 4f20 3838 3739 3a31 3938  ....|SO 8879:198
00001310: 3622 2c22 476c 6f73 7344 6566 223a 7b22  6","GlossDef":{"
...
00001390: 476c 6f73 7353 6565 223a 226d 6172 6b75  GlossSee":"marku
000013a0: 7022 7d7d 7d7d 7dff ff7b 2267 6c6f 7373  p"}}}}}..{"gloss
                           ^^ ^^
```
(looking at binary dump via esptool read_flash 0x300000 0x100000 for something resembling file chunks)

There were no changes to spiffs source itself 2.7.4 and 3.0.0 though, and it is deprecated for a variety of reasons (https://github.com/esp8266/Arduino/issues/7095 plus https://github.com/esp8266/Arduino/pull/7263), so it is a pretty low chance this is going to get fixed.

There are some avenues to explore
- gcc version bump from 4.8.2 to 10.3 *might* sometimes be an issue (see https://github.com/esp8266/Arduino/issues/8261), but adding -fno-strict-aliasing to the build does not do anything helpful
- perhaps 79ea883fb3099e86030e69f867648dfabcd270b5 is related, as it did change the flash write behaviour that spiffs *might've* relied upon in the previous Core version(s) (ref. cores/esp8266/spiffs_api.cpp and flash hal we use).
reverting the commit seems to fix the issue

fabianoriccardi, 2021-11-14:
I missed you post, thanks for linking!

I will look at  [#79ea883](https://github.com/esp8266/Arduino/commit/79ea883fb3099e86030e69f867648dfabcd270b5) for further insights.

Bodmer, 2022-02-16:
I seem to be having the same problem.  I have saved data into a SPIFFS file, but after reading ~1kbyte back there are spurious FF's in the data read. 

Read back:
```
E5 D9 5F E1 A8 99 E7 4E FF FF FF B5 29 87 DD 66
                                        ^^^^^ incorrect bytes
```

Correct data:
```
E5 D9 5F E1 A8 99 E7 4E 1B F5 6D B5 29 87 DD 66
                                        ^^^^^ correct data
```

Bodmer, 2022-02-16:
Rolling back from 3.0.x to 2.7.4 fixes the problem so it does appear to be a bug in the latest versions.

fabianoriccardi, 2022-05-22:
I can confirm that the  issue is on "write" and not in "read" process.

In particular, it seems that the bug is in `bool EspClass::flashWrite(uint32_t address, const uint8_t *data, size_t size)`, and in particular in the else-branch at line.

If I understood the idea of the algorithm, to write in flash the proper 4-bytes alignment must be guaranteed for both RAM and flash addresses. The issue is that if you try to align one, you will break the other alignment. In that branch you may have a wrong RAM alignment (there is no check about it), so in this case it may fail the writing.

In other words, when you enter that branch, if the flash address is not aligned, you should write a partial 4-bytes block and then re-check the alignment of the start address of remaining data. This may lead to a reallocation of the data on the stack, (it already occurs in flashWriteUnalignedMemory function).

If the maintainer would consider the idea, I can create a pull request.

mcspr, 2022-05-23:
Can you provide an example with specific numbers and the call chain that breaks?
[BTW, also see RTOS implementation for both u32 that we have and 'generic' void-ptr variant](https://github.com/espressif/ESP8266_RTOS_SDK/blob/d45071563cebe9ca520cbed2537dc840b4d6a1e6/components/spi_flash/src/spi_flash.c)

fabianoriccardi, 2022-05-23:
The minimal sketch to reproduce is the following one with core v3.0.2.

```
#include <FS.h>

void setup() {
  Serial.begin(115200);
  while(!Serial);
  Serial.println();
  
  // You mays need to format the flash before using it
  //SPIFFS.format();

  if(SPIFFS.begin()){
    Serial.println("Filesystem mounted successfully");
  }else{
    Serial.println("Filesystem NOT mounted. System halted");
    while(1) delay(100);
  }
}

void loop() {
  for(int i=0; i<10; ++i){
    Serial.println(i);
    File f = SPIFFS.open("/mytext.txt", "a");
    if(f){
      // Print 200 chars (+2 "\r\n")
      int n = f.println("12345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890");      
      f.close();
      Serial.println(String("I have printed: ") + n );
    } else {
      Serial.println("error");
    }
    delay(500);
  }

  File f = SPIFFS.open("/mytext.txt", "r");
  while(f.available()){
    Serial.println(f.readStringUntil('\n'));
  }
  f.close();
  SPIFFS.remove("/mytext.txt");
}
```

When you print the file content on serial port, you will see some prints showing "??" instead of "\r\n".

fabianoriccardi, 2022-05-29:
I have pushed a temptative fix that fix the current issue, but it may break other behaviors which I'm not aware of...

closed 2022-07-03 by mcspr