esp8266/Arduino #8384: LittleFS missing availableForWrite()

CapnBry, 2021-11-24:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12F]
- Core Version:			[3.0.2]
- Development Env:		[VSMicro]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[LOLIN(WEMOS) D1 D2 & mini]
- Flash Mode:			[DIO]
- Flash Size: 		[4MB]
- lwip Variant:   [Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [921600] (serial upload only)

### Problem Description

Attempting to use HTTPClient to stream directly to a LittleFS `File` times out with error code -11. This is due to the `LittleFSFileImpl` not implementing `availableForWrite()`, so the base implementation returns 0 and the streaming can't write and times out. I added this implementation but I do not know if this is the right thing to return. It probably isn't but hey it works! It would probably work even if I returned 1 so I'm not going to throw myself a parade or anything for getting it working.
```
// in LittleFS.h LittleFSFileImpl
    int availableForWrite() override {
      if (!_opened || !_fd) {
        return 0;
      }
      return _fs->_lfs_cfg.block_size;
    }
```

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
void taskcbIcsCheck(void)
{
  Serial.println(F("Performing ICS check..."));

  WiFiClient *client;
  if (strncmp(ICS_URL, "https://", 8) == 0)
  {
    client = new BearSSL::WiFiClientSecure();
    ((BearSSL::WiFiClientSecure *)client)->setInsecure();
  }
  else
    client = new WiFiClient();

  HTTPClient *https = new HTTPClient();
  int httpCode = -1;
  if (https->begin(*client, ICS_URL))
  {
    https->setUserAgent("Google-Calendar-Importer");
    https->setFollowRedirects(HTTPC_FORCE_FOLLOW_REDIRECTS);
    httpCode = https->GET();
    Serial.print(F("Response code ")); Serial.println(httpCode, DEC);

    if (httpCode == HTTP_CODE_OK) {
      File f = LittleFS.open("cal.ics", "w");
      int n = https->writeToStream(&f);
      f.close();
      Serial.print("Download complete n="); Serial.println(n, DEC);

      g_IcsLastCheck = time(nullptr);
      taskIcsRead.restartDelayed(100);
    }
    https->end();
  }
  else
    Serial.print("Connect failed");

  delete https;
  delete client;
}
```

### Debug Messages

```
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] size: 8718
// and then nothing, return code -11
```

d-a-v, 2021-11-25:
@CapnBry thanks for reporting this issue to us.
LittleFS wrapper indeeds lacks of this implementation.
Returning `block_size` indeed unlocks `Stream*Send()` methods but this may be a little to rough.
We'll address a fix very soon.

CapnBry, 2021-11-25:
Awesome, thanks for the reply. Yeah I wasn't sure what value to return considering it could be up to the remaining space of the entire partition! Or be really inefficient and return as low as 1 haha.

Appreciate you putting this on your radar for a proper fox and thanks so much for all the other tons of code that makes my projects just an include and a few lines.

PR #8386 (2021-11-29): LittleFS: add overrides for Stream::send
fixes #8384 

`::availableForWrite()` could unconditionnally return a full sector as free space like proposed in #8384.

This proposal aims at returning a non-0 size fitting the current file block.
If the block is full and the filesystem is not full, it returns a full-FS-block size.
This way, the API allows users to take proper actions without losing data from the input stream when the filesystem is full, even if there is generally little to do in such case.
closed 2021-11-29 by d-a-v