esp8266/Arduino #8385: Arduino: problem with I2C after powering off

Lava999, 2021-11-25:
Hi everyone

This is the first time for me looking for help on GitHub, so please let me know if I can improve the way of asking :)

I am working with an JOY-IT R3DIP (Arduino compatible) to read RFID-chips. Since I need several RFID readers I use a "Grove- 8 Channel I2C Hub". In the end the state of each reader will be sent to a server by a post request but at the moment, in order to check weheather the program does work: every time a new card is detected by one of the reader and if a card is removed, a tone is played (using an active piezo buzzer).
I try to upload a picture of the hardware and hope that it works.

So far, everything works as expected when I upload the code. 
But if it is powerd of and on again (by just pull out the plug and plug it in again), the program still works but no cards are detected anymore.  By pressing the reset button of the arduino or uploading the sketch again, the problem resolves. 

Since theoretically, there should be no difference between resetting/uploading the sketch and powering off and on, I suppose that the bus somehow hangs itself when powered off.
Furthermore, I tried to directly connect one reader to the board without the intermediate bus and the problem did not occure then. 
In order to clear and restart the bus I implemented the method that is described [here](http://www.forward.com.au/pfod/ArduinoProgramming/I2C_ClearBus/index.html). The method did work and the function `I2C_ClearBus()` did return 0, however, the cards were still not detected after powering off and on. 

Does someone have an idea what the problem could be and how to resolve it?

Here is my code:

```
#include <Wire.h>
#include "MFRC522_I2C.h"
#include <SPI.h>

#define RST_PIN 6                     // Arduino UNO Pin
MFRC522_I2C mfrc522(0x28, RST_PIN);   // Create MFRC522 instance.

const int amountOfScanners = 1;       // set amount of scanners here
String    content = "";

void TCAChannel(uint8_t i) {          // I2C Multiplexer
  Wire.beginTransmission(0x70);
  Wire.write(1 << i);
  Wire.endTransmission();
}

byte mac[] = { 0x2C, 0xF7, 0xF1, 0x08, 0x32, 0x6B };

String queryString[amountOfScanners];                          // Liste mit jeweils den zu sendenden Daten in der Form eines "query strings"
String saveQueryString[amountOfScanners];                      // Liste in welcher die letzten gesendeten Daten für jeden Leser gespeichert werden -> um eine Änderung feststellen zu können
int    buzzer = 9;                                             // Pin, an welchem der active piezo buzzer angeschlossen ist

void setup() {  // -------------------------------------------------------------------------------------------------------
  delay(5000);
  Wire.begin(); 
  mfrc522.PCD_Init(); // Init MFRC522   
 
  }
 
void loop() {   //-------------------------------------------------------------------------------------------------------
  
  // Going through amount of connected scanners to read
  for (int n = 0; n < amountOfScanners; n++) {  
    TCAChannel(n);                                                                                     // Bestimmt den Eingang beim Multiplexer, der untersucht werden soll
    mfrc522.PICC_ReadCardSerial();
   
    // Look for new cards, and select one if present
    if (  ! mfrc522.PICC_IsNewCardPresent() || ! mfrc522.PICC_ReadCardSerial() ) {                     // if no card, do nothing
      content= "";
      queryString[n] = content;
      }
    else { // if new card, do this
      content = "";
      for (byte i = 0; i < mfrc522.uid.size; i++) {                                                    //generiert die ausgelesene HEX ID
        content.concat(String(mfrc522.uid.uidByte[i] < 0x10 ? "0" : ""));
        content.concat(String(mfrc522.uid.uidByte[i], HEX));
      }
      queryString[n] = content;                                                                        // setzt den aktuellen "query string" zusammen                                                              
    }   
  }
  for (int n = 0; n < amountOfScanners; n++) {   // Durchführung für jeden Eingang separat
    // überprüft, ob sich der Status vom Lesegerät verändert hat -> API und ersetzt den Speicher
    if(queryString[n] != saveQueryString[n]){    
      tone(buzzer,500);
      delay(500);  
      noTone(buzzer);
      delay(500);
      saveQueryString[n] = queryString[n];      // überschreibt den Speicher mit dem aktuell ausgelesenen Status vom Lesegerät
    }
  }
  
}
```

And here the picture of the hardware setting: 

[image]

I am happy for any help :)</div>

fsommer1968, 2021-11-27:
Try to play with the status of the IC2 pins in the setup routine, before wire.begin. That is set them to high or low level and see if it changed something. This link or similar could be of interest: https://www.esp8266.com/viewtopic.php?f=32&t=16141
Anyway in my opinion it´s somehow related to your card reader, not the 8266 itself

Lava999, 2021-11-29:
I am sorry for bothering the wrong forum. Thank you very much for your answer.

closed 2021-11-29 by Lava999