esp8266/Arduino #8387: error: 'uint16_t makeWord' redeclared as different kind of entity

Schawen, 2021-11-26:
I tried to compile an old program after updating VS + PIO and received the following errors: 

Compiling .pio\build\d1_mini_pro\FrameworkArduino\WString.cpp.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\abi.cpp.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\aes_unwrap.cpp.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\base64.cpp.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\cbuf.cpp.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\cont.S.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\cont_util.cpp.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\core_esp8266_app_entry_noextra4k.cpp.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\core_esp8266_eboot_command.cpp.o
In file included from C:\users\miko\.platformio\lib\NtpClientLib_ID727\src\NtpClientLib.h:59,
                 from C:\users\miko\.platformio\lib\NtpClientLib_ID727\src\NtpClientLib.cpp:32:
C:\users\miko\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:247:19: error: 'uint16_t makeWord' redeclared as different kind of entity
  247 | uint16_t makeWord(byte h, byte l);
      |                   ^~~~
C:\users\miko\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:246:10: note: previous declaration 'uint16_t makeWord(uint16_t)'
  246 | uint16_t makeWord(uint16_t w);
      |          ^~~~~~~~
C:\users\miko\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:247:19: error: reference to 'byte' is ambiguous
  247 | uint16_t makeWord(byte h, byte l);
      |                   ^~~~
In file included from c:\users\miko\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\bits\stl_algobase.h:61,
                 from c:\users\miko\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\array:40,
                 from c:\users\miko\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\tuple:39,
                 from c:\users\miko\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\functional:54,
                 from C:\users\miko\.platformio\lib\NtpClientLib_ID727\src\NtpClientLib.h:44,
                 from C:\users\miko\.platformio\lib\NtpClientLib_ID727\src\NtpClientLib.cpp:32:
c:\users\miko\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\bits\cpp_type_traits.h:404:30: note: candidates are: 'enum class 
std::byte'
  404 |   enum class byte : unsigned char;
      |                              ^~~~
In file included from C:\users\miko\.platformio\lib\NtpClientLib_ID727\src\NtpClientLib.h:59,
                 from C:\users\miko\.platformio\lib\NtpClientLib_ID727\src\NtpClientLib.cpp:32:
C:\users\miko\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:160:17: note:                 'typedef uint8_t byte'
  160 | typedef uint8_t byte;
      |                 ^~~~
In file included from C:\users\miko\.platformio\lib\NtpClientLib_ID727\src\NtpClientLib.h:59,
                 from C:\users\miko\.platformio\lib\NtpClientLib_ID727\src\NtpClientLib.cpp:32:
C:\users\miko\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:247:27: error: reference to 'byte' is ambiguous
  247 | uint16_t makeWord(byte h, byte l);
      |                           ^~~~
In file included from c:\users\miko\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\bits\stl_algobase.h:61,
                 from c:\users\miko\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\array:40,
                 from c:\users\miko\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\tuple:39,
                 from c:\users\miko\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\functional:54,
                 from C:\users\miko\.platformio\lib\NtpClientLib_ID727\src\NtpClientLib.h:44,
                 from C:\users\miko\.platformio\lib\NtpClientLib_ID727\src\NtpClientLib.cpp:32:
c:\users\miko\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\bits\cpp_type_traits.h:404:30: note: candidates are: 'enum class 
std::byte'
  404 |   enum class byte : unsigned char;
      |                              ^~~~
In file included from C:\users\miko\.platformio\lib\NtpClientLib_ID727\src\NtpClientLib.h:59,
                 from C:\users\miko\.platformio\lib\NtpClientLib_ID727\src\NtpClientLib.cpp:32:
C:\users\miko\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:160:17: note:                 'typedef uint8_t byte'
  160 | typedef uint8_t byte;
      |                 ^~~~
In file included from src\main.cpp:12:
C:\users\miko\.platformio\lib\NtpClientLib_ID727\src/NtpClientLib.h:501:32: error: reference to 'byte' is ambiguous
  501 |     bool summertime (int year, byte month, byte day, byte hour, byte weekday, byte tzHours);
      |                                ^~~~

Found this #8089 ,  but need more help ...

d-a-v, 2021-11-26:
There is [a fix](https://github.com/gmag11/NtpClient/pull/115).
You might consider updating to their latest version.

closed 2021-11-26 by d-a-v

Schawen, 2021-11-27:
Thx for your help.
I was able to fix the "byte" related errors in NTPClientLib and arduino.h, but there are much more errors:

- ESP8266WebServer-impl.h:624:21: error: could not convert 'args#0' from 'const char const [2]' to 'String'*
- ESP8266WebServer-impl.h:624:21: error: conversion from 'const int' to 'String' is ambiguous
- ESP8266WebServer-impl.h:624:21: **error: invalid conversion from 'int' to 
'const __FlashStringHelper*' [-fpermissive]**

In file included from C:\users\miko\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/ESP8266WebServer.h:340,
                 from src\main.cpp:10:
C:\users\miko\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h: In instantiation of 'void esp8266webserver::ESP8266WebServerTemplate<ServerType>::collectHeaders(const Args& ...) [with Args = {const char* [2], int}; ServerType = WiFiServer]':
src\main.cpp:2897:38:   required from here
C:\users\miko\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:624:21: **error: could not convert 'args#0' from 'const char* const [2]' to 'String'**
  624 |   _currentHeaders = new RequestArgument[_headerKeysCount = sizeof...(args) + 2] {
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                     |
      |                     const char* const [2]
  625 |     { .key = FPSTR(AUTHORIZATION_HEADER), .value = emptyString },
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  626 |     { .key = FPSTR(ETAG_HEADER), .value = emptyString },
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  627 |     { .key = args, .value = emptyString } ...
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  628 |   };
      |   ~
C:\users\miko\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:624:21: error: conversion from 'const int' to 'String' is ambiguous
In file included from C:\users\miko\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:286,
                 from src\main.cpp:1:
C:\users\miko\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/WString.h:59:9: note: candidate: 'String::String(const __FlashStringHelper*)' (near match)
   59 |         String(const __FlashStringHelper *str);
      |         ^~~~~~
C:\users\miko\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/WString.h:59:9: note:   conversion of argument 1 would be ill-formed:
In file included from C:\users\miko\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/ESP8266WebServer.h:340,
                 from src\main.cpp:10:
C:\users\miko\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:624:21: **error: invalid conversion from 'int' to 
'const __FlashStringHelper*' [-fpermissive]**
  624 |   _currentHeaders = new RequestArgument[_headerKeysCount = sizeof...(args) + 2] {
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                     |
      |                     int
  625 |     { .key = FPSTR(AUTHORIZATION_HEADER), .value = emptyString },
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  626 |     { .key = FPSTR(ETAG_HEADER), .value = emptyString },
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  627 |     { .key = args, .value = emptyString } ...
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  628 |   };
      |   ~
In file included from C:\users\miko\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:286,
                 from src\main.cpp:1:
C:\users\miko\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/WString.h:57:9: note: candidate: 'String::String(const char*)' (near match)
   57 |         String(const char *cstr);
      |         ^~~~~~
C:\users\miko\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/WString.h:57:9: note:   conversion of argument 1 would be ill-formed:
In file included from C:\users\miko\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/ESP8266WebServer.h:340,
                 from src\main.cpp:10:
C:\users\miko\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:624:21: error: invalid conversion from 'int' to 
'const char*' [-fpermissive]
  624 |   _currentHeaders = new RequestArgument[_headerKeysCount = sizeof...(args) + 2] {
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                     |
      |                     int
  625 |     { .key = FPSTR(AUTHORIZATION_HEADER), .value = emptyString },
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  626 |     { .key = FPSTR(ETAG_HEADER), .value = emptyString },
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  627 |     { .key = args, .value = emptyString } ...
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  628 |   };
      |   ~
src\main.cpp: In function 'bool readDHTSensor()':
src\main.cpp:2111:1: error: control reaches end of non-void function [-Werror=return-type]
 2111 | }
      | ^

d-a-v, 2021-11-27:
Apparently there is a mix of versions.
Did you copy older internal libraries outside ?
Did you consider removing and reinstalling this core ?
Don't forget the 'arduino15' directory (google about it).

Schawen, 2021-11-28:
**>> Did you copy older internal libraries outside ?**
I don't think so ...

**>> Did you consider removing and reinstalling this core ?**
Sorry, which core should I reinstall? PlatformIO?

removed this from my projects platformio.ini:
platform = https://github.com/platformio/platform-espressif8266.git

Uninstalled and reinstalled Arduino IDE
Deleted and reinstalled all platforms (atmelAVR, ESP32, esp8266) in PlatformIO

Now only the errors regarding ESP8266WebServer-impl.h are left over

-   ESP8266WebServer-impl.h:624:21: error: could not convert 'args#0' from 'const char const [2]' to 'String'*
-   ESP8266WebServer-impl.h:624:21: error: conversion from 'const int' to 'String' is ambiguous
-   ESP8266WebServer-impl.h:624:21: error: invalid conversion from 'int' to 'const __FlashStringHelper' [-fpermissive]*

I also found these folders under 
C:\Users\xyz\.platformio\packages:

- framework-arduinoespressif8266
- framework-arduinoespressif8266@3.20704.0
- framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910

and deleted them completely.

**>> Don't forget the 'arduino15' directory (google about it).**
Uninstalled ArduinoIDE again,
also deleted the aduino15 directory, but 
these errors stay:
-   ESP8266WebServer-impl.h:624:21: error: could not convert 'args#0' from 'const char const [2]' to 'String'*
-   ESP8266WebServer-impl.h:624:21: error: conversion from 'const int' to 'String' is ambiguous
-   ESP8266WebServer-impl.h:624:21: error: invalid conversion from 'int' to 'const __FlashStringHelper' [-fpermissive]*

What else can I do???