esp8266/Arduino #8400: Version 3.0.0 up break my92xx library

CBMalloch, 2021-12-06:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266
- Core Version:			3.0.2
- Development Env:		Arduino IDE
- Operating System:		MacOS

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			dout
- Flash Size: 		1MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		OTA or serial
- Upload Speed:   115200 when serial

### Problem Description

Sending code to an Itead SONOFF B1 light bulb, which uses a my9231 chip to control the 5 channels of LED. When I use ESP8266 board definitions 2.7.4, everything works as planned. If I update to 3.0.0 or above, the bulb flashes more or less at random as I adjust values. The brightnesses no longer scale smoothly with the commands, nor go to the proper channels.
Seems funny, since the my92xx library isn't using much - only digitalWrites and the c code for os_delay_us. I tried replacing the os_delay_us definition with delayMicroseconds, without any better luck.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

#define PROGNAME "ESP8266_B1_bulb"
#define VERSION  "1.1.0"
#define VERDATE  "2021-12-06"

  /*
    ESP8266_B1_bulb.ino
    2021-02-15
    Charles B. Malloch, PhD
    
    Program to control a Sonoff B1 Bulb
    Target hardware: ESP8266 on a Sonoff B1 RGB WW CW LED light bulb on Edison base
      
    v0.0.1  2021-02-13 cloned from ESP8266_Sonoff v1.1.1
    v1.2.1  2021-02-16 cbm repaired timer
    v0.2.4  2021-04-06 cbm ready for production test
    v0.2.15 2021-04-27 cbm added veil of indication
    v0.3.0  2021-11-29 cbm modified to alternately start up with red or white
    v0.3.7  2021-12-02 cbm removed timed auto-reboot, 'cause we don't retain color over
                           a soft restart. Could, with work, retrieve it from MQTT "status"
    v0.4.0  2021-12-02 cbm now fixed; writing and reading soft-restart file
    v1.0.0  2021-12-02 cbm calling it production!
    v1.1.0  2021-12-06 cbm substituting on-time for reset reason to keep old color
    
    Sonoff B1 Bulb Pinout, CW from CCW, for wiring to an ESP-01 programming adapter:
      1 Vcc (3.3) red
      2 RX        yellow NOTE this is the RX of the ESP, so RX header hole ( next to 3V3 )
      3 TX        green
      4 GND       black
      5 GPIO0     purple
    
    Use adapter from Sonoff box
        
    Schematic:
      https://user-images.githubusercontent.com/17343162/29202638-6b2c39ee-7e6a-11e7-93eb-727d7efaa24c.png
      3-color LEDs are controlled by my9231 chip
[... 2046 lines cut ...]

  // try to write status to LittleFS
  
  // unsigned long s = millis();

  File f;
  f = LittleFS.open ( "/soft_reset_color.txt", "w" );
  f.printf ( "%d,%d,%d, %d,%d, %d", currentR, currentG, currentB,
                                    currentW, currentC,
                                    outputState );
  f.close();
  
  // unsigned long interval = millis() - s;
  // Serial.printf ( "LittleFS file write took %lu ms\n", interval );
}

// *****************************************************************************
// *****************************************************************************
// *****************************************************************************

### Debug Messages

```
Debug messages go here
```

mcspr, 2021-12-06:
You sure we are better served with the issue here instead of https://github.com/xoseperez/my92xx/issues?

Is this the actual MCVE, and channels don't properly adjust from 0 to 255?
```cpp
#include <Arduino.h>
#include <my92xx.h>

#define MY92XX_MODEL        MY92XX_MODEL_MY9231
#define MY92XX_CHIPS        2
#define MY92XX_DI_PIN       12
#define MY92XX_DCKI_PIN     14

#define MY92XX_COLD         0
#define MY92XX_WARM         1
#define MY92XX_RED          4
#define MY92XX_GREEN        3
#define MY92XX_BLUE         5

static constexpr unsigned char MY92XX_CHANNELS[] {
    MY92XX_RED,
    MY92XX_GREEN,
    MY92XX_BLUE,
    MY92XX_WARM,
    MY92XX_COLD
};

static my92xx* _my92xx { nullptr };

void setup() {
    _my92xx = new my92xx ( MY92XX_MODEL, MY92XX_CHIPS,
            MY92XX_DI_PIN, MY92XX_DCKI_PIN,
            { \
            .scatter = MY92XX_CMD_SCATTER_APDM, \
            .frequency = MY92XX_CMD_FREQUENCY_DIVIDE_1, \
            .bit_width = MY92XX_CMD_BIT_WIDTH_8, \
            .reaction = MY92XX_CMD_REACTION_FAST, \
            .one_shot = MY92XX_CMD_ONE_SHOT_DISABLE, \
            .resv = 0 \
            }
        );

    _my92xx->setState(true);

    for (;;) {
        for (unsigned char value = 0; value < 255; ++value) {
            for (auto channel : MY92XX_CHANNELS) {
                _my92xx->setChannel(channel, value);
            }
            _my92xx->update();
            delay(100);
        }
    }
}
                                                                                                                                                                                                                                             void loop() {                                                                                                                                                                                                                                }
```

`delayMicroseconds(...)` is `os_delay_us(...)` so no surprise there :)
https://github.com/esp8266/Arduino/blob/c312a2eaf1356ceaafad7c4935fa850e087c84fe/cores/esp8266/core_esp8266_wiring.cpp#L189-L191

Since we are expected to digitalWrite in a tight sequence... Have you tried locking interrupts before doing `_my92xx->update()` in the `sendOutput()`, like the library source have tried originally (but commented out)? Something like
```cpp
 ets_intr_lock();
 _my92xx->update();
 ets_intr_unlock();
```

CBMalloch, 2021-12-24:
I just tried locking interrupts before the update call. No change. :(