esp8266/Arduino #8401: Compiler not ignoring stray { curly-braces in “//” comments and “#if (false)” sections

Rob58329, 2021-12-07:
### Basic Infos

- [yes ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [yes ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [yes ] I have tested that the issue is present in current master branch (aka latest git).
- [yes ] I have searched the issue tracker for a similar issue.
- [n/a] If there is a stack dump, I have decoded it.
- [yes ] I have filled out all fields below.

#### Platform

- Hardware:			ESP8266 D1mini or Generic Board
- Core Version:			github.com/esp8266/Arduino downloaded 7Dec21
- Development Env:		[Arduino IDE v1.8.16]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module &/or Wemos D1 mini]
- Flash Mode:			[dio]
- Flash Size: 		        [4MB]
- lwip Variant:                   [v2 Lower Memory]
- Reset Method:		[?]
- Flash Frequency:	        [?]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:               [921600]

### Problem Description

Compiler not ignoring stray { curly-braces in “//” comments and “#if (false)” sections.

I have had issues with the above on a number of different projects over the past few years.  Basically even though a sketch compiles without warnings, an update for a project will stop it working, and it will take some time to find out it is a stray curly-brace in a comment somewhere which is causing the issue.

The below sketch demonstrates this issue.  As included below it will give the incorrect 65535 when run, but DELETE either of the two comment lines, and it will give the correct approx 3000 when run.
(The D1mini has a voltage divider on the A0 pin, which includes a resistor from A0 to ground, which will cause the above 3000 result to be lower than the real approx 3300 you get with a generic board, however it still demonstrates the issue OK.}

I note that there are several discussions regarding curly-braces for the Arduino-IDE (eg. https://forum.arduino.cc/t/curly-braces-in-strings-or-quotes/212575 )

I also note that in the Arduino-IDE, the “control-T” function USED to include/notice curly-braces in comments and format the sketch including them, making them easier to spot.  But sadly this changed a number of years ago, so “control-T” now ignores curly-brackets in comments, and so no longer helps in finding curly-braces buried in comments!

As this would seem to be a compiler "feature", I am unsure if anything can be done about it, but thought I would raise it as an issue just in case!

```
#if (false) // The below lines should not be enabled, and so should not have any effect 
#error "Should not get to here"
// {               // either delete this line,
// extern "C" {}   // or delete this line, and the below code works fine!
#endif

ADC_MODE(ADC_VCC);

void setup() {
  Serial.begin(74880);
}

void loop() {
  delay(5000);
  Serial.print("raw_battery (should be between 3000 and 3300)="); Serial.println(ESP.getVcc());
  // unless you DELETE one of the above two COMMENT lines, you will get "=65535" and not "=3017"
}

// Note that for working AND non-working, we always get the same resulting sketch size:
// Sketch uses 265437 bytes (25%) of program storage space. Maximum is 1044464 bytes.
// Global variables use 28044 bytes (34%) of dynamic memory, leaving 53876 bytes for local variables. Maximum is 81920 bytes.

```

mcspr, 2021-12-08:
This is an https://github.com/arduino/arduino-cli problem though? It is my understanding that it handles the .ino -> .cpp conversion and causes the issue described here (at least in the recent IDE versions).
I'd suggest to forward the issue there, as this will happen with any Core trying to build this sketch.

Note that you can add a separate `.cpp` file, which includes the `#include <Arduino.h>` and `ADC_MODE(ADC_VCC);` lines to have more consistent compilation results.

---
.ino -> .cpp conversion results can be seen after enabling verbose mode in the IDE logs and opening the temporary directory location. Or, by manually building with `arduino-cli` 'compile --verbose ...'
(or, just change '--build-path' to some known location)

8401.ino.cpp with the code above, .ino -> .cpp erroneously adds extern "C" to everything below the comments
```cpp
#include <Arduino.h>
#line 1 "C:\\Users\\maxim\\Documents\\Arduino\\esp8266-test\\8401\\8401.ino"
#if (false) // The below lines should not be enabled, and so should not have any effect
#error "Should not get to here"
// {               // either delete this line,
// extern "C" {}   // or delete this line, and the below code works fine!
#endif

#line 7 "C:\\Users\\maxim\\Documents\\Arduino\\esp8266-test\\8401\\8401.ino"
 extern "C" int __get_adc_mode(void);
#line 9 "C:\\Users\\maxim\\Documents\\Arduino\\esp8266-test\\8401\\8401.ino"
 extern "C" void setup();
#line 13 "C:\\Users\\maxim\\Documents\\Arduino\\esp8266-test\\8401\\8401.ino"
 extern "C" void loop();
#line 7 "C:\\Users\\maxim\\Documents\\Arduino\\esp8266-test\\8401\\8401.ino"
ADC_MODE(ADC_VCC);

void setup() {
  Serial.begin(74880);
}

void loop() {
  delay(5000);
  Serial.print("raw_battery (should be between 3000 and 3300)="); Serial.println(ESP.getVcc());
  // unless you DELETE one of the above two COMMENT lines, you will get "=65535" and not "=3017"
}
```

Otherwise, it works as expected
```cpp
#include <Arduino.h>
#line 1 "C:\\Users\\maxim\\Documents\\Arduino\\esp8266-test\\8401\\8401.ino"
#if (false) // The below lines should not be enabled, and so should not have any effect
#error "Should not get to here"
#endif

#line 5 "C:\\Users\\maxim\\Documents\\Arduino\\esp8266-test\\8401\\8401.ino"
int __get_adc_mode(void);
#line 7 "C:\\Users\\maxim\\Documents\\Arduino\\esp8266-test\\8401\\8401.ino"
void setup();
#line 11 "C:\\Users\\maxim\\Documents\\Arduino\\esp8266-test\\8401\\8401.ino"
void loop();
#line 5 "C:\\Users\\maxim\\Documents\\Arduino\\esp8266-test\\8401\\8401.ino"
ADC_MODE(ADC_VCC);

void setup() {
  Serial.begin(74880);
}

void loop() {
  delay(5000);
  Serial.print("raw_battery (should be between 3000 and 3300)="); Serial.println(ESP.getVcc());
  // unless you DELETE one of the above two COMMENT lines, you will get "=65535" and not "=3017"
}
```

closed 2021-12-08 by mcspr

Rob58329, 2021-12-15:
@mcspr: Very many thanks for your above comments!

The “arduino_build_xxx/sketch/yyy.cpp” temporary file is very usefull in finding these stray curly-brackets - thank you!

I have now submitted this issue as https://github.com/arduino/arduino-cli/issues/1591, and it will be interesting to see if they can do anything about it!