esp8266/Arduino #8402: SNTP does not start if WiFi.config called with staticIP

drmpf, 2021-12-08:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					ESP-01
- Core Version:			Ver 3.0.2
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			DOUT
- Flash Size: 		1MB]
- lwip Variant:   v2 Lower Memory
- Reset Method:		dtr
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		OTA
- Upload Speed:   115200

### Problem Description

SNTP fails to connect if WiFi uses static IP

Open supplied example file  ESP8266 -> NTP-TZ-DST and replace
  // start network
  WiFi.mode(WIFI_STA);
  WiFi.begin(STASSID, STAPSK);

with 
  // start network
  WiFi.mode(WIFI_STA);
  IPAddress staticIP; staticIP.fromString("10.1.1.212");
  IPAddress gatewayIP; staticIP.fromString("10.1.1.1");
  IPAddress subnetIP; staticIP.fromString("255.255.255.255");
  WiFi.config(staticIP,gatewayIP,subnetIP);
[... 275 lines cut ...]
  yield();

  // start network
  WiFi.mode(WIFI_STA);
  IPAddress staticIP; staticIP.fromString("10.1.1.212");
  IPAddress gatewayIP; staticIP.fromString("10.1.1.1");
  IPAddress subnetIP; staticIP.fromString("255.255.255.255");
  WiFi.config(staticIP,gatewayIP,subnetIP);
  WiFi.begin(STASSID, STAPSK);

  // don't wait for network, observe time changing
  // when NTP timestamp is received
  showTime();
}

void loop() {
  if (showTimeNow) {
    showTime();
  }
}

```

### Debug Messages
settimeofday(SNTP)  does NOT appear in the console if using staticIP

mcspr, 2021-12-08:
```cpp
  configTime(MYTZ, "pool.ntp.org");
```
```cpp
  IPAddress staticIP; staticIP.fromString("10.1.1.212");
  IPAddress gatewayIP; gatewayIP.fromString("10.1.1.1");
  IPAddress subnetIP; subnetIP.fromString("255.255.255.0");
  WiFi.config(staticIP,gatewayIP,subnetIP);
```
Can't resolve `pool.ntp.org` without DNS :)
I'd guess, `configTime(MYTZ, "10.1.1.1")` might also work if your gateway has NTP server.

closed 2021-12-08 by mcspr

drmpf, 2021-12-08:
So what does not having DHCP to do with DNS?
setting a staticIP just set the IP of the ESP8266 without using DHCP.  
How does that effect the operation of DNS?

Pablo2048, 2021-12-08:
On any ip configuration (DHCP/Static) you need DNS server for acessing symbolic names like "pool.ntp.org". In DHCP configuration the DHCP server is responsible for setting the primary and secondary DNS servers. In static configuration the responsibility is up to you.

drmpf, 2021-12-08:
Ok found this
Edit 
   bool config(IPAddress local_ip, IPAddress gateway, IPAddress subnet, IPAddress dns1 = (uint32_t)0x00000000, IPAddress dns2 = (uint32_t)0x00000000);
i.e. the default config( ) call does disable DNS

Pablo2048, 2021-12-08:
Yes, you can set the addresses for example to public Google DNS (8.8.8.8 , 8.8.4.4) and give it a try...

drmpf, 2021-12-08:
Yes thanks just tried that and that worked thanks

Nourbakhsh-Rad, 2021-12-15:
I tried to test with WiFi and static IP, everything was OK
But with LwIP_Ethernet, even though I registered DNS, S-NTP does not work