esp8266/Arduino #8404: About Websocket

Ciapas-Linux, 2021-12-12:
I warmly welcome
My problem is that the websocket libraries that I use to communicate esp with the browser do not work with the current version 3.02 of this software. I am using version 2.7.4 and websocket works here. I have tried several websoket libraries for esp and none of them work and they work with older versions. Does anyone know anything about it? I asked one creator of the websocket library but he didn't answer, maybe he doesn't have time. I do not know if it is good that I wrote on the issue here, but I do not know where to ask about it, let the admin delete my entry, cheers.

hbosko, 2021-12-12:
Hi,

I am using [Links2004 ](https://github.com/Links2004/arduinoWebSockets) library, and works great.
Esp8266 3.02.

d-a-v, 2021-12-13:
Also if I may, with this library, port 80 is shared between webserver and websocketserver.

Ciapas-Linux, 2022-01-01:
welcome

Eventually the problem was solved.

In the old version 2.7.4, you could directly assign an int value to a String:

String sTmpStr;
int data = 10;
sTmpStr = data; // this produces error with new version not utf-8 data and error
						  // when send text over websockets

Maybe it's strange but it worked on the old version but on the new 3
here was an error when sending a wrong utf-8 frame then I changed it to:

String sTmpStr;
int data = 10;
char TempCharBuffer [12] = "0000000000 \ 0";
itoa (data, TempCharBuffer, 10);
sTmpStr = TempCharBuffer;
sTmpStr.trim();

it works on version 3.02

greetings

closed 2022-01-01 by Ciapas-Linux