esp8266/Arduino #8405: Backport getLocalTime from ESP32

h1aji, 2021-12-13:
Was trying to port this code from ESP32, but found that getLocalTime doesnt exist. Is there a replacement?

```
void gettime()
{
  static int16_t delaycount = 0 ;                         // To reduce number of NTP requests
  static int16_t retrycount = 100 ;

  if ( timeinfo.tm_year )                                 // Legal time found?
  {
    sprintf ( timetxt, "%02d:%02d:%02d",                  // Yes, format to a string
              timeinfo.tm_hour,
              timeinfo.tm_min,
              timeinfo.tm_sec ) ;
  }
  if ( --delaycount <= 0 )                                // Sync every few hours
  {
    delaycount = 7200 ;                                   // Reset counter
    if ( timeinfo.tm_year )                               // Legal time found?
    {
      dbgprint ( "Sync TOD, old value is %s", timetxt ) ;
    }
    dbgprint ( "Sync TOD" ) ;
    if ( !getLocalTime ( &timeinfo, 5000 ) )              // Read from NTP server
    {
      dbgprint ( "Failed to obtain time!" ) ;             // Error
      timeinfo.tm_year = 0 ;                              // Set current time to illegal
      if ( retrycount )                                   // Give up syncing?
      {
        retrycount-- ;                                    // No try again
        delaycount = 5 ;                                  // Retry after 5 seconds
      }
    }
    else
    {
      sprintf ( timetxt, "%02d:%02d:%02d",                // Format new time to a string
                timeinfo.tm_hour,
                timeinfo.tm_min,
                timeinfo.tm_sec ) ;
      dbgprint ( "Sync TOD, new value is %s", timetxt ) ;
    }
  }
}

```

Or is there any chance to add this from [esp32-hal-time.c](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-time.c)

```
bool getLocalTime(struct tm * info, uint32_t ms)
{
    uint32_t start = millis();
    time_t now;
    while((millis()-start) <= ms) {
        time(&now);
        localtime_r(&now, info);
        if(info->tm_year > (2016 - 1900)){
            return true;
        }
        delay(10);
    }
    return false;
}

```

d-a-v, 2021-12-13:
Why don't you directly use [localtime()](https://linux.die.net/man/3/localtime) ?

Anyway if you think this core should get `bool getLocalTime(struct tm * info, uint32_t ms)` for better inter-operability, then your PR is welcome.

h1aji, 2021-12-13:
Can you give me a tip? do you mean `if ( !localtime ( &timeinfo ) )  ` ?

d-a-v, 2021-12-13:
Yes. There is [an example](https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino).
I guess you can also simply copy-paste and use your above `getLocalTime()`.

h1aji, 2021-12-14:
Tried localtime_r() but it shows incorrect time, I assume its because of 5 sec delay.
Anyway, its good to have this function for ESP8266 available. Created PR, please review here https://github.com/esp8266/Arduino/pull/8406
Thanks

h1aji, 2021-12-14:
new PR here https://github.com/esp8266/Arduino/pull/8407