esp8266/Arduino #8412: Wifi connection fails - only works after router reboot

Toto5931, 2021-12-15:
### Basic Infos

- [X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X] I have tested that the issue is present in current master branch (aka latest git).
- [X ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ X] I have filled out all fields below.

#### Platform

- Hardware: esp8266
- Core Version: 3.0.2
- Development Env: Arduino IDE
- Operating System: Windows

### Settings in IDE

- Module:  Wemos D1 mini 
- Flash Mode:	DOUT
- Flash Size:  4MB
- lwip Variant: v2 Lower Memory
- Flash Frequency:	40Mhz
- CPU Frequency:	80Mhz
- Upload Using: SERIAL
- Upload Speed: 921600

### Problem Description

The WIFI connection is very unstable. It often works just after router restart but few times later the connection fails.
The debug message tells me that I am connected but the status stays at 0 and never goes to WL_CONNECTED
I have tried with many ESP8266 wemos d1 mini hardware it is always the same problem
The connection works fine with an ESP32.
Thanks for your help

```
#include <ESP8266WiFi.h> 

void setup()
{
  Serial.begin(9600);
  Serial.println();

  WiFi.begin("SFR_5758", "xxxxxxxxxxx");

  Serial.print("Connecting");

  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.print(".");
  }
  
  if (WiFi.status() == WL_CONNECTED) {
    Serial.print("Connected, IP address: ");
    Serial.println(WiFi.localIP());
  }
}

void loop() {}
```

### Debug Messages

```
22:29:28.936 -> SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
22:29:29.076 -> 
22:29:29.076 -> fpm close 1 
22:29:29.076 -> mode : sta(8c:aa:b5:7b:1f:5f)
22:29:29.076 -> add if0
22:29:29.123 -> Connectingwifi evt: 8
22:29:29.962 -> .wifi evt: 2
22:29:30.991 -> .scandone
22:29:32.724 -> state: 0 -> 2 (b0)
22:29:32.724 -> .state: 2 -> 3 (0)
22:29:32.770 -> state: 3 -> 5 (10)
22:29:32.770 -> add 0
22:29:32.770 -> aid 3
22:29:32.770 -> cnt 
22:29:32.770 -> state: 5 -> 2 (2c0)
22:29:32.770 -> rm 0
22:29:32.816 -> wifi evt: 1
22:29:32.816 -> STA disconnect: 2
22:29:33.703 -> .reconnect
22:29:33.749 -> state: 2 -> 0 (0)
22:29:33.841 -> scandone
22:29:33.841 -> state: 0 -> 2 (b0)
22:29:33.841 -> state: 2 -> 3 (0)
22:29:33.887 -> state: 3 -> 5 (10)
22:29:33.935 -> add 0
22:29:33.935 -> aid 3
22:29:33.935 -> cnt 
22:29:33.935 -> 
22:29:33.935 -> connected with SFR_5758, channel 11
22:29:33.983 -> dhcp client start...
22:29:33.983 -> wifi evt: 0
22:29:34.686 -> ..........pm open,type:2 0
22:29:44.684 -> .............................................................................................
```

1d4rk, 2021-12-16:
You can try to force 802.11g mode:

WiFi.setPhyMode(WIFI_PHY_MODE_11G);

With Asus router it worked.

Toto5931, 2021-12-16:
Thanks a lot ! I solved the problem
I do not understand why v3.0.2 takes always 7s to connect to the wifi whereas v2.7.4 takes either 7s, either 3s. So much faster in average...

d-a-v, 2021-12-16:
@Toto5931
Can you have a look and try https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/generic-class.html#persistent and report it it changes connection delay ?

Toto5931, 2022-01-04:
Thank you @d-a-v !

nathanielarking, 2022-01-12:
Hi there Toto, are you able to share exactly which solution solved your problem? I'm having a similar problem where my ESP8266 will be able to connect to WiFi for 20 minutes or so and then will require a restart of the router to connect again.

Toto5931, 2022-01-16:
Just adding WiFi.setPhyMode(WIFI_PHY_MODE_11G); as proposed by 1d4rk solved my problem

sblantipodi, 2022-02-06:
I have the same problem, downgrading the connection from N to G does not solve the problem

nathanielarking, 2022-02-06:
@sblantipodi It didn't solve mine either, and I ended up fixing it by throwing a whole bunch of resets: I cleared the flash and wifi settings on the ESP, reverted to a previous version of my code, and defaulted the router settings, so I'm not exactly sure what works. Try re starting your code with the most basic WiFi settings possible (use the basic ESP example for connecting to WiFi) then change things from their until you get the functionality you did before.

cxcorp, 2022-04-09:
I had this issue and it was related to #7965.

billjuv, 2022-08-20:
> You can try to force 802.11g mode:
> 
> `WiFi.setPhyMode(WIFI_PHY_MODE_11G);`
> 
> With Asus router it worked.

I wish I had found this comment about a half a day earlier. Solved the issue for me. My newer Asus router is set to “smart connect“ instead of separate 2.4 and 5 bands. This may have been the issue but I didn’t troubleshoot further once I was able to move forward on my project.

1d4rk, 2023-01-31:
I tested the new Arduino ESP8266 v3.1.1 Library (based on nonos_SDK v3.0.5) and it is working without any issue for now.
The connection is working good in 802.11n mode + WMM enabled (so no need to force 802.11g mode). 
The problem was due to the older ESP8266 Arduino core, starting from 2.6.0, that disabled WMM/WME support in 802.11n mode and it was added in the last release (because it is based on Espressif nonos_SDK 3.0.5 library).