esp8266/Arduino #8415: UDP services dos not work properly with LwIP_Ethernet and static IP

Nourbakhsh-Rad, 2021-12-17:
I am using ESP12-E with ENC28J60 ethernet module for my tests

I tried to get the date-time with configTime and AsyncUDP (ESPNtpClinent library)

1. With DHCP everything is OK  ( LwIP_Ethernet and WiFi )
2. When I am using static IP, even though I registered DNS, my sketch dos not work with LwIP_Ethernet  but with WiFi everything is OK

It seems the UDP services dos not work properly with LwIP_Ethernet and static IP

test sketch: [001.txt](https://github.com/esp8266/Arduino/files/7735414/001.txt)

d-a-v, 2021-12-17:
I / We / one have yet to try to run and check some examples involving asyncTCP/UDP/Webserver with the lwIP on ethernet driver.

In the meantime did you try with [this example](https://github.com/esp8266/Arduino/tree/master/libraries/esp8266/examples/NTP-TZ-DST) ?

Nourbakhsh-Rad, 2021-12-17:
> I / We / one have yet to try to run and check some examples involving asyncTCP/UDP/Webserver with the lwIP on ethernet driver.
> 
> In the meantime did you try with [this example](https://github.com/esp8266/Arduino/tree/master/libraries/esp8266/examples/NTP-TZ-DST) ?

my test sketch (attached) is similar that example, can you test it?

d-a-v, 2021-12-17:
I will when I get my hands back on my ethernet setup.

Nourbakhsh-Rad, 2021-12-23:
Hi @d-a-v 
As you can see in the attached image, static IP connection is established ( lwIP_Ethernet ) and DNS works fine ( NTP server address resolved ), but data is not sent through AsyncUDP

Nourbakhsh-Rad, 2023-02-24:
@d-a-v Hi, my problem was not solved yet! There is no solution?