esp8266/Arduino #8421: Problem with int function which doesn't return a value

thierry7100, 2021-12-24:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ]X I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X] I have tested that the issue is present in current master branch (aka latest git).
- [ X] I have searched the issue tracker for a similar issue.
- [ X] If there is a stack dump, I have decoded it.
- [ X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[3.0.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Wemos D1 mini r2]
- Flash Mode:			[?]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I have build a function returning an int but I forgot to return a value. This error is NOT detected by the IDE and worse the function doesn't not exist properly. In fact, the function doesn't exit and resume the loop (see below)

In the simple example provided, I got an exception, but in the complete program, nothing at all, the loop doesn't stop running after the limit was reached without error on the console.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

int TestFunction()
{
long int val = 1;
  for ( int r = 0; r < 10; r++ )
  {
    for ( int c = 0; c < 10; c++)
    {
      val = val + 5;
    }
  }
}

void setup() {

  Serial.begin(115200);
  delay(500);
  Serial.println("Starting and calling TestFunction");
  TestFunction();
  Serial.println("Done.\n");

}

void loop() {
  // put your main code here, to run repeatedly:

}

### Debug Messages
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00042760
~ld

SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
Starting and calling TestFunction
Fatal exception 9(LoadStoreAlignmentCause):
epc1=0x40201050, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000003b, depc=0x00000000

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (9):
epc1=0x40201050 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000003b depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffe00 end: 3fffffc0 offset: 0190
3fffff90:  3fffdad0 00000000 3ffee568 40201050  
3fffffa0:  feefeffe feefeffe 3ffee5bc 40201cb4  
3fffffb0:  feefeffe feefeffe 3ffe85d8 40100ea9  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

mcspr, 2021-12-24:
should be a fatal error since https://github.com/esp8266/Arduino/pull/8165 via `-Werror=return-type`
https://github.com/esp8266/Arduino/blob/b5f3d1d6d57281642f3e034a215146da61b0771e/platform.txt#L24-L28

> Core Version: [3.0.2]

you should not be able to build the binary, can you check that the verbose build logs for .ino.cpp compilation actually have the flag above?

mcspr, 2021-12-24:
...but, there's still an issue with the g++ arguments in 'no warnings' mode, `-w` takes precedence over `-Werror=...`
```
$ g++ -v
...
gcc version 11.2.1 20210728 (Red Hat 11.2.1-1) (GCC)
$ g++ -w -Werror=return-type -o /dev/null no-return.cpp
$ cat no-return.cpp
int test() {
}

int main() {
    test();
}
```
but I'm not sure whether we can do something about it

thierry7100, 2021-12-24:
First, thanks for your prompt answer. 
There is indeed a problem with gcc options. 

But I think there is also an issue with the generated code. As explained, when the loop was over, the function did NOT return properly and the loop resumed its execution with vale’s of loop index well above the limit. It was very strange and it was quite difficult to understand the real error (lack of return value). Once the function modified with a return value everything is working fine.

mcspr, 2021-12-24:
Generated code is subject to the undefined behaviour, so anything could
happen at runtime :)

"Unlike in C, in C++, flowing off the end of a non-void function other than
main results in undefined behavior even when the value of the function is
not used."

https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#:~:text=Unlike%20in%20C%2C%20in%20C%2B%2B%2C%20flowing%20off%20the%20end%20of%20a%20non%2Dvoid%20function%20other%20than%20main%20results%20in%20undefined%20behavior%20even%20when%20the%20value%20of%20the%20function%20is%20not%20used

closed 2022-02-20 by mcspr