esp8266/Arduino #8422: LiquidCrystal_I2C library for ESP8266 core 2.6.0+

cniedzi, 2021-12-25:
I successfuly use LiquidCrystal_I2C library (https://github.com/lucasmaziero/LiquidCrystal_I2C) with Wemos D1 mini Pro & LCD 16x2 display with ESP8266 core 2.5.0.
With any ESP core version starting from 2.6.0 or higher this library doesn't work. Is there any LiquidCrystal_I2C library working with ESP8266 core 2.6.0+?

mcspr, 2021-12-25:
```patch
diff --git a/LiquidCrystal_I2C.cpp b/LiquidCrystal_I2C.cpp
index 7be5887..d74f3a5 100644
--- a/LiquidCrystal_I2C.cpp
+++ b/LiquidCrystal_I2C.cpp
@@ -239,8 +239,9 @@ inline void LiquidCrystal_I2C::command(uint8_t value) {
        send(value, 0);
 }

-inline size_t LiquidCrystal_I2C::write(uint8_t value) {
+size_t LiquidCrystal_I2C::write(uint8_t value) {
        send(value, Rs);
+    return 1;
 }

diff --git a/LiquidCrystal_I2C.h b/LiquidCrystal_I2C.h
index d5ef237..af29e5a 100644
--- a/LiquidCrystal_I2C.h
+++ b/LiquidCrystal_I2C.h
@@ -165,7 +165,7 @@ public:
        void noAutoscroll();
        void createChar(uint8_t, uint8_t[]);
        void setCursor(uint8_t, uint8_t);
-       virtual size_t write(uint8_t);
+       size_t write(uint8_t) override;
        void command(uint8_t);

        inline void blink_on() { blink(); }
```

If you meant that it does not build, check out whether these changes help. For the issue, let's use https://github.com/lucasmaziero/LiquidCrystal_I2C/issues/3 for tracking instead

Notice that we have discussions, gitter and a community forum for general questions. You might find an answer for the 2nd question there

closed 2021-12-25 by mcspr

cniedzi, 2021-12-25:
Thx for reply. The code does build, but the LCD display doesn't display any text, it is backlighted only. Ok, let's switch to lucasmaziero/LiquidCrystal_I2C#3 issues board.

mathertel, 2021-12-27:
If you use an adapter with a PCF8574 chip over i2c you may try LiquidCrystal_PCF8574.

cniedzi, 2021-12-27:
> If you use an adapter with a PCF8574 chip over i2c you may try LiquidCrystal_PCF8574.

Thx @mathertel Matthias, it does work for me !!! :)

But I have to migrate some code from core 2.5.0 to 2.5.1+. Especially I have used interrupts without IRAM_ATTR, but with 3.0.0+ it doesn't work without this attribute so I have to use it. Do you know is IRAM_ATTR the same as ICACHE_RAM_ATTR? With core 2.5.0 the IRAM_ATTR doesn't compile and I have to use ICACHE_RAM_ATTR instead.