esp8266/Arduino #8446: HTTPClient problems with 2 parameters and MQTT

CCVEI, 2022-01-09:
The HTTPClient in esp8266 has problems if we use at the same time the MQTT protocol.
The problem appear since the http.begin starts to force the use of more than one paremeter.

The last version that works is the esp8266 2.7.4 (if we use it with just one parameter)

Simptoms:
It will execute just one time the http.begin (inside the main loop) and block.

d-a-v, 2022-01-10:
Can you adapt your code according to [examples](https://github.com/esp8266/Arduino/blob/2c5885e29d7a5bf0095684d96662f52cf7e9565d/libraries/ESP8266HTTPClient/examples/BasicHttpClient/BasicHttpClient.ino#L43-L48) ?

For further investigation, you must show an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example).

CCVEI, 2022-01-10:
void loop()   
  nodeEVMQTT.loop(); 
  
  if(!nodeEVMQTT.connected()){ 
         nodeEVMQTT.connect("NodeHome", mqttUser, mqttPassword);
  }
   
    if (timeElapsed >= interval) {
              
              v1 = pzem1.voltage();
              i1 = pzem1.current();
              p1 = pzem1.power();
              phi1 = pzem1.pf();
                
                     //send data to server to insert in DB
                     getData = "?local=GERAL&amps=" + String (i1) + "&volts=" + String (v1) + "&watts=" + String (p1); //construo o link a enviar
                     link = "http://192.168.1.105/consumo.php" + getData;
                     http.begin(nodeEV20,link);
                     int httpCode = http.GET();
                     http.end();
              
           dtostrf(i1,2,1,i_str);
           dtostrf(v1,2,1,v_str);
           dtostrf(p1,2,1,p_str);

           nodeEVMQTT.publish("/HomeAmps", i_str);
           nodeEVMQTT.publish("/HomeVolts",v_str);
           nodeEVMQTT.publish("/HomeWatts",p_str);
          
          timeElapsed = 0;
            
    }
    httpServer.handleClient();    
}

d-a-v, 2022-01-10:
There is no indication of what is `nodeEV20`.
Your `timeElapsed` is not increased.

This is not an MCVE.
You did not even try to declare a local wifi client like in the example I showed you.
You don't help us help you.

CCVEI, 2022-01-10:
timeElapsed works well.
nodeEV20 is a WifiClient object.

Until v2.7.4 all works fine, since http.begin has 2 parameters, it will not work.
____________________________________________________________________________________________________
#include <elapsedMillis.h>
#include <PZEM004Tv30.h>
#include <Wire.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266mDNS.h>
#include <ESP8266HTTPUpdateServer.h>
#include <PubSubClient.h>
//#include <LiquidCrystal_I2C.h>
#include <WiFiClient.h>

PZEM004Tv30 pzem1(&Serial,0x01);

ESP8266WebServer httpServer(80);       // Create a webserver for the updates
ESP8266HTTPUpdateServer httpUpdater;   // and the webserver update module

WiFiClient nodeEV20;                     // wifi client name
PubSubClient nodeEVMQTT(nodeEV20);

const char* mqttServer = "192.168.1.143";
const int mqttPort = 1883;
const char* mqttUser = "mqttusername";
const char* mqttPassword = "mqttpassword";

HTTPClient http;  

elapsedMillis timeElapsed;
const unsigned int  interval = 10000;           // interval of reading (10sec)
unsigned long previousMillis=0;         // used to control the send of data to the server
unsigned long currentMillis;            // in constant time periods (10s)

//WIFI credentials go here
const char* ssid     = "MYSSID";
const char* password = "MYPASS";
const char* host="nodemeter-update";                    // name for the webupdate server

unsigned long lastMsg = 0;
#define MSG_BUFFER_SIZE  (50)
char msg[MSG_BUFFER_SIZE];
int value = 0;

char i_str[10]; 
char v_str[10];
char p_str[10];
char e_str[10];

String getData, link;

float v1,i1,p1,phi1,e1;                            // used to store the data from the PZEM
WiFiServer server(80);

void startWIFI(void) {
    // Connect to WiFi network
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
    }
    delay(300);
}

void setup() {  
      //Serial.begin(115200);
      startWIFI();
      timeElapsed = 0;

       nodeEVMQTT.setServer(mqttServer, mqttPort);

       while (!nodeEVMQTT.connected()) {
            if (nodeEVMQTT.connect("NodeHome", mqttUser, mqttPassword )) {
            } else {
              delay(1000);
            }
         }
      // start the webupdate service!
      MDNS.begin(host);
      httpUpdater.setup(&httpServer);
      httpServer.begin();
      MDNS.addService("http", "tcp", 80);
}

void loop() {

  nodeEVMQTT.loop(); 
  // In case the connection to the server is lost
  if(!nodeEVMQTT.connected()){ 
         nodeEVMQTT.connect("NodeHome", mqttUser, mqttPassword);
  }
    if (timeElapsed >= interval) {
              
              v1 = pzem1.voltage();
              i1 = pzem1.current();
              p1 = pzem1.power();
              phi1 = pzem1.pf();
              e1 = pzem1.energy();
              if (v1 < 0) v1 = 0.0;
              if (p1 <0 ) p1 = 0.0;
              if (i1 <0) i1 = 0.0;
              
              /*v1 = random(220,230);
              i1 = random(0,30);
              p1 = i1*v1;
              //delay(100);*/
              
                     //send data to server to insert in DB
                     getData = "?local=GERAL&amps=" + String (i1) + "&volts=" + String (v1) + "&watts=" + String (p1); //construo o link a enviar
                     link = "http://192.168.1.105/consumo.php" + getData;
                     http.begin(link);     //Specify request destination
                     //http.begin(nodeEV20,link);     //Specify request destination
                     int httpCode = http.GET();
                     http.end();
              
           dtostrf(i1,2,1,i_str);
           dtostrf(v1,2,1,v_str);
           dtostrf(p1,2,1,p_str);
           dtostrf(e1,2,1,e_str);
           
           nodeEVMQTT.publish("/HomeAmps", i_str);
           nodeEVMQTT.publish("/HomeVolts",v_str);
           nodeEVMQTT.publish("/HomeWatts",p_str);
           nodeEVMQTT.publish("/HomeWh",e_str);

          timeElapsed = 0;
            
    }
    httpServer.handleClient();            
    delay(1000); 
}

d-a-v, 2022-01-10:
```cpp
WiFiClient nodeEV20;                     // wifi client name
PubSubClient nodeEVMQTT(nodeEV20);
```
Use a separate WiFiClient for mqtt and http client.

```cpp
    delay(1000); 
```
This is forbidden (counter-productive) in a network loop.