esp8266/Arduino #8452: default implementation for buffered Stream.read only reads every second byte

INemesisI, 2022-01-18:
The current Stream::read (uint8_t* buffer, size_t maxLen) provides a standard implementation for writing a stream to a buffer.
It uses the virtual int method Stream::read() to read the stream byte by byte until no more bytes are available.

However, it calls read() twice in each loop. This results in only every 2nd byte actually being read into the destination buffer, as the read() method will always return the next byte in the stream.
At least that is my understanding of how the read() method of streams is supposed to work.

The lines of code involved:
https://github.com/esp8266/Arduino/blob/9fcf14f81fa9be589530e9596b7c5a264dc81ee8/cores/esp8266/Stream.cpp#L267-L279

d-a-v, 2022-01-18:
Nice catch !
We didn't see it because this function is often overridden in subclasses.
Would you like to submit a pull request for fixing this ?

INemesisI, 2022-01-18:
That was also my first guess, that it is probably overloaded everywhere, in favour of a faster version.
The PR is submitted!

d-a-v, 2022-01-18:
Thanks !
You may edit your *first* post in #8453 and add a new line "fixes #8452"

INemesisI, 2022-01-18:
Like this?

d-a-v, 2022-01-18:
> Like this?

Yes. Doing this adds a reference to the issue (look to the right column on this page) and automatically closes the issue (this post) when the PR is merged.

PR #8453 (2022-01-18): fix Stream.read into buffer ignoring every second byte 
As shown in #8452, the default implementation for `int Stream::read (uint8_t* buffer, size_t maxLen)` only writes every second byte into the given buffer.

fixes #8452