esp8266/Arduino #8475: All Wemos D1 mini code stopped working after Boards upgrade from 2.5.2 to 3.0.2

TSBrownie, 2022-02-02:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ?] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [ x] If there is a stack dump, I have decoded it.
- [ x] I have filled out all fields below.

#### Platform
- Hardware:			Wemos D1 mini  (reports: ESP8266EX) w/ RTC Shield/SD
- Core Version:			[latest git hash or date]
- Development Env:		1.8.16
- Operating System:		Win 10

### Settings in IDE

- Module:  			Wemos D1 mini ESP8266 3.0.2 (from 2.5.2)
- Flash Mode:			qio
- Flash Size: 		        4MB
- lwip Variant:                   ? 
- Reset Method:		nodemcu, hard, power
- Flash Frequency:	        ?? Mhz
- CPU Frequency:		Crystal is 26MHz
- Upload Using:		SERIAL
- Upload Speed:               460800 (auto set)

### Problem Description
Working code (plus 10 past versions) stopped working immediately after updating from Boards 2.5.2 to 3.0.2.

Output before:
========= 20220201,Tue,07:54:28
Device Altitude(abs): 6.00 meters AMSL
Temperature: 28.47C; 83.25F  Falling
Barometric Pressure(abs): Rising-Fair Weather
  955.597 hPa(mbar); 95559.720 Pa
  950.916 hPa, Short Term Past Average 
  716.757 mmHg
  0.943101 Atm 
Barometric Pressure-Comp'ed To Sealevel
  956.247 hPa(mbar); 95624.692 Pa
  717.244 mmHg
  0.943742 Atm 

Output after:
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)
wdt reset
load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00055920
~ld

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

One sample of previously working code here (compile info below):
https://github.com/TSBrownie/Arduino-Wemos-D1-Weather-Basic/tree/main/2022WeatherBMP280-04x

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Executable segment sizes:
ICACHE : 32768           - flash instruction cache 
IROM   : 247428          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 27369   / 32768 - code in IRAM          (IRAM_ATTR, ISRs...) 
DATA   : 1520  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 1152  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 26216 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 277469 bytes (26%) of program storage space. Maximum is 1044464 bytes.
Global variables use 28888 bytes (35%) of dynamic memory, leaving 53032 bytes for local variables. Maximum is 81920 bytes.
esptool.py v3.0
Serial port COM4
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: e8:db:84:dc:c5:06
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 281616 bytes to 205845...
Writing at 0x00000000... (7 %)
Writing at 0x00004000... (15 %)
Writing at 0x00008000... (23 %)
Writing at 0x0000c000... (30 %)
Writing at 0x00010000... (38 %)
Writing at 0x00014000... (46 %)
Writing at 0x00018000... (53 %)
Writing at 0x0001c000... (61 %)
Writing at 0x00020000... (69 %)
Writing at 0x00024000... (76 %)
Writing at 0x00028000... (84 %)
Writing at 0x0002c000... (92 %)
Writing at 0x00030000... (100 %)
Wrote 281616 bytes (205845 compressed) at 0x00000000 in 4.6 seconds (effective 488.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

```

TSBrownie, 2022-02-02:
Another version of the same weather station code that was working, now does this:
```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------
Exception (28):
epc1=0x40211734 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
>>>stack>>>
ctx: cont
sp: 3ffffd20 end: 3fffffc0 offset: 0190
3ffffeb0:  00000000 00000000 3ffeebd8 3fff0738  
3ffffec0:  3fff072c 00000003 3ffeea34 40201709  
3ffffed0:  00000000 00000000 3ffffef0 3ffeea9c  
3ffffee0:  3fffff08 3ffeea34 3fffff30 4021184c  
3ffffef0:  40210fa4 00000000 000003e8 40100cd4  
3fffff00:  00000000 00000000 00000000 40204d4c  
3fffff10:  00000000 00000000 3ffeebd8 4020c8d5  
3fffff20:  4020fddc 4020fdc8 3ffee9d8 40201860  
3fffff30:  3fff05b4 000c000f 80000000 4020c9d1  
3fffff40:  4020fddc 4020fdc8 3ffee9d8 40203d00  
3fffff50:  3fff069c 0037003f 80000000 00000000  
3fffff60:  40203488 00000000 4020fddc 4020fdc8  
3fffff70:  40203488 08080808 feefeffe feefeffe  
3fffff80:  00000000 feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3ffeeedc  
3fffffa0:  3fffdad0 00000000 3ffeeec8 4020d9c4  
3fffffb0:  feefeffe feefeffe 3ffe8688 40100f2d  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

THIS IS THE CODE throwing the above errors. This was working archival code:
```
//Add led on / off during file update if file corruption is issue during eject.
//Arduino Wemos D1R1 mini w/ RTC, SD Shield (ESP8266 Boards 2.5.2), WiFi Viewing
//RTC, SD File: Absolute Barometric Pressure (SL & Comp), Temp, no humidity
//Exponential Moving Average EMA improvement
//Wemos SDA(D2)-->BMP280 SDA pin; Wemos SCL(D1)-->BMP280 SCL
//Wemos 3v3-->BMP280 Vdd; Wemos GND-->BMP280 GND
//UNUSED: Wemos Vdd or GND-->BMP280=SDO (Address control)
//UNUSED: Wemos Vdd or GND-->BMP280=CSB (I2C select, GND=SPI)
//BMP280 Data is LSBMSB ijklmnopabcdefgh order (not abcdefghijklmnop)
//WeMos Micro SD Shield (uses HSPI(12-15) not (5-8), 3V3, G)
//GPIO12(D6)=MISO (main in, secondary out); GPIO13(D7)=MOSI (main out, secondary in) 
//GPIO14(D5)=CLK (Clock); GPIO15(D8)=CS (Chip select)
//SD library-->8.3 filenames (ABCDEFGH.txt = abcdefgh.txt)
//RTC DS1307. I2C--> SCL(clk)=D1, SDA(data)=D2 (Shared with BMP280)
//20220122 - TSBrownie.  Non-commercial use.
#include <SD.h>                      //SD card library
#include <SPI.h>                     //COM Serial Peripheral Interface bus for COMM, SD com

//WiFi Serving Def ===============================
#include <ESP8266WiFi.h>             //WiFi library
#include <WiFiClient.h>              //WiFi Client library
#include <ESP8266WebServer.h>        //Web Server library
#include <DNSServer.h>               //Domain Name Server
#include <ESP8266mDNS.h>             //ESP8266 mDNS specific
#include <EEPROM.h>                  //ESP eeprom library
#define DBG_OUTPUT_PORT Serial       //Direct debug to Serial
//Set to desired softAP credentials. NOT configurable at runtime
#ifndef APSSID                       //Prevents incursive lib inclusions
#define APSSID "FreeLocalWeather"    //USER SUPPLIED: SSID
#define APPSK  ""                    //#define APPSK  "12345678"
#endif                               //

const char *softAP_ssid = APSSID;    //Soft access point ssid (def above)
const char *softAP_password = APPSK; //Soft access point password (def above)
const char *myHostname = "FreeInfo"; //Hostname for multicast DNS. Works on win. Try http://esp8266.local
char ssid[32] = "";                  //Don't set this credential. Configured at runtime & stored on EEPROM
char password[32] = "";              //Don't set this credential. Configured at runtime & stored on EEPROM
const byte DNS_PORT = 53;            //DNS server.  Port 53 for UDP activities & as server port for TCP
[... 378 lines cut ...]
    S_pln(outBuff);print4File(outBuff);
    outBuff = "  "+String(pressure * 0.000986923, 6)+" Atm ";//Print Pressure in atm
    S_pln(outBuff);print4File(outBuff);
    outBuff = "Barometric Pressure-Comp'ed To Sealevel";
    S_pln(outBuff);print4File(outBuff);
    outBuff = "  "+String(p0 * 100, 3)+" hPa(mbar); "+String(p0 * 10000, 3)+" Pa";//Comp'ed Pressure in Pa, mbar, hPa
    S_pln(outBuff);print4File(outBuff);
    outBuff = "  "+String(p0 * 075.0061683, 3)+" mmHg";       //Comp'ed Pressure in mmHg
    S_pln(outBuff);print4File(outBuff);
    outBuff = "  "+String(p0 * 000.0986923, 6)+" Atm ";      //Comp'ed Pressure in Atm
    S_pln(outBuff);print4File(outBuff);
   
    GetRTCTime();                                             //Get time from RTC
//    SDData = "D,"+String(Dh)+','+timeString+','+
//      String(pressure)+','+String(p0*100)+','+String(cTemp);  //Ongoing SD weather data
//    print3File(SDData);                                       //Write weather data to SD

    outBuff = "D,"+String(Dh)+','+timeString+','+
      String(pressure)+','+String(p0*100)+','+String(cTemp);  //Ongoing SD weather data
    print3File(outBuff);                                       //Write weather data to SD

  }                                                           //End: if(millis() >= time_last + time2update)
}                                                             //End: Loop()

```

d-a-v, 2022-02-02:
Thanks for the report.
[Please use the exception decoder for your stack trace above](https://arduino-esp8266.readthedocs.io/en/3.0.2/faq/a02-my-esp-crashes.html#exception-decoder).

TSBrownie, 2022-02-02:
I used github's code sample. Ran it. It blew up.
Copied the dump (below), but 
IDE 1.8.16 does not have the ESP exception decoder.

```
ctx: cont
sp: 3ffffdf0 end: 3fffffc0 offset: 01a0
3fffff90:  3fffdad0 00000000 3ffee4f8 40201050  
3fffffa0:  feefeffe feefeffe 3ffee54c 40201894  
3fffffb0:  feefeffe feefeffe 3ffe85d8 40100b51  
```

I have been testing all my Wemos code from simple to 500 line programs
and none of it works; most of it produces chksums or such, some of it 
produces Serial COM output, but anything that's not just text (ex: numbers)
 is either junk, or wrong (probably from memory issues).

I used to write code for GE/Honeywell mainframes in the late 1970s, and 
that's the last time I dug through a dump. Now I'm coding for fun, 
but when all your code goes bad because of a board library change, that
sucks. Any suggestions on low power boards with solid support? (And RTCs, SD
cards, etc?)
Thanks.

Bighoneypot, 2022-02-02:
@TSBrownie 

Try this tool

https://www.filemail.com/d/nzlpfgbjpananhv

TSBrownie, 2022-02-02:
> > but  IDE 1.8.16 does not have the ESP exception decoder.
> 
> You must install it as instructed in the documentation. No one except you can decode the stack trace.
> 
> > I have been testing all my Wemos code from simple to 500 line programs
> > and none of it works
> 
> Did you try the [erase "All flash" arduino IDE option](https://arduino-esp8266.readthedocs.io/en/3.0.2/ideoptions.html#erase-flash) ?
Yes, I tried all 3 options. No difference.

EXCEPTION DECODER:
Decoding stack results
0x40201050: setup() at D:\Prgms\Arduino\ArduinoWemosD1\2022GithubExceptDecode01a/2022GithubExceptDecode01a.ino line 22
0x40201894: loop_wrapper() at C:\Users\Family\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 198

--------------------------------------
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffdf0 end: 3fffffc0 offset: 01a0
3fffff90:  3fffdad0 00000000 3ffee4f8 40201050  
3fffffa0:  feefeffe feefeffe 3ffee54c 40201894  
3fffffb0:  feefeffe feefeffe 3ffe85d8 40100b51  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

TSBrownie, 2022-02-02:
> @TSBrownie
> 
> Try this tool
> 
> https://www.filemail.com/d/nzlpfgbjpananhv
Your user name is "Bighoneypot" and you are offering a link to an unknown source? 
Hmmmmmmm.

Bighoneypot, 2022-02-02:
@TSBrownie 
You're nice ;-)

My drive
[https://drive.google.com/drive/folders/1D1lp4OoMuAaTxWt1QXtWsm0RYKwtgMQQ?usp=sharing](url)

Try it!

xdubx, 2022-02-05:
Hi i updated my projects too. After the breaking change that http.begin() is deprecated with one parameter. I updated from esp8266/2.7.4 to 3.0.2. Now my projects breaks. The possible solution is don't use any Hardware pin when you send a request. But its not very handy for the code.

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4021871c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000264 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffae0 end: 3fffffc0 offset: 0190
3ffffc70:  00000000 00000000 4bc6a7f0 00000000  
3ffffc80:  3fff0404 00000000 40100501 00000000  
3ffffc90:  0000470f 3fff16a8 00001a6b 00000000  
3ffffca0:  000000c4 000000c4 3ffe85e4 ffffffff  
3ffffcb0:  0000470f 3ffffe68 3fff0ec4 40203220  
3ffffcc0:  0000012c 00000020 3fff16a4 00000000  
3ffffcd0:  3ffffe28 00000020 3fff0ec4 00000000  
3ffffce0:  3ffffe28 3ffffe68 00000001 402036b1  
3ffffcf0:  00000274 00000274 3ffe85e4 402036e5  
3ffffd00:  3ffffe28 3ffffd30 3ffffe30 4020370c  
3ffffd10:  3ffffe28 3ffffef0 3ffffe30 40206b5c  
3ffffd20:  3ffe88a1 3ffffef0 3ffffe30 40202024  
3ffffd30:  3ffffe68 3fff0f3c 0016001f 80000000  
3ffffd40:  00010050 00001388 3fff0f64 0051005f  
3ffffd50:  80000000 70747468 00000000 04000000  
3ffffd60:  3fff17d4 0000001f 80000000 3fff0e44  
3ffffd70:  0011001f 80000000 00000000 00000023  
3ffffd80:  00000000 00000000 00000000 00000000  
3ffffd90:  ffffffff 00000001 00000000 0000000a  
3ffffda0:  00000000 00000023 00000000 00000000  
3ffffdb0:  00000000 3ffffe48 3fff1d74 3ffffe48  
3ffffdc0:  3ffe8899 00000000 3ffe88a6 3fff1a0a  
3ffffdd0:  0000003a 3ffffe48 30303633 32383700  
3ffffde0:  00003534 00000005 3ffffe54 4020afbc  
3ffffdf0:  3ffffe55 7fc40000 00000050 00000023  
3ffffe00:  00000000 00000002 3ffffe54 0000000a  
3ffffe10:  40207be8 3ffffe50 00000004 00000000  
3ffffe20:  000d000f 00000000 00000000 40007364  
3ffffe30:  00000000 4023d70a c02d70a4 00000007  
3ffffe40:  00000052 3ffefd70 3fff1c7c 3fff1d74  
3ffffe50:  3fff1d80 3fff1fa0 00000001 3fff1f90  
3ffffe60:  3fff1d80 3ffe8c20 4020d7d8 00000000  
3ffffe70:  00001388 000046ec 00000000 00000000  
3ffffe80:  3fff0ec4 00000000 00000000 020e020f  
3ffffe90:  00000000 3fff12a4 006e006f 80000000  
3ffffea0:  3fff1f30 3ffffe48 3fff1f20 3ffffe48  
3ffffeb0:  00000000 00000000 00000190 3fffff40  
3ffffec0:  40207be8 3ffe8c7a 3ffefefc 3fff0040  
3ffffed0:  3fffdad0 3fff113c 3ffefd84 4020211c  
3ffffee0:  3ffefd9a 00000039 3ffefefc 40207f38  
3ffffef0:  3fff131c 000d000f 80000000 756f6c43  
3fffff00:  40007364 06000000 4023d70a c02d70a4  
3fffff10:  00000007 00000052 3fffff80 4020133e  
3fffff20:  00000000 00000001 00000000 009c1001  
3fffff30:  000000ad 00000001 00000000 3fff0040  
3fffff40:  3fffdad0 00000000 3ffefed2 40205cdd  
3fffff50:  3fffdad0 3ffee998 3ffefd54 40205461  
3fffff60:  3fffdad0 3ffefefc 3ffefc98 3fff0040  
3fffff70:  3fffdad0 3ffefefc 3ffefd84 4020264e  
3fffff80:  00000000 0014001f 00000000 00000000  
3fffff90:  0011001f 00000000 feefeffe feefeffe  
3fffffa0:  feefeffe 00000000 3fff002c 402097fc  
3fffffb0:  feefeffe feefeffe 3ffe85e0 40100f0d  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```
and decoded 

```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x4021871c: tcp_output at core/tcp_out.c line 1264
EXCVADDR: 0x00000264

Decoding stack results
0x40100501: millis() at /home/hp/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/core_esp8266_wiring.cpp line 193
0x40203220: ClientContext::wait_until_acked(int) at /home/hp/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/include/ClientContext.h line 347
0x402036b1: WiFiClient::flush(unsigned int) at /home/hp/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/WiFiClient.cpp line 311
0x402036e5: WiFiClient::stop(unsigned int) at /home/hp/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/WiFiClient.cpp line 319
0x4020370c: WiFiClient::stop() at /home/hp/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/WiFiClient.h line 75
0x40206b5c: HTTPClient::~HTTPClient() at /home/hp/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 60
0x40202024: getWeather() at /home/hp/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/WString.h line 79
0x4020afbc: uart_write(uart_t*, char const*, size_t) at /home/hp/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/uart.cpp line 546
0x40207be8: HardwareSerial::write(unsigned char const*, unsigned int) at /home/hp/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/HardwareSerial.h line 193
0x40207be8: HardwareSerial::write(unsigned char const*, unsigned int) at /home/hp/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/HardwareSerial.h line 193
0x4020211c: drawInterface() at /media/hp/Data/Repo/home-station/main/main.ino line 386
```

TSBrownie, 2022-02-07:
This is a video of Setting the Real Time Clock (code from above) showing it worked when it was compiled using Board 2.5.2.
https://www.youtube.com/watch?v=dAs0Xzp2GdQ

mcspr, 2022-02-07:
Node that the code from https://github.com/esp8266/Arduino/issues/8475#issuecomment-1027706966 cannot be built using 3.0.2 *with* warnings enabled

```
/home/builder/dev/arduino8475/src/main.ino: In function 'void avgPTCalc()':
/home/builder/dev/arduino8475/src/main.ino:92:17: warning: comparison of integer expressions of different signedness: 'int' and 'const unsigned int' [-Wsign-compare]
   92 |   for(int k=1; k<=avgPTsz; k++){      //Loop thru avg readings
      |                ~^~~~~~~~~
/home/builder/dev/arduino8475/src/main.ino: In function 'char openFile3(byte, String)':
/home/builder/dev/arduino8475/src/main.ino:167:33: error: no return statement in function returning non-void [-Werror=return-type]
  167 |   saveWeather = SD.open(FN, RW);}        //Open Read at end.  Open at EOF for write/append
      |                                 ^
/home/builder/dev/arduino8475/src/main.ino: In function 'char openFile4(byte, String)':
/home/builder/dev/arduino8475/src/main.ino:173:3: error: no return statement in function returning non-void [-Werror=return-type]
  173 |   }
      |
``` 

I suppose that is b/c you are using 'no-warnings' option?
Can you retry by either using `void` as return type for `openFile3` and `openFile4` or add `return` with some meaningful value?

TSBrownie, 2022-02-07:
> Note that the code from [#8475 (comment)](https://github.com/esp8266/Arduino/issues/8475#issuecomment-1027706966) cannot be built using 3.0.2 _with_ warnings enabled
> 
> ```
> /home/builder/dev/arduino8475/src/main.ino: In function 'void avgPTCalc()':
> /home/builder/dev/arduino8475/src/main.ino:92:17: warning: comparison of integer expressions of different signedness: 'int' and 'const unsigned int' [-Wsign-compare]
>    92 |   for(int k=1; k<=avgPTsz; k++){      //Loop thru avg readings
>       |                ~^~~~~~~~~
> /home/builder/dev/arduino8475/src/main.ino: In function 'char openFile3(byte, String)':
> /home/builder/dev/arduino8475/src/main.ino:167:33: error: no return statement in function returning non-void [-Werror=return-type]
>   167 |   saveWeather = SD.open(FN, RW);}        //Open Read at end.  Open at EOF for write/append
>       |                                 ^
> /home/builder/dev/arduino8475/src/main.ino: In function 'char openFile4(byte, String)':
> /home/builder/dev/arduino8475/src/main.ino:173:3: error: no return statement in function returning non-void [-Werror=return-type]
>   173 |   }
>       |
> ```
> 
> I suppose that is b/c you are using 'no-warnings' option? Can you retry by either using `void` as return type for `openFile3` and `openFile4` or add `return` with some meaningful value?

Do you mean Tools -->Debug Level: None??? or Exceptions: Disabled???

TSBrownie, 2022-02-08:
Uninstalled IDE 1.8.16, installed 1.8.19, and esp8266 by ESP8266 Community. Tried versions below on the weather program above (closed IDE between each upgrade + "Erase all flash content" + hardware reset of Wemos). 
2.5.2 --> Compiles. Runs correctly.
2.6.2 --> Compiles. Runs correctly.
2.7.2 --> Compiles. Runs correctly.
3.0.0 --> Compiles, but gives similar runtime error as above
3.0.1 --> Compiles, but gives similar runtime error as above
3.0.2 --> Compiles, but gives the runtime errors shown above

Then I tried a simple program that reads the Wemos D1 mini SD card/RTC shield's clock (per above).
2.5.2 --> Compiles. Runs correctly.
2.6.2 --> Compiles. Runs correctly.
2.7.2 --> Compiles. Runs correctly.
3.0.0 --> Compiles, Runs correctly.
3.0.1 --> Compiles, Runs correctly.
3.0.2 --> Compiles, Runs correctly but prints out the following line mid-output
"SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635"

TSBrownie, 2022-02-08:
> Do you mean Tools -->Debug Level: None??? or Exceptions: Disabled???

I did one better and integrated those into the only routine that called them. (The program is 4 programs I cobbled together, so I'm now removing redundancy and such.) The results are the same.

`//SD CARD FUNCTIONS =================================
void openSD() {                          //Routine to open SD card
  S_pln(); S_pln("Open SD card");        //User message
  if (!SD.begin(15)) {                   //If not open, print message.
    S_pln("Open SD card failed");
    return;}
  S_pln("SD Card open");
}

void print2File(String B){               //Write connect data to SD file
  SDconFile.close();                     //Ensure file status, before re-opening
  SDconFile = SD.open(FName2, FILE_WRITE); //Open Read at end.  Open at EOF for write/append
  if (SDconFile) {                       //If file there & opened --> write
    SDconFile.println(B);                //Write to SD
    SDconFile.close();}                   //Close file, flush buffer (reliable but slow)
  else {S_pln("Error opening Connect file for write");}  //File didn't open
}

void print3File(String B) {              //Print data to SD file
  saveWeather.close();                   //Ensure file status, before re-opening
  saveWeather = SD.open(FName3, FILE_WRITE);  //Open Read at end.  Open at EOF for write/append
  if (saveWeather) {                     //If file there & opened --> write
    saveWeather.println(B);              //Print string to file
    saveWeather.close();                 //Close file, flush buffer (reliable but slower)
  } else {S_pln("Error opening saveWeather file for write");}   //File didn't open
}

char openFile4(byte RW, String FN) {     //Open SD FName3 file. char RW. Only 1 at a time.
  dataTemp.close();                      //Ensure file status, before re-opening
  dataTemp = SD.open(FN, RW);            //Open Read at end.  Open at EOF for write/append
//  digitalWrite(LED_BUILTIN, LOW);        //D1 Mini: turns LED *on*. Eject SD if off
}

void print4File(String B) {              //Print data to SD file
  openFile4(FILE_WRITE,FName4);          //Open user SD file for write
  if (dataTemp) {                        //If file there & opened --> write
    dataTemp.println(B);                 //Print string to file
    dataTemp.close();                    //Close file, flush buffer (reliable but slower)
//    digitalWrite(LED_BUILTIN, HIGH);     //D1 Mini: turns LED *on*.Eject SD if off
  } else {S_pln("Error opening dataTemp file for write");}   //File didn't open
}

void initTempFile(){                     //Initialize SD card connect file
  SDconFile.close();                     //Ensure file is closed         
  SD.remove(FName4);                     //Delete connection info file
  SDconFile.close();                     //Ensure file status, before re-opening
  SDconFile = SD.open(FName4, FILE_WRITE); //Open Read at end.  Open at EOF for write/append
  SDconFile.close();                     //Close SD connect file
//  server.send(200, "text/html", "Connect Data File Initialized");
}`

mcspr, 2022-02-08:
My mistake, I was referring to the Arduino IDE preferences 'Compiler warnings'
(see File > Preferences, at least that's what it is called on the 2.0.0rc version I have locally. PIO has slightly different naming)

The code **cannot** omit return statement if the function was declared as something other than `void`, openfile4 function above still does not have it. We can't exactly force this setting though, due to the fact `-w` overrides the `-Werror=...`

```cpp
int something() { 
}

void setup() {
  something();
}

void loop() {
}
```
```
C:\Users\test\AppData\Local\Temp\.arduinoIDE-unsaved202218-9456-1ebm4ou.xdoe\sketch_feb8a\sketch_feb8a.ino: In function 'int something()':
C:\Users\test\AppData\Local\Temp\.arduinoIDE-unsaved202218-9456-1ebm4ou.xdoe\sketch_feb8a\sketch_feb8a.ino:8:1: error: no return statement in function returning non-void [-Werror=return-type]
    2 | }
      | ^
```

mcspr, 2022-02-22:
Assuming this was eventually solved
#8492 adds documentation entry in the crash FAQ about compiler warnings and their possible impact on the resulting program

closed 2022-02-22 by mcspr

TSBrownie, 2022-02-22:
No, it was not resolved. I went back to pre-3.xx board manager. 3.xx still does not work.

TSBrownie, 2022-02-22:
No, not resolved. 3.xx board manger does not work. Still waiting, still using version 2.7.3