esp8266/Arduino #8479: HTTPClient client crashed with parallel hardware serial using

xdubx, 2022-02-05:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266
- Core Version:			3.0.2
- Development Env:		Arduino IDE or VSC
- Operating System:		[Windows|Linux]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v3.0]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description
After you send a request with the wifiClient the esp crashes with the error below. I tried out with 2 separat projects. I updated from 2.7.4 to 3.0.2 and the error appears. One project was with UART, and with this example with SPI.

```cpp

    HTTPClient http;
    WiFiClient wifiClient;
    http.begin(wifiClient, url);

```

### Debug Messages

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4021871c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000264 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffae0 end: 3fffffc0 offset: 0190
3ffffc70:  00000000 00000000 4bc6a7f0 00000000  
3ffffc80:  3fff0404 00000000 40100501 00000000  
3ffffc90:  0000470f 3fff16a8 00001a6b 00000000  
3ffffca0:  000000c4 000000c4 3ffe85e4 ffffffff  
3ffffcb0:  0000470f 3ffffe68 3fff0ec4 40203220  
3ffffcc0:  0000012c 00000020 3fff16a4 00000000  
3ffffcd0:  3ffffe28 00000020 3fff0ec4 00000000  
3ffffce0:  3ffffe28 3ffffe68 00000001 402036b1  
3ffffcf0:  00000274 00000274 3ffe85e4 402036e5  
3ffffd00:  3ffffe28 3ffffd30 3ffffe30 4020370c  
3ffffd10:  3ffffe28 3ffffef0 3ffffe30 40206b5c  
3ffffd20:  3ffe88a1 3ffffef0 3ffffe30 40202024  
3ffffd30:  3ffffe68 3fff0f3c 0016001f 80000000  
3ffffd40:  00010050 00001388 3fff0f64 0051005f  
3ffffd50:  80000000 70747468 00000000 04000000  
3ffffd60:  3fff17d4 0000001f 80000000 3fff0e44  
3ffffd70:  0011001f 80000000 00000000 00000023  
3ffffd80:  00000000 00000000 00000000 00000000  
3ffffd90:  ffffffff 00000001 00000000 0000000a  
3ffffda0:  00000000 00000023 00000000 00000000  
3ffffdb0:  00000000 3ffffe48 3fff1d74 3ffffe48  
3ffffdc0:  3ffe8899 00000000 3ffe88a6 3fff1a0a  
3ffffdd0:  0000003a 3ffffe48 30303633 32383700  
3ffffde0:  00003534 00000005 3ffffe54 4020afbc  
3ffffdf0:  3ffffe55 7fc40000 00000050 00000023  
3ffffe00:  00000000 00000002 3ffffe54 0000000a  
3ffffe10:  40207be8 3ffffe50 00000004 00000000  
3ffffe20:  000d000f 00000000 00000000 40007364  
3ffffe30:  00000000 4023d70a c02d70a4 00000007  
3ffffe40:  00000052 3ffefd70 3fff1c7c 3fff1d74  
3ffffe50:  3fff1d80 3fff1fa0 00000001 3fff1f90  
3ffffe60:  3fff1d80 3ffe8c20 4020d7d8 00000000  
3ffffe70:  00001388 000046ec 00000000 00000000  
3ffffe80:  3fff0ec4 00000000 00000000 020e020f  
3ffffe90:  00000000 3fff12a4 006e006f 80000000  
3ffffea0:  3fff1f30 3ffffe48 3fff1f20 3ffffe48  
3ffffeb0:  00000000 00000000 00000190 3fffff40  
3ffffec0:  40207be8 3ffe8c7a 3ffefefc 3fff0040  
3ffffed0:  3fffdad0 3fff113c 3ffefd84 4020211c  
3ffffee0:  3ffefd9a 00000039 3ffefefc 40207f38  
3ffffef0:  3fff131c 000d000f 80000000 756f6c43  
3fffff00:  40007364 06000000 4023d70a c02d70a4  
3fffff10:  00000007 00000052 3fffff80 4020133e  
3fffff20:  00000000 00000001 00000000 009c1001  
3fffff30:  000000ad 00000001 00000000 3fff0040  
3fffff40:  3fffdad0 00000000 3ffefed2 40205cdd  
3fffff50:  3fffdad0 3ffee998 3ffefd54 40205461  
3fffff60:  3fffdad0 3ffefefc 3ffefc98 3fff0040  
3fffff70:  3fffdad0 3ffefefc 3ffefd84 4020264e  
3fffff80:  00000000 0014001f 00000000 00000000  
3fffff90:  0011001f 00000000 feefeffe feefeffe  
3fffffa0:  feefeffe 00000000 3fff002c 402097fc  
3fffffb0:  feefeffe feefeffe 3ffe85e0 40100f0d  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```
and decoded 

```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x4021871c: tcp_output at core/tcp_out.c line 1264
EXCVADDR: 0x00000264

Decoding stack results
0x40100501: millis() at /home/hp/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/core_esp8266_wiring.cpp line 193
0x40203220: ClientContext::wait_until_acked(int) at /home/hp/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/include/ClientContext.h line 347
0x402036b1: WiFiClient::flush(unsigned int) at /home/hp/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/WiFiClient.cpp line 311
0x402036e5: WiFiClient::stop(unsigned int) at /home/hp/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/WiFiClient.cpp line 319
0x4020370c: WiFiClient::stop() at /home/hp/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/WiFiClient.h line 75
0x40206b5c: HTTPClient::~HTTPClient() at /home/hp/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 60
0x40202024: getWeather() at /home/hp/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/WString.h line 79
0x4020afbc: uart_write(uart_t*, char const*, size_t) at /home/hp/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/uart.cpp line 546
0x40207be8: HardwareSerial::write(unsigned char const*, unsigned int) at /home/hp/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/HardwareSerial.h line 193
0x40207be8: HardwareSerial::write(unsigned char const*, unsigned int) at /home/hp/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/HardwareSerial.h line 193
0x4020211c: drawInterface() at /media/hp/Data/Repo/home-station/main/main.ino line 386
```

d-a-v, 2022-02-06:
```
    HTTPClient http;
    WiFiClient wifiClient;
    http.begin(wifiClient, url);
```
 Try with
```
    WiFiClient wifiClient;
    HTTPClient http;
    http.begin(wifiClient, url);
```

xdubx, 2022-02-06:
> ```
>     HTTPClient http;
>     WiFiClient wifiClient;
>     http.begin(wifiClient, url);
> ```
> 
> Try with
> 
> ```
>     WiFiClient wifiClient;
>     HTTPClient http;
>     http.begin(wifiClient, url);
> ```

It works. Thx. So simple fix. uff....

closed 2022-02-06 by xdubx

zellsito, 2022-07-27:
> ```
>     HTTPClient http;
>     WiFiClient wifiClient;
>     http.begin(wifiClient, url);
> ```
> 
> Try with
> 
> ```
>     WiFiClient wifiClient;
>     HTTPClient http;
>     http.begin(wifiClient, url);
> ```

it works for me too! thanks