esp8266/Arduino #8480: Forced light sleep being locked in delay function

rafaelgrosko, 2022-02-07:
I need the ESP to sleep and be able to wake up by both the GPIO and the timer, follow the functions below:

void esp8266_disable_wifi()
{
  WiFi.enableAP(false);
  WiFi.mode(WIFI_OFF);
  WiFi.disconnect(true);
  WiFi.forceSleepBegin();
  wifi_set_opmode(NULL_MODE);
}

void esp8266_light_sleep(int wakeup_pin, long time_sleep){
  long sleepTimeMilliSeconds = time_sleep * 1e3;

  extern os_timer_t *timer_list;
  timer_list = nullptr;
  Serial.print("system time start Sleep"); Serial.println(RTCmillis());

  wifi_fpm_set_sleep_type(LIGHT_SLEEP_T);
  wifi_fpm_set_wakeup_cb(wakeupCallback);
  gpio_pin_wakeup_enable(GPIO_ID_PIN(wakeup_pin), GPIO_PIN_INTR_HILEVEL);
  wifi_fpm_open();
  wifi_fpm_do_sleep(sleepTimeMilliSeconds * 1000);
  delay(sleepTimeMilliSeconds + 1);
}

ESP goes into forced light sleep mode, but when the time defined in wifi_fpm_do_sleep ends, the esp is stopped in delay(sleepTimeMilliSeconds + 1) consuming 60mA

Another problem I found is that esp only wakes up with the timer after the first interruption by gpio, is this correct?

maniacMatt, 2022-02-12:
I encountered the same issue with your code.  I've modified it to allow the interrupt to occur at any time without the need for a GPIO event first.  
`#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

extern "C" {
  #include "gpio.h"
}

extern "C" {
  #include "user_interface.h"
}

uint32_t RTCmillis() {
  // system_get_rtc_time() is in us (but very inaccurate anyway)
  return (system_get_rtc_time() * (system_rtc_clock_cali_proc() >> 12)) / 1000;
}

void esp8266_light_sleep(int wakeup_pin, long time_sleep)
{
  long sleepTimeMilliSeconds = time_sleep * 1e3;
  
  extern os_timer_t *timer_list;
  timer_list = nullptr;
  Serial.print("system time start Sleep"); Serial.println(RTCmillis());
  
  wifi_fpm_set_sleep_type(LIGHT_SLEEP_T);
  wifi_fpm_set_wakeup_cb(wakeup_cb);
  gpio_pin_wakeup_enable(GPIO_ID_PIN(wakeup_pin), GPIO_PIN_INTR_LOLEVEL);
  wifi_fpm_open();
  wifi_fpm_do_sleep(sleepTimeMilliSeconds * 1000);
  delay(sleepTimeMilliSeconds + 1);
}

void wakeup_cb() {
  Serial.println(RTCmillis());
  if (!digitalRead(0))
      Serial.println("tip event!");
    else
      Serial.println("time event!");
  Serial.flush();
}

void setup()
{
   /* connect to WiFi/normal setup stuff */
  Serial.begin(115200);
  gpio_init(); // Initialize GPIO pins
  wifi_fpm_set_sleep_type(LIGHT_SLEEP_T); // set sleep type, the above posters wifi_set_sleep_type() didnt seem to work for me although it did let me compile and upload with no errors 
}

void loop()
{
    wifi_fpm_close();  
    delay(150);
    Serial.println("hi,again!");
     /* do whatever you want */
    
    esp8266_light_sleep(0,10);
 }

`

dok-net, 2022-02-13:
There's a pending PR #7979, that solves issues with finding the right calls and their correct order by providing a cleanup and tested API.
Please look at [libraries/esp8266/examples/ForcedLightSleep/ForcedLightSleep.ino](https://github.com/esp8266/Arduino/pull/7979/files#diff-b7bddb696ab152d41ae05b21fc792ebabb2689455e852551726d290f29bca4fc) for your use case involving both timeout and GPIO-based wakeup.
I'm interested in any unclarities and suggestions for improvements.