esp8266/Arduino #8487: minor documentation improvement

petaflot, 2022-02-13:
on https://arduino-esp8266.readthedocs.io/en/latest/installing.html

```
cd esp8266/tools
python3 get.py
```

should be

```
cd tools
python3 get.py
```

.. because we already cd'ed into esp8266 before doing `git submodule update --init`

also, a direct link to the github in the left menu (ie. "Source code") would be appreciated

NaincyKumariKnoldus, 2022-02-13:
Hey @petaflot I have fixed this and raised a PR for this. 
Please review my PR.
Thanks!

petaflot, 2022-02-13:
looks good :-)

doesn't add the link to the sources on the github repos ; I find many projects lack a direct link and it takes a bit of trial-and-error and/or cross-linking sources when one doesn't want to post a bug report on a fork of the project (or an unrelated project with the same name , which does happen from time to time..)

petaflot, 2022-02-13:
on the left of the page lives a menu (CONTENTS: Installing/Arduino IDE options/Reference/...)  I am of the opinion that a link to this Github project would

1) not harm
2) be useful

I would call the link "Source code" or "Get the source" or something similar.

Thanks a lot for the quick response.

closed 2022-02-15 by earlephilhower