esp8266/Arduino #8497: How to install Sketch Data Upload

neo305, 2022-02-26:
Hi sorry I don't know where else to ask. I just got my pc back on fresh Windows 11 so have to install Arduino again and I forget how to add the option under Tools Tab to upload sketch data to esp8266. I found it and did it for ESP32 tho.

Bighoneypot, 2022-03-09:
@neo305 
Add this http://arduino.esp8266.com/stable/package_esp8266com_index.json in 
Preference -- Additional Boards

In your Arduino IDE

redPrint7, 2022-05-25:
If I understand correctly, you want to install the plugins for data-upload to ESP8266 and ESP32 in the ArduinoIDE which show under the tools menu. 

**Instruction**
This currently works only for the ArduinoIDE 1.X (current version 1.8.19), not for the new IDE2.x (current 2.0 RC6) -> Reason plugins currently not supported (Although there are strong wishes/pledges from the community).
There are different plugins (depending on board family and File Sytem used)

**Download-Links for plugins**
ESP8266 / SPIFFS: https://github.com/esp8266/arduino-esp8266fs-plugin/releases
ESP8266 / LittleFS: https://github.com/earlephilhower/arduino-esp8266littlefs-plugin/releases
ESP32 LittleFS / SpifFS / FatFS partion.csv Support: https://github.com/lorol/arduino-esp32fs-plugin
(Outdated) ESP32 FatFS: https://github.com/lorol/arduino-esp32fatfs-plugin

**Installation:**
Make sure you use one of the supported versions of Arduino IDE (1.8.x) and have ESP8266/ESP32 core installed.
Download espxxxxfs.zip zipped tool from [latest release]Links above
In your Arduino sketchbook directory, create tools directory if it doesn't exist yet.
Unpack the tool into the path you look up with "Setup->sketchbook location" \tools directory (C:\<my_development_dir>\Arduino). Example: (C:\\<my_development_dir>\Arduino)\tools\ESPxxxxFS\tool\espxxxxfs.jar 
Make sure you have mklittlefs[.exe] and mkfatfs[.exe] available at esp8266/esp32 core installation folder. Example for ESP8266: Look inside of C:\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs or in path from "Setup->sketchbook location" eg: (C:\\<my_development_dir>\Arduino)\hardware\esp8266\tools\mklittlefs
For MacOS /Linux adjust the path like instead of (C:\\<my_development_dir>\Arduino)\hardware\esp8266\tools\mklittlefs use eg (<my_development_dir>/Arduino)/hardware/esp8266/tools/mklittlefs

**Additional tip**
For easier debugging I recommend the plugin ESPExeptionDecoder: https://github.com/me-no-dev/EspExceptionDecoder
Follow the instructions given on the page.