esp8266/Arduino #8499: Making the Core ESP8266 Support pre-compiled libraries

InqOnThat, 2022-03-02:


mcspr, 2022-03-03:
Should already be fixed by https://github.com/esp8266/Arduino/pull/8392 for the (pending) 3.1.0
https://github.com/esp8266/Arduino/blob/ead5f94dd3b24229818876877a9591e9ec51ce7e/platform.txt#L67-L68

I don't believe .local.txt should not be provided by us
https://arduino.github.io/arduino-cli/0.21/platform-specification/#platformlocaltxt

closed 2022-03-03 by mcspr

InqOnThat, 2022-03-03:
Thank you for your quick response.

You are totally right about the second statement.  I forgot who was going
to receive the request... since you are the library provider, you are
supposed to put it in platform.txt as I see you have done in your first
statement.

Do you have an estimate when normal Arduino IDE users might see 3.1.0?