esp8266/Arduino #8509: Wrong value on EEPROM.Read since 3.0.0 core arduino version

miky-boy, 2022-03-13:
Hi guys,

I developed my code using ESP8266 Wemos D1 on Arduino IDE and i use the EEPROM to store my Wifi details and my sensors parameters.

On my last version developed under the core 2.7.4 the code below works perfectly, but today i decided to made some other changes on my program and upgraded also the core version from the esp8266 to the last 3.0.5 version and i realized a strange behaviour on my EEPROM readings. Seems is not reading correctly the value. I saw the byte is not existing anymore on this last core versions, but on this case i don't use it and i don't know what is wrong.

The code is the below one:

       case 208://IP-1
        {              
          String Conteudo_Memoria;
          Conteudo_Memoria = EEPROM.read(ADD_IP_1);//Obtém o valor como Integer  

          Serial.println("READ CASE:");
          Serial.println(ADD_IP_1);
          Serial.println(Conteudo_Memoria);
                       
          return Conteudo_Memoria;

Without any write change on EEPROM, if i compile and upload with the core 2.7.4 i have the following output:

READ CASE:
208
192 <- This value from EEPROM

But, if i compile and upload with the core 3.0.5 i have the following output:
READ CASE:
208
⸮ <- This value from EEPROM

Someone could gave an hand please?
Thanks and Regards,
Miky

d-a-v, 2022-03-13:
Try with `Conteudo_Memoria = (int)EEPROM.read(ADD_IP_1);//Obtém o valor como Integer `

miky-boy, 2022-03-14:
Hi d-a-v,
I tested your sugestion but still the same :(
Any other idea/reason for that?
Thanks and Regards

d-a-v, 2022-03-14:
Sorry, this is a known bug per #8430 which still needs to be fixed.
A workaround for both 3.0 and 2.x is
```
String Conteudo_Memoria(EEPROM.read(ADD_IP_1));//Obtém o valor como Integer  
```

miky-boy, 2022-03-14:
Hi d-a-v,
Thanks for your support. 

I really made a test now with int and worked. I will wait for the bug fix on the old version.

int test = EEPROM.read(ADD_IP_1);//Obtém o valor como Integer 
          Serial.println("READ CASE:");
          Serial.println(ADD_IP_1);
          Serial.println(test);

Thanks and Best Regards,
Miky

closed 2022-04-11 by mcspr