esp8266/Arduino #8515: analogWrite disables WDT (both soft and hard)

asturcon3, 2022-03-18:
### Basic Infos

- [X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ X] I have filled out all fields below.

#### Platform

- Hardware:					[new NodeMCU v3 ESP8266]
- Core Version:			[3.0.2]
- Development Env:		[Arduino IDE 1.8.19]
- Operating System:		[Windows 11]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[DOUT]
- Flash Size: 		[4MB (2+1)]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[dtr]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [921600]

### Problem Description

No watchdog when using analogWrite inside a infinite loop

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```
void setup() { ; }

void loop() {
	pinMode(2,OUTPUT);
	while (true) analogWrite(2,millis() % 0xff);
	//while (true) digitalWrite(2,millis() % 1);
}
```

As provided, the sketch runs forever.

Comment the analogWrite line, uncomment the digitalWrite one, and you get the expected behaviour (soft WDT resets the device)

mcspr, 2022-03-21:
Analog writes will call our internal task switch, wd timer gets fed and we never get it to fail.
(specifically, see _notifyPWM func and the instances of where esp_yield is used)

Notice that while the Arduino API treats this as almost the same thing, it is heavily dependent on the actual implementation and the way analog write actually happens. If it happened to be some peripheral, we may have had a wdt-prone analogWrite. What we have is a software-based one though.

asturcon3, 2022-03-23:
Ok, I understand.

I'm replacing all calls to analogWrite with setting a global variable which is tested for once at every loop, doing the actual analogWrite call just one time out of every possible infinite loop. I mean, in a production environment I need a reliable watchdog, not a instantaneous update of a glowing led.

analogWrite(pin, value,rightNow=true) would be wellcomed. With a clear explanation text in the docs.

closed 2022-03-23 by asturcon3