esp8266/Arduino #8517: esp8266wifi compilation error

Manojj1603, 2022-03-21:
Arduino: 1.8.19 (Windows 10), Board: "Arduino Yún"

Code_ESP01_SinricPro_4Relay:26:10: fatal error: ESP8266Wifi.h: No such file or directory

 #include <ESP8266Wifi.h>

          ^~~~~~~~~~~~~~~

compilation terminated.

exit status 1

ESP8266Wifi.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

mcspr, 2022-03-21:
https://github.com/esp8266/Arduino/blob/3.0.2/libraries/ESP8266WiFi/src/ESP8266WiFi.h

Unless you are using case-insensitive filesystem, `F` should be upper case
`#include <ESP8266WiFi.h>`

closed 2022-03-21 by mcspr

Manojj1603, 2022-03-22:
Thanks for your solution I'll try

Manojj1603, 2022-03-22:
The error is not rectified sir

d-a-v, 2022-03-22:
> Arduino: 1.8.19 (Windows 10), Board: "Arduino Yún"

Did you select the right board ?