esp8266/Arduino #8520: ESP8266 Can't connect to WIFI network and can't work as access point

mohamedsemz, 2022-03-22:
### Problem Description
I used the WiFiAccessPoint example and I couldn't find the network using both laptop and mobile phone.

I tried to run the WiFiScan example to check if the hardware is working or not. The ESP couldn't find my wifi, phone hotspot. But it only finds mobile hotspot created from laptop. I checked the router and confirmed that the router is working on 2.4 GHz not 5 GHz.

I tried the HelloServer example with my wifi network name (which is couldn't be found in the Scan example) and the status is WL_DISCONNECTED for sometime and then WL_NO_SSID_AVAIL.

I tried the HelloServer example with hotspot created from the laptop (which is found in the Scan example) and the status is WL_DISCONNECTED for sometime and then WL_CONNECT_FAILED.

What is the issue and how to fix it ? Or how o debug this kind of issues ?

Note:
I'm using Core version 3.0.2, Nodemcu with ESP8266 and Arduino IDE.
I also tested 3.0.0, 2.7.2, 2.4.2 and all of them not working too.

lorerave85, 2022-06-30:
I have a very similar problem:

the same code that I use regularly for other esp8266 (and that work) has stopped working and now going to debug it no longer sees any wifi network.

i installed arduino IDE 1.8.19 on my new mac with M1 processor, and it doesn't seem to work anymore.

core version 3.0.2 which has been standing still for almost a year.

d-a-v, 2022-07-01:
Did you try the option "erase flash > all flash content" ?

mohamedsemz, 2022-07-01:
I still have the same issue and didn't fix it yet. 

> Did you try the option "erase flash > all flash content" ?

I tried it back then and had the same issue.

lorerave85, 2022-07-01:
I tested all core versions from 2.4 up to 3.0.2.
every time I erased everything.
today I try with an external power supply, maybe the USBC-USB-ESP8266 adapter can't power it

lorerave85, 2022-07-01:
no change, same error: no SSID found

voblaunsane, 2022-07-03:
same problem, running this basic code does not set a SSID nor does it set a password. what's going on?
erasing all contents; usually powering from PC usb, but also tried 2A wall chargers with same results..

```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>

ESP8266WebServer server(80);

void handleRoot() {
  server.send(200, "text/html", "<h1>You are connected</h1>");
}

void setup() {
  delay(1000);
  Serial.setDebugOutput(true);
  Serial.begin(115200);
  
  WiFi.softAP("ESPap", "thereisnospoon");

  server.on("/", handleRoot);
  server.begin();

  WiFi.printDiag(Serial);
}

void loop() {
  server.handleClient();
}
```

```
SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
fpm close 3 
mode : softAP(26:a1:60:3a:b7:98)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
bcn 0
del if1
usl
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
Mwifi evt: 8
ode: AP
PHY mode: N
Channel: 1
AP id: 0
Status: 255
Auto connect: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
```

config: https://i.postimg.cc/GhcP23ZJ/Untitled.jpg

lorerave85, 2022-07-04:
I made an even simpler test with the example of the CORE 3.0.2 library - ESP8266WiFi.h
WifiScan.ino

result:
wemos d1 mini (on the right): everything ok
wemos d1 mini v3 (on the left): cannot find any networks

[image]