esp8266/Arduino #8530: Add support for inverting UART1

PaulWieland, 2022-04-06:
I found this [pull request](https://github.com/esp8266/Arduino/pull/6816/files), and consequently [this section](https://github.com/esp8266/Arduino/blob/584d2f239290ab533fe513ac399ee17091c5f954/cores/esp8266/uart.cpp#L699) of the library which makes it possible to invert UART0.

This request is to add the same functionality for UART1.

In the meantime, I found this [blog post](https://www.analysir.com/blog/2017/01/29/updated-esp8266-nodemcu-backdoor-upwm-hack-for-ir-signals/) which gives a workaround that does work (tested on a Wemos D1 Mini)

```
#include "uart_register.h"  // Place this at the top of the sketch 
 
#define SET_PERI_REG_MASK(reg, mask) WRITE_PERI_REG((reg), (READ_PERI_REG(reg)|(mask)))
SET_PERI_REG_MASK(UART_CONF0(UART1) , BIT22);
```

mcspr, 2022-04-06:
Just for verbosity sake, the source you mentioned is this one.
https://github.com/esp8266/Arduino/blob/584d2f239290ab533fe513ac399ee17091c5f954/cores/esp8266/uart.cpp#L699-L702
https://github.com/esp8266/Arduino/blob/584d2f239290ab533fe513ac399ee17091c5f954/cores/esp8266/esp8266_peri.h#L268
The equivalent operation would be to set `BIT(UCTXI)` in the config for the uart object when `uart_nr == UART1`
`UCTXI` is `22` aka 22th bit you set via the mask. And, assuming we don't need the rest of the bits mentioned in the peripherals header.

mcspr, 2022-04-08:
...do you want to do a PR? :)

mcspr, 2022-06-13:
based on both RTOS and our code, it is
setting the rest of the bits is useless in case of uart1, but it if does not seem to care - it just might be ok

closed 2022-06-14 by mcspr