esp8266/Arduino #8536: esp8266webserver question: How to run some own code before server.serveStatic();

chess-levin, 2022-04-09:
How do I run some own code before `server.serveStatic();` ?

I have a html file stored in LittleFS and serve it via `httpServer.serveStatic("/action", LittleFS, "/action.html");`
Now I want to execute some code first when someone requests `/action` and then send `action.html` , but I can't find no other way than to write a handler function that executes my code and then reads the file into a `String fileBuffer` and finally sending the String via `httpServer.send(200, HTML, fileBuffer);`

Does anybody has a hint for a better way of doing this?

I'using latest version of esp8266webserver on an wemos d1 mini with 4MB.

paulocsanz, 2022-04-09:
You have to create your own handler, it's not hard tho. You just have to reimplement two places, the boilerplate can get annoying tho.

Implement your custom handler class, like this code:

https://github.com/esp8266/Arduino/blob/3f4bcbe483b73a02ea49eb65493dc95ae6e8a678/libraries/ESP8266WebServer/src/detail/RequestHandlersImpl.h#L213

Just copy this entire class and add your logic before the last line of the handle function (the sendStream call).

And set this custom handler class instance to a route like this line: 

https://github.com/esp8266/Arduino/blob/f401f08aba74f6dd0f6dbeb9b1b6b6d94b174b31/libraries/ESP8266WebServer/src/ESP8266WebServer-impl.h#L275

Something like

```
server.addHandler(new YourStaticFileRequestHandler<WiFiServer>(fs, path, uri, cache_header));
```

chess-levin, 2022-04-09:
Great! I'll give it a try! THX

chess-levin, 2022-04-11:
Hello @paulocsanz, I've tried to write my own  `StaticRequestHandler` but getting a compiler error. I have no experiences with writing templates in c++.  I put this code in a file named ``MyRequestHandler.cpp` next to my `main.cpp`.

Could you give me some more hints on how to get this running. Big thank you.

```c++
template<typename ServerType>
class MyStaticFileRequestHandler
    :
public StaticRequestHandler<ServerType> {

    using SRH = StaticRequestHandler<ServerType>;
    using WebServerType = ESP8266WebServerTemplate<ServerType>;

public:
    MyStaticFileRequestHandler(FS& fs, const char* path, const char* uri, const char* cache_header)
        :
    StaticRequestHandler<ServerType>{fs, path, uri, cache_header}
    {
    }

    bool canHandle(HTTPMethod requestMethod, const String& requestUri) override  {
        return SRH::validMethod(requestMethod) && requestUri == SRH::_uri;
    }

    bool handle(WebServerType& server, HTTPMethod requestMethod, const String & requestUri) override {
        if (!canHandle(requestMethod, requestUri))
            return false;

        if (server._eTagEnabled) {
            if (server._eTagFunction) {
                _eTagCode = (server._eTagFunction)(SRH::_fs, SRH::_path);
            } else  if (_eTagCode.isEmpty()) {
                _eTagCode = esp8266webserver::calcETag(SRH::_fs, SRH::_path);
            }

            if (server.header("If-None-Match") == _eTagCode) {
                server.send(304);
                return true;
            }
        }

        File f = SRH::_fs.open(SRH::_path, "r");

        if (!f)
            return false;

        if (!f.isFile()) {
            f.close();
            return false;
        }

        if (SRH::_cache_header.length() != 0)
            server.sendHeader("Cache-Control", SRH::_cache_header);

        if ((server._eTagEnabled) && (_eTagCode.length() > 0)) {
            server.sendHeader("ETag", _eTagCode);
        }

        // my code
        // lsPause();

        server.streamFile(f, mime::getContentType(SRH::_path), requestMethod);
        return true;
    }

protected:
    String _eTagCode; // ETag code calculated for this file as used in http header include quotes.
    
};
```

```
Compiling .pio\build\esp12e\src\RequestHandler.cpp.o
src\RequestHandler.cpp:4:28: error: expected template-name before '<' token
    4 | public StaticRequestHandler<ServerType> {
      |                            ^
src\RequestHandler.cpp:4:28: error: expected '{' before '<' token
Archiving .pio\build\esp12e\lib186\libESP8266WiFi.a
Archiving .pio\build\esp12e\libcfd\libesp8266_mdns.a
Archiving .pio\build\esp12e\libedf\libWire.a
Archiving .pio\build\esp12e\lib5d5\libBH1750.a
Archiving .pio\build\esp12e\libbfb\libSimpleTimer.a
Compiling .pio\build\esp12e\lib44b\ESP8266WebServer\detail\mimetable.cpp.o
Compiling .pio\build\esp12e\libe74\ESP8266mDNS\ESP8266mDNS.cpp.o
Compiling .pio\build\esp12e\libe74\ESP8266mDNS\LEAmDNS.cpp.o
Compiling .pio\build\esp12e\libe74\ESP8266mDNS\LEAmDNS_Control.cpp.o
Compiling .pio\build\esp12e\libe74\ESP8266mDNS\LEAmDNS_Helpers.cpp.o
*** [.pio\build\esp12e\src\RequestHandler.cpp.o] Error 1
Compiling .pio\build\esp12e\libe74\ESP8266mDNS\LEAmDNS_Structs.cpp.o
```